emsesp/EMS-ESP32 #766: Changing a value works at first, but is then reset to it's original value

daad919, 2022-11-23:
**Bug description**
I'm using home assistant & node red to set mode and temp, and every couple of days, i'm not able to set the mode and temp anymore. If i change the mode from ex day to auto, it changes back to day after a second delay.
Wen i'm using the webbrowser to write the command, i get the error: write command failed.

Only thing that helps is to restart the ems in the webbrowser. Then i can set the temp and mode as i wish. 

I've already tried the fix with sending 0 to boiler/selflowtemp every few seconds, but the system does not respond.

I'm on ems V3.4.4

**Steps to reproduce**
Just happens random

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
{
  "System Status": {
    "version": "3.4.4",
    "uptime": "000+04:32:48.537",
    "freemem": 128,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Status": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "8C:CE:4E:94:BC:17",
    "IPv4 address": "10.0.1.201/255.255.255.0",
    "IPv4 gateway": "10.0.1.1",
    "IPv4 nameserver": "10.0.1.1",
    "phy type": 1,
    "eth power": 16,
    "eth phy addr": 1,
    "eth clock mode": 0,
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 38182,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 36000,
    "clean session": true,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 1,
    "publish time solar": 10,
[... 146 lines cut ...]
2022-11-23 16:33:22.021 D 60621: [mqtt] Publishing topic ems-esp/thermostat_data (#41365, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:22.122 D 60622: [mqtt] Publishing topic ems-esp/switch_data (#41366, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:23.031 D 60623: [mqtt] Publishing topic ems-esp/thermostat_data (#41367, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:23.132 D 60624: [mqtt] Publishing topic ems-esp/switch_data (#41368, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:24.043 D 60625: [mqtt] Publishing topic ems-esp/thermostat_data (#41369, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:24.144 D 60626: [mqtt] Publishing topic ems-esp/switch_data (#41370, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:25.053 D 60627: [mqtt] Publishing topic ems-esp/thermostat_data (#41371, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:25.154 D 60628: [mqtt] Publishing topic ems-esp/switch_data (#41372, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:26.063 D 60629: [mqtt] Publishing topic ems-esp/thermostat_data (#41373, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:26.164 D 60630: [mqtt] Publishing topic ems-esp/switch_data (#41374, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:27.073 D 60631: [mqtt] Publishing topic ems-esp/thermostat_data (#41375, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:27.174 D 60632: [mqtt] Publishing topic ems-esp/switch_data (#41376, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:28.083 D 60633: [mqtt] Publishing topic ems-esp/thermostat_data (#41377, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:28.184 D 60634: [mqtt] Publishing topic ems-esp/switch_data (#41378, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:29.093 D 60635: [mqtt] Publishing topic ems-esp/thermostat_data (#41379, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:29.194 D 60636: [mqtt] Publishing topic ems-esp/switch_data (#41380, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:30.007 D 60637: [mqtt] Publishing topic ems-esp/boiler_data_ww (#41381, retain=0, retry=1, size=484, pid=1)
2022-11-23 16:33:30.108 D 60638: [mqtt] Publishing topic ems-esp/boiler_data (#41382, retain=0, retry=1, size=794, pid=1)
2022-11-23 16:33:30.209 D 60639: [mqtt] Publishing topic ems-esp/thermostat_data (#41383, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:30.310 D 60640: [mqtt] Publishing topic ems-esp/mixer_data (#41384, retain=0, retry=1, size=117, pid=1)
2022-11-23 16:33:30.411 D 60641: [mqtt] Publishing topic ems-esp/switch_data (#41385, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:31.017 D 60642: [mqtt] Publishing topic ems-esp/thermostat_data (#41386, retain=0, retry=1, size=1871, pid=1)
2022-11-23 16:33:31.118 D 60643: [mqtt] Publishing topic ems-esp/switch_data (#41387, retain=0, retry=1, size=30, pid=1)
2022-11-23 16:33:31.185 D 60644: [command] Calling command 'thermostat/mode' (mode) with value night
2022-11-23 16:33:31.185 E 60645: [emsesp] Write command failed callback function failed (Error)

daad919, 2022-11-23:
> Heatingactivated is set to on, and i can't write to this command.
This is normal with BC10 controller, the boiler ignores the values. There is nothing we can do. It's the same with my                 GB125/MC10/BC10, bus protocol is ok, but value is unchanged.

Ok!

>You are using very, very short publish intervals, it could be memory related. Try to set all intervals to 10sec or                             longer. the data from bus normally are not faster than 10 sec, mixer publishes only every minute.

They where at 1 sec, changed back to 10 sec!

>I'm supprised that the first mode command (D655) works (D660 to D662), the initialization of thermostat hasn't finished yet, the telegram is not registered. How often are you sending commands?

Multiple times a day (20?30?) 

> 022-11-23 16:33:31.185 D 60644: [command] Calling command 'thermostat/mode' (mode) with value night
2022-11-23 16:33:31.185 E 60645: [emsesp] Write command failed callback function failed (Error)
If you try again, is it always the same result? What is shown for free heap /free block when this happends?

What do you mean with free heap/free block?

 > Is there a possibility to restart the ems from within HA or NR? This way i could temporary fix the problem this way.

>Sure, command is: system/restart

Will look into it to tackle the problem!

MichaelDvP, 2022-11-23:
> What do you mean with free heap/free block?

Heap (Free/Max Alloc) from this page.
[image]