emsesp/EMS-ESP32 #768: EMS-ESP Version v3.5.0b10 - "wwactive" no values ???

stbuerger, 2022-11-25:
I resently switched from V2 to ESP32 v3.5.0b10 and don't see any data for wwactive ("dhw active ").
Neither in DASHBOARD nor MQTT.
In DASHBOARD: dhw active constantly shows as "off"
In MQTT - which is configured with "Boolean Format 1/0" - value is constantly "0". Similar behavior with "Boolean Format on/off".

May I have missed something when moving from V2 to ESP32 V3.x ?

How can I debug what the program gets from EMS bus and how it processes this ?

Thx for this wonderfull software anyway
-stefan

proddy, 2022-11-25:
was it called `wwactive` before in the MQTT ?

stbuerger, 2022-11-25:
in V2 MQTT it must have been `ems-esp/boiler_data_ww { .. "wWActive": .. }` - unfortunately already deleted all old configs :-(
What I'm actually looking for is an indicator for "burning gas because of warm water request" equivalent to and to differentiate from "heatingactive".

MichaelDvP, 2022-11-25:
Please specify more exact, do you *"don't see any data"* or *"constantly shows off"*?
Are you sure v2 shows this value/a change in this value? i can not see any difference in handling of v2:
https://github.com/emsesp/EMS-ESP/blob/dev/src/devices/boiler.cpp#L1137
and v3:
https://github.com/emsesp/EMS-ESP32/blob/dev/src/devices/boiler.cpp#L898
only the renameing from `wWActive` to `wwactive`.
For ems+ boilers (telegram 0xE7) this value is not shown for v2 and v3, also no difference.
 > How can I debug what the program gets from EMS bus and how it processes this ?

Use terminal and `watch 8 34` and check 6'th byte (offset 5) for changes in bit 6.

MichaelDvP, 2022-11-25:
I've checked on my boiler (Buderus GB125, ems, wwtype buffer, chargepump): wwactive is never set by the boiler, `wwcharge` is set while dhw preparation with burner, when burner goes off, it switches to `wwrecharge`, chargepump continous some minutes and dhw is heated by the hotter boiler water. Also the `tapwater_active` topic is set the whole time of dhw preparation.

stbuerger, 2022-11-25:
Sorry - with "don't see any data", I ment: Value is constantly showing "off" or "0" depending on "Format Boolean".
I'm absolutely sure that in V2 I've changes - just not sure what parameter/topic this was. Only proof that I still have available is an annotated screenshot from Grafana:
[image]
which makes mw wonder about this parameter might have been named "wWHeat" or so ???

stbuerger, 2022-11-25:
Thx a lot @MichaelDvP ! I've found that ems-esp/boiler_data_ww {.. "wwcharging": ..} seems to be what I'm looking for.
Sofar I couldn't notice any change in { .. "wwrecharging": .. } so. But that may be related to the  settin of warm water "mode" ("wwcomfort" ???) as this is set to AUTO on my BC10 ("wwcomfort" : "hot"). Will test what it changes when I set it to ECO (as in the past).

closed 2023-02-19 by proddy