emsesp/EMS-ESP32 #769: E32 with v3.5.0b10 - application crashing / blank pages

lmbd, 2022-11-25:
**Bug description**
*I've started to observe recently some blank pages since running I'm running my E32 on v3.5.0b10.  when trying to open e.g. system log.*

**Steps to reproduce**
*Use similar config on an E32 (see attached config file*
*open /system/log/ with max buffer size and log level all*

**Screenshots**
https://user-images.githubusercontent.com/3472161/204029795-8da8af59-1c26-41ab-bb69-1ac565b9f86e.mov

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api/system*

lmbd, 2022-11-25:
disabling multiple functionalities (e.g. telnet console, telnet console) helped to get rid of in most cases the blank pages. Also  http://ems-esp.local/api/system started to return info:

{
  "System Info": {
    "version": "3.5.0b10",
    "platform": "ESP32",
    "uptime": "000+00:07:50.001",
    "freemem": 70,
    "free_app": 75,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -65,
    "IPv4 address": "192.168.188.110/255.255.255.0",
    "IPv4 gateway": "192.168.188.1",
    "IPv4 nameserver": "192.168.188.250",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": true,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 554,
    "MQTT queued": 1,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 724,
    "bus reads (tx)": 211,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 8,
    "bus reads failed": 2,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 5,
    "tx gpio": 17,
    "dallas gpio": 4,
    "pbutton gpio": 33,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": false
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",
      "device id": "0x08",
      "product id": 123,
      "version": "07.00",
      "entities": 67,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "74.01",
      "entities": 48,
      "handlers received": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x02CE 0x02D0 0x02D2 0x031E",
      "handlers ignored": "0x1A 0x23 0x35"
    },
    {
      "type": "Solar",
      "name": "SM100/MS100",
      "device id": "0x30",
      "product id": 163,
      "version": "27.05",
      "entities": 36,
      "handlers received": "0x0362 0x0363 0x0366 0x0364 0x036A",
      "handlers fetched": "0x0358 0x035A 0x0380 0x038E 0x0391",
      "handlers pending": "0x035D 0x035F 0x035C 0x0361",
      "handlers ignored": "0x0368 0x03A9 0x03AA 0x35 0xBF"
    },
    {
      "type": "Controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.05",
      "entities": 0
    }
  ]
}

proddy, 2022-11-25:
You only have 4 devices, I'm not sure why your free heap memory is so low

lmbd, 2022-11-25:
free memory the last 5 days
[image]
- I've enable analog sensors / added a sensor with counter on the 25th around noon

Could the issue be related with analog sensors?

proddy, 2022-11-25:
but still this doesn't look normal. Are you using a BBQKees board?

MichaelDvP, 2022-11-25:
The system/info shows the maxAllocHeap, this can be much lower than freeHeap, especially after HA autoconfig with all devices recognized: This is mine after a HA creation:
[image]
The slow decrease is imo the weblog buffer getting filled, this is eating much ram. I think we should auto-reduce the buffer at higher memory, maybe 64k. I have only tested with ESP32-S2, but this chip can alloc the json buffers in psram ,The ESP32 does not get to limit in my system. I'll change to E32-Gateway to test.

lmbd, 2022-11-25:
> but still this doesn't look normal. Are you using a BBQKees board?

yes, it's a E32 bought directly from BBQKees (Q3 / 2022)

proddy, 2022-11-29:
>The slow decrease is imo the weblog buffer getting filled, this is eating much ram.

@MichaelDvP  but he hits 70 KB free heap after 7 minutes, with only 3 devices. It doesn't look right. Shall we include the `settings.weblog_buffer` in the system info command so it's captured in the support json?

proddy, 2022-12-09:
@lmbd we're not sure how to help with this and need more details from you

lmbd, 2022-12-14:
@proddy any hints how I can help or which information could I provide that would be helpful?

proddy, 2022-12-15:
I think it's memory related. A few things to get clarity on
 - does the system crash? If it crashes, are you doing anything (making an API call, on the web UI, telnet etc...)
 - the Web blank screen, when does that occur? Soon after EMS-ESP boots or a long time after? Does it always happen?
 - what is the available heap memory (check via system API call)
 - did you reduce the weblog buffer like Michael recommended?
 - does it happen when you disable the analog sensor (maybe this is eating the memory as you said)
 
Also upgrade to the latest b12 as Michael had put more memory checks into the web code. And attach the latest support info file.