emsesp/EMS-ESP32 #770: Question about how to turn on and off the heating (keeping warmwater on) with EMS-ESP32

joeheaven, 2022-11-27:
Hi there,

maybe it's a silly question: My familiy sometimes forgets to turn off the heating when opening the windows. Currently I use the turning knob to set a heating temperature of 0, which turns of the heating completely (no pump running, no gas used). I don't know of any other way to prevent gas being used to heat, maybe summer/winter-setting?

Closing the valves is not really an option, since not all of them are remote controlled and some (AVM) are really slow in adopting the new value. And the auto-windows-mode seems to do nothing.

Anyhow, I would love top have an option to remotely turn on/off the heating (but keeping the warm water running).

My installation (as reported by EMS-ESP32):
Boiler BK13/BK15/Smartline/GB1x2
Thermostat RC35
EMS-ESP33 v3.4.2 (soon to be updated)

Are there any ideas? After I turned the knob quite many values change in the status reported by EMS-ESP32, and I'm not sure which one is possible and safe to write.

Again, please excuse this maybe simple question, if there is a quick solution, it might not bother you too much.

Best regards,
Jochen

proddy, 2022-11-27:
I'm not an expert here. Which boiler do you have? On a combi gas boiler you perhaps could force the 3-way-valve? @BBQKees and @MichaelDvP would know more than me.

joeheaven, 2022-11-27:
Proddy, thanks a lot for the prompt response.
I have a Buderus Logamax plus GB152-24 T 170SR and a RC35.
I don't expect to get advice on how I to use the system as such, my question is simply if (and how) I could simulate turning the knob for the heating temperature to "off", so that in my case the pump stops and no gas is used for heating. 
I tried several ways (ems-esp/boiler_data/heatingactivated, ems-esp/boiler_data/heatingtemp), but the values jump back to the previous values. Which is maybe the right thing, since this is what is selected at the panel.

MichaelDvP, 2022-11-27:
> if (and how) I could simulate turning the knob for the heating temperature to "off"

Simple answer: you can't.
The control panel sets the values to the boiler according o the knob position, you can query it from contoller `read 9 16` give back 2 bytes  for heating on/off and temperature. If you write these values to the boiler, the controller immediatly overwrites to knob position. You also can't write the values to the controller. Also flowtemp is controlled by thermostat, if you try to set it to boiler, the thermostat will overwrite.

You can try two ways:
1. set thermostat summertemp to 9 (always summermode) and the thermostat wll stop pump and heating, but it takes some minutes.
2. Set boiler into test-mode and switch off burner and pump. `system send "0B 08 1D 00 5A 00 00 00"` and to end testmode `system send "0B 08 1D 00 00"`

joeheaven, 2022-11-30:
@MichaelDvP The first way ist absolutely sufficent and more or less exactly what I wanted! I will use that to turn off the heating when windows are open for a certain time below a certain temperature. Perfect!

Thanks very, very much!

closed 2023-02-19 by proddy

RonnAndrea, 2025-04-14:
Dear MichaelDvP, joeheaven, proddy,

My case is: https://github.com/emsesp/EMS-ESP32/discussions/2516

I think that a good goal for me would be to write values to the two knob position variables in my Bosch Junkers Cerapur boiler as MichaelDvP explains above...!!!

I'm new to this, and there are several levels of complexity, and I’m short of time…
I see it like this:
1.	HA Data from/to EMS-ESP Gateway over MQTT
2.	EMS-ESP Gateway Data from/to Boiler over EMS-Bus 
---
3.	In the EMS-ESP Gateway I would like to create two knob setpoint variables (heating temperature, and the DHW – Domestic Hot Water).
4.	The EMS-ESP Gateway, on its startup, reads the two knob settings from the boiler into the knob setpoints, after that, the EMS-ESP Gateway cyclic writes the two setpoints to the boiler.
5.	When HA takes control, it then writes the two calculated setpoints to the EMS-ESP Gateway cyclic. This way HA takes control of the two knob positions in centigrade…
6.	Best of all, also be able to write the setpoints from the EMS-ESP-Gateway web interface…

As far as I can see, I have two options: 
1.	That somebody helps me as a paid service, is this a valid option?, 
2.	or I must study, read the manuals, understand the stuff, and make it work when things get under control… Not the best solution for me!
Thanks… 
and please let me know if you have a good and qualified proposal...