emsesp/EMS-ESP32 #772: home assistant themorstate mode change doesn't work in beta 3.5.0.b10 (or earlier)

jtonk, 2022-11-27:
**Bug description**
changing mode to OFF or HEAT in Home assistant doesn't work in version v3.5.0b10 or earlier.
in version 3.4.4 it works fine.

**Steps to reproduce**
upgrade to beta v3.5.0b10. change the mode from AUTO to HEAT or OFF. nothing happens on the thermostat.

**Expected behavior**
change of heating mode.

**Screenshots**
n/a

**Device information**
I don't thing the message below is related...
{
  "message": "unknown device"
}

**Additional context**
mqtt home assistant auto discovery
`{
	"~": "ems-esp",
	"object_id": "thermostat_hc1",
	"name": "Hc1",
	"uniq_id": "thermostat_hc1",
	"mode_stat_t": "~/thermostat_data",
	"mode_stat_tpl": "{%if value_json.hc1.mode=='manual'%}heat{%elif value_json.hc1.mode=='day'%}heat{%elif value_json.hc1.mode=='night'%}off{%elif value_json.hc1.mode=='off'%}off{%else%}auto{%endif%}",
	"temp_cmd_t": "~/thermostat/hc1/seltemp",
	"temp_stat_t": "~/thermostat_data",
	"temp_stat_tpl": "{{value_json.hc1.seltemp}}",
	"mode_cmd_t": "~/thermostat/hc1/mode",
	"curr_temp_t": "~/thermostat_data",
	"curr_temp_tpl": "{{value_json.hc1.currtemp}}",
	"min_temp": "5",
	"max_temp": "30",
	"temp_step": "0.5",
	"modes": ["auto", "heat", "off"],
	"dev": {
		"ids": ["ems-esp-thermostat"]
	}
}`

proddy, 2022-11-27:
Yeah, I get the same error. Will dig into what's broke. Thanks for reporting.

proddy, 2022-11-27:
Looks like publishing to the topic 'ems-esp/thermostat/hc1/mode' with any value fails.

MichaelDvP, 2022-11-27:
@jtonk Could you please check with https://github.com/MichaelDvP/EMS-ESP32/releases if my fix is working (i don't use HA).

proddy, 2022-11-27:
I just tested your build @MichaelDvP with my HA and it works perfectly.

jtonk, 2022-11-27:
I just tested it as well, it works again!

commit d583409a (2022-11-28): set mode for ha climate, fix #772
closed 2022-11-28 by proddy