emsesp/EMS-ESP32 #773: Write command failed callback function failed on Bosch FW200/FB10 and FB100

akanto, 2022-11-27:
**Bug description**
I have upgraded to 3.4.4 and after that, I can't change any thermostat value anymore due to a "write command failed error message".

Before the upgrade, I used this version: https://github.com/emsesp/EMS-ESP32/issues/336#issuecomment-1055643227

**Steps to reproduce**
I have tried to change from "auto" heating mode to "eco", and "heat" and it just failed:

```
2022-11-27 17:44:41.760 DEBUG 2932: [emsesp] Updating details for already active deviceID 0x20
2022-11-27 17:44:41.892 TRACE 2933: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 30 01 C6 01 C6 A1 00 00 03 00 01 74 63 00 A7 94 00 80 00
2022-11-27 17:44:42.097 TRACE 2934: [emsesp] Thermostat(0x1A) <- 30(0x30), Version(0x02), length: 0x0A
2022-11-27 17:44:42.189 DEBUG 2935: [command] Calling command 'thermostat/mode' (mode) with value eco
2022-11-27 17:44:42.189 ERROR 2936: [emsesp] Write command failed callback function failed (Error)
2022-11-27 17:44:42.329 TRACE 2937: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 01 05 07 00 00 00 00 00 00 00 00 00
2022-11-27 17:44:42.609 TRACE 2938: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 23 01 5E 64 26 09 03 25 C0 80 00 01 C6 01 43 00 6C FF 00 00 00 C8 00 02 18
2022-11-27 17:44:42.914 TRACE 2939: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 00 3A 80 00 80 00 00 00 00 64 01 12 8A 09 1C 06 00 00 00 07 A7 A3 00 6A F6 80 00
2022-11-27 17:44:43.132 TRACE 2940: [emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 90 03 0B 12 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2022-11-27 17:44:43.464 TRACE 2941: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 05 01 05 80 00 00 80 00 80 00 80 00 00
2022-11-27 17:44:43.668 TRACE 2942: [emsesp] Thermostat(0x10) <- Mixer(0x21), Version(0x02), length: 0x0A
2022-11-27 17:44:43.733 TRACE 2943: [emsesp] Mixer(0x21) -> Thermostat(0x10), Version(0x02), data: 66 14 07 00 00 00 00 00 64 01
```
[log.txt](https://github.com/emsesp/EMS-ESP32/files/10098776/log.txt)

I have seen the same error when I just tried to control the temperature, but it looks like that reading data works correctly and whenever I change anything directly on the thermostat it is reflected back to ems-esp.

proddy, 2022-11-27:
similar to #772

something is broken for sure!

MichaelDvP, 2022-11-27:
@akanto Does it work with  hc set: `thermostat/hc1/mode`, is mode the only command with this message? Is it on all devices or thermostat only?

@proddy No, not similar, #772 is v3.5 with HA climate (off/heat/auto), here it is v3.4.4 and Junkers commands. The mode setting in v3.5 is different because of localization and a climate enum is missing, Set [here](https://github.com/emsesp/EMS-ESP32/blob/af8e82a86090ae5cf3a25cde2ce01542a76202de/src/devices/thermostat.cpp#L2252):
```
        if (!Helpers::value2enum(value, enum_index, mode_list)) {
            mode_list = FL_(enum_mode_ha);
            if (Mqtt::ha_enabled() && !Helpers::value2enum(value, enum_index, mode_list)) {
                LOG_WARNING("wrong mode: %s", value);
                return false; // not found
            }
        }
```
with a new enum_mode_ha (off, heat, auto)

akanto, 2022-11-28:
I have seen https://github.com/emsesp/EMS-ESP32/issues/772, but I am using v3.4.4 release, and to narrow down the error surface, I am just using the EMS ESP frontend directly and not HA.

@MichaelDvP Changing values on the boiler work correctly. I have tried to change the "dhw comfort" to eco/hot on the boiler and it was correctly applied, so it seems that  I have problems with the thermostats only.

MichaelDvP, 2022-11-28:
What about other thermostat commands?
Does it work if you specify the hc? You have 3 hcs, it's uncommon to send a command without hc.

akanto, 2022-11-28:
No write command works on any thermostat, only changing values directly on the boiler works correctly.

proddy, 2022-11-28:
from the Console too? Can you show the output so we can reproduce

akanto, 2022-11-28:
I have tried from the UI only, but I will try to figure out how it can be done directly from the console.

proddy, 2022-11-28:
https://emsesp.github.io/docs/#/Console

akanto, 2022-12-26:
Unfortunately, I was unable to figure out what is the proper telnet command to set anything. 

```
ems-esp:$ show
....
Thermostat: FW200 (DeviceID:0x10, ProductID:106, Version:12.13) (29)
  date/time: 26.12.2022 10:23
...
  hc1 selected room temperature: 20.0 °C
...
  hc1 mode: auto
  hc1 mode type: heat
....
```

The help command did not really guide me on how to set either the "hc1 mode" to e.g. "eco" or the selected room temperature. When I did some educated guesses, and I issued commands like:
```
ems-esp:# call thermostat hc1 mode eco

Connection closed by foreign host
```

The entire device was just restarted automatically.

proddy, 2022-12-26:
> Unfortunately, I was unable to figure out what is the proper telnet command to set anything.

Typing `help` into the console will show the syntax, which is `call [device] [cmd] [data] [id|hc]` (you need to su first)

And when you in the console hit TAB and it'll show you the next available commands. So in your case the command is `call thermostat mode eco 1`.

You can also call this direct from an MQTT topic or the HTTP API using `ems-esp/thermostat/hc1/mode`.

> ems-esp:# call thermostat hc1 mode eco
> Connection closed by foreign host

That's a new bug! I'll make a new GH issue for that to fix.

akanto, 2022-12-26:
Thanks. So I was able to test the command:
```
ems-esp:# call thermostat mode eco 1
ems-esp:# call thermostat mode eco 2
ems-esp:# call thermostat mode eco 3
```

It looks like hc1 and hc2 can be controlled with the above commands over telnet, what is odd, is that if I go to the UI and change the settings there it is also successful now... So I don't understand, because it seems that the original issue has disappeared for hc1 and hc2.

Regarding hc3, it seems that the `call` command has no impact. Kindly note, that I had a Bosch setup that you had patched a few months back: https://github.com/emsesp/EMS-ESP32/issues/336#issuecomment-1055643227

So, in a nutshell, I can control hc1 and hc2 over telnet console and over UI as well (what I don't understand, since I had changed nothing), but at least it works now. 

The hc3 can't be controlled either from the telnet console or from UI, might be related to your former patch that was a fix for the tricky bosch setup:  https://github.com/emsesp/EMS-ESP32/issues/336#issuecomment-1055643227

I have executed a few ```ems-esp:# call thermostat mode eco 3``` commands, you can find the logs, here:

[log.txt](https://github.com/emsesp/EMS-ESP32/files/10304122/log.txt)

MichaelDvP, 2022-12-26:
> I have executed a few ems-esp:# call thermostat mode eco 3 

This is the same problem with multiple thermostats as in #826 and is fixed in v3.5.

akanto, 2022-12-26:
Ah, ok, let me try to upgrade to the latest dev version again. What I have seen in v3.5 is that you have changed mqtt entity names, so I need to do some changes in HA, which might take some time.

proddy, 2022-12-27:
> Ah, ok, let me try to upgrade to the latest dev version again. What I have seen in v3.5 is that you have changed mqtt entity names, so I need to do some changes in HA, which might take some time.

if you take the latest dev build, it wil be backwards compatible and your old HA entities will still work.

akanto, 2022-12-27:
Thanks, you are right it works with the latest dev build, so we can consider this issue as solved. However the frontend is broken for me on the latest dev build, I have filed this issue: https://github.com/emsesp/EMS-ESP32/issues/847

MichaelDvP, 2022-12-27:
I think your'e not on the latest. This was s Copy past error in my first commit.