emsesp/EMS-ESP32 #774: minor UI bugs with decimal values (adding / editing analog sensor dialog)

lmbd, 2022-11-27:
**Bug description**
While playing around with the analog sensor dialog I've identified 2 minor UI bugs
1. hint texts for start value and factor are misleading as they're stating that only integers are valid (which is not true)
2. it's not possible to enter decimal values with an leading 0 after decimal point (only possible with a workaround - see recording)

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

https://user-images.githubusercontent.com/3472161/204154951-e346bc48-c187-4b6a-be0b-72b21b4aa99e.mov

[image]

**Device information**
[emsesp_settings (1).txt](https://github.com/emsesp/EMS-ESP32/files/10099004/emsesp_settings.1.txt)

proddy, 2022-11-27:
very cool the video! As they say,  a picture is a thousand words. We can update the hint text. The leading zero we need to fix. Both @MichaelDvP and myself have tried different techniques but it remains challenging.

MichaelDvP, 2022-11-27:
We had this issue in: https://github.com/emsesp/EMS-ESP32/pull/755#issuecomment-1320900326
It depends on browser and the "step" setting of input field.

lmbd, 2022-11-27:
I would suggest to support decimals as there are gasmeters which are sending an impulse for each 0,01 m^3
[image]

[image]

MichaelDvP, 2022-11-27:
Decimals are supported, value is float, only the browsers handles the input different and give warning if input does not match the step setting. We can remove the up/down arrows and step setting to solve this.

MichaelDvP, 2022-11-27:
Ok, removing input props seems to work, please try https://github.com/MichaelDvP/EMS-ESP32/releases

closed 2022-12-06 by proddy