emsesp/EMS-ESP32 #777: Cannot get Wifi to connect, always comes up in access point mode

chris-0xcd, 2022-11-27:
**Question**
My EMS-ESP doesn't want to connect to my WIFI network. It always starts up in access point mode.
My access points show up in the network scanner and I select the one with the best reception. I've tried playing with the network settings but nothing seems to help. The EMS-ESP is only a meter away from the closest access point so I don't think reception is an issue.  
Not sure if it makes any difference but I have multiple access points with the same SSID but on different channels to extend the range of the Network. I never had a problem getting other devices to connect.

Any help would be appreciated.

**Device information**
```json{

  "System Status": {

    "version": "3.5.0b7",

    "platform": "ESP32",

    "uptime": "000+00:13:23.770",

    "freemem": 103,

    "free_app": 74,

    "reset reason": "Software reset CPU / Software reset CPU"

  },

  "Network Status": {

    "static ip config": false,

    "enable IPv6": false,

    "low bandwidth": false,

    "disable sleep": false,

    "AP provision mode": "disconnected",

    "AP security": "wpa2",

    "AP ssid": "ems-esp"

  },

  "NTP Status": {

    "network time": "disconnected",

    "enabled": false,

    "server": "time.google.com",

    "tz label": "Europe/Amsterdam"

  },

  "OTA Status": {

    "enabled": false,

    "port": 8266

  },

  "MQTT Status": {

    "MQTT status": "disconnected",

    "enabled": false,

    "client_id": "ems-esp",

    "keep alive": 60,

    "clean session": false,

    "base": "ems-esp",

    "discovery prefix": "homeassistant",

    "nested format": 1,

    "ha enabled": false,

    "mqtt qos": 0,

    "mqtt retain": false,

    "publish time boiler": 10,

    "publish time thermostat": 10,

    "publish time solar": 10,

    "publish time mixer": 10,

    "publish time other": 10,

    "publish time sensor": 10,

    "publish single": false,

    "publish2command": false,

    "send response": false

  },

  "Syslog Status": {

    "enabled": false

  },

  "Sensor Status": {

    "temperature sensors": 0,

    "temperature sensor reads": 0,

    "temperature sensor fails": 0,

    "analog sensors": 0,

    "analog sensor reads": 0,

    "analog sensor fails": 0

  },

  "API Status": {

    "API calls": 0,

    "API fails": 0

  },

  "Bus Status": {

    "bus status": "connected",

    "bus protocol": "HT3",

    "bus telegrams received (rx)": 917,

    "bus reads (tx)": 174,

    "bus writes (tx)": 0,

    "bus incomplete telegrams": 0,

    "bus reads failed": 0,

    "bus writes failed": 0,

    "bus rx line quality": 100,

    "bus tx line quality": 100

  },

  "Settings": {

    "board profile": "E32",

    "locale": "en",

    "tx mode": 1,

    "ems bus id": 11,

    "shower timer": false,

    "shower alert": false,

    "rx gpio": 5,

    "tx gpio": 17,

    "dallas gpio": 4,

    "pbutton gpio": 33,

    "led gpio": 2,

    "hide led": false,

    "notoken api": false,

    "readonly mode": false,

    "fahrenheit": false,

    "dallas parasite": false,

    "bool format": 1,

    "bool dashboard": 1,

    "enum format": 1,

    "analog enabled": true,

    "telnet enabled": true

  },

  "Devices": [

    {

      "type": "Boiler",

      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",

      "device id": "0x08",

      "product id": 95,

      "version": "38.02",

      "entities": 59,

      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34",

      "handlers fetched": "0x14 0x16 0x33",

      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A",

      "handlers ignored": "0x05"

    },

    {

      "type": "Thermostat",

      "name": "RC200/CW100",

      "device id": "0x18",

      "product id": 157,

      "version": "41.08",

      "entities": 39,

      "handlers received": "0x06 0xA2 0x031D",

      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02F5 0x023A",

      "handlers pending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x02CE 0x02D0 0x02D2 0x03",

      "handlers ignored": "0x1A 0x35 0x23 0xBF"

    },

    {

      "type": "Controller",

      "name": "HT3",

      "device id": "0x09",

      "product id": 95,

      "version": "38.02",

      "entities": 0

    },

    {

      "type": "Unknown",

      "name": "unknown",

      "device id": "0x13",

      "product id": 104,

      "version": "21.04",

      "entities": 0

    }

  ]

}
```

**Additional context**
system log right after reboot:

000+00:00:00.000 NOTICE 0: [emsesp] System is running in Debug mode
000+00:00:00.000 INFO 1: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
000+00:00:00.000 INFO 2: [emsesp] Loading board profile E32
000+00:00:00.000 INFO 3: [emsesp] Starting EMS-ESP version 3.5.0b7 (hostname: ems-esp)
000+00:00:00.000 INFO 4: [dallassensor] Starting Dallas sensor service
000+00:00:00.000 INFO 5: [analogsensor] Starting Analog sensor service
000+00:00:00.000 INFO 6: [emsesp] Loaded EMS device library (105 records)
000+00:00:00.000 INFO 7: [emsesp] Starting Access Point with captive portal on 192.168.4.1
000+00:00:09.377 INFO 8: [emsesp] WiFi connection dropped, stopping NTP
000+00:00:09.377 WARNING 9: [emsesp] WiFi disconnected. Reason code=15
000+00:00:14.162 INFO 10: [emsesp] WiFi connection dropped, stopping NTP
000+00:00:14.162 WARNING 11: [emsesp] WiFi disconnected. Reason code=15
000+00:00:15.967 INFO 12: [emsesp] Recognized new boiler with deviceID 0x08
000+00:00:15.987 INFO 13: [emsesp] Recognized new controller with deviceID 0x09
000+00:00:16.388 NOTICE 14: [emsesp] Unrecognized EMS device (deviceID 0x13, productID 104). Please report on GitHub.
000+00:00:16.626 INFO 15: [emsesp] Recognized new thermostat with deviceID 0x18
000+00:00:20.166 INFO 16: [emsesp] WiFi connection dropped, stopping NTP
000+00:00:20.166 WARNING 17: [emsesp] WiFi disconnected. Reason code=15
000+00:00:26.169 INFO 18: [emsesp] WiFi connection dropped, stopping NTP
000+00:00:26.169 WARNING 19: [emsesp] WiFi disconnected. Reason code=15
000+00:00:32.168 INFO 20: [emsesp] WiFi connection dropped, stopping NTP
000+00:00:32.168 WARNING 21: [emsesp] WiFi disconnected. Reason code=15
000+00:00:38.165 INFO 22: [emsesp] WiFi connection dropped, stopping NTP
000+00:00:38.165 WARNING 23: [emsesp] WiFi disconnected. Reason code=15
000+00:00:44.168 INFO 24: [emsesp] WiFi connection dropped, stopping NTP
000+00:00:44.168 WARNING 25: [emsesp] WiFi disconnected. Reason code=15
000+00:00:50.177 INFO 26: [emsesp] WiFi connection dropped, stopping NTP
000+00:00:50.177 WARNING 27: [emsesp] WiFi disconnected. Reason code=15
000+00:00:56.167 INFO 28: [emsesp] WiFi connection dropped, stopping NTP
000+00:00:56.167 WARNING 29: [emsesp] WiFi disconnected. Reason code=15
000+00:01:02.149 INFO 30: [emsesp] WiFi connection dropped, stopping NTP
000+00:01:02.149 WARNING 31: [emsesp] WiFi disconnected. Reason code=15
000+00:01:08.173 INFO 32: [emsesp] WiFi connection dropped, stopping NTP
000+00:01:08.173 WARNING 33: [emsesp] WiFi disconnected. Reason code=15

proddy, 2022-11-28:
are you using a BBQKees E32 board with ethernet?

chris-0xcd, 2022-11-28:
Yes but I don't have an ethernet cable connected. I plan to run a cable eventually but wanted to run it on wifi until I get around to that.

MichaelDvP, 2022-11-28:
Reason code 15 is 4way handshake timeout (v3.5.0b10 shows this as text). 
Maybe the AP does not answer to wrong password (some AP do that as DDOS protection). Do you have different passwords to the APs with same SSID?
Depending what APs used, it could help to reboot the APs.

chris-0xcd, 2022-11-28:
Thank you for your help! Turns out the 2.4 GHz interface on the closest AP had an old password set. *D'oh!*  
I didn't even realize the passwords for the two interfaces are controlled separately. Would never have checked that without you asking if all APs had the same password. Looks like all my other devices work on 5GHz so I never even noticed.  

It works perfectly now, thanks again!

closed 2023-02-19 by proddy