emsesp/EMS-ESP32 #783: Web Interface and documentation wrong description of RX/TX pins

van-shorty, 2022-11-30:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/emsesp/EMS-ESP32/issues) (both open and closed)*
* *searched the [documentation help section](https://emsesp.github.io/docs)*

*Completing this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api/system*

**Additional context**
*Add any other context about the problem here.*

Hello together,

to day i setup the EMS-ESP and had over an hour to get this working. The problem was, that my RX and TX pins were twistet on the ESP32 and thats because the description of the pin assignement

[image]

in the Webinterface (see picture) it looks like the "GPIO TX" is the pin where the TX is on the ESP32, BUT its not. I had to configure there the pin where the TX is coming from the EMS interface board.

as example:

ESP32 AZDelivery:

[image]

TX pin is 17 and RX is 16 --> my configuration wich is working --> (see picture from settings) TX is 16 and RX is 17.

The documentation says:

_Rx GPIO - Which pin the Rx is assigned to. It can be any pin.
Tx GPIO - Which pin the Tx is assigned to. It can be any pin._

here its not clear thats the meaning of "pin" is the pin from the EMS interface!

For me it should be the PIN on the ESP32 which i assign, not something else i connect to it.

Thanks for the great work, iam realy happy now.

MichaelDvP, 2022-11-30:
Read: https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/tutorials/io_mux.html
and: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#uart-api-setting-communication-pins

BTW: v3.4.4 and before uses uart2, v3.5.0 uses uart1 (for esp32, C3, S2, S3)

MichaelDvP, 2022-12-01:
>  it looks like the "GPIO TX" is the pin where the TX is on the ESP32, BUT its not. I had to configure there the pin where the TX is coming from the EMS interface board.

Yes, the gateway is DCE, esp is DTE, you have to connect TX(esp)-TX(gateway) and RX-RX. TX on esp is sending data, TX on gateway is the input for sending data to the ems-bus.
(A crossed (nullmodem) connection is only used for DTE-DTE connections)

van-shorty, 2022-12-01:
> Yes, the gateway is DCE, esp is DTE, you have to connect TX(esp)-TX(gateway) and RX-RX. TX on esp is sending data, TX on gateway is the input for sending data to the ems-bus. (A crossed (nullmodem) [...]

ahhhh, that explains my issue, but how to know :D

proddy, 2022-12-01:
@MichaelDvP @bbqkees do we need to update the doc for EMS-ESP?

bbqkees, 2022-12-02:
We can close this one. I updated the EMS-ESP wiki and the BBQKees wiki to clarify the proper pinout.

van-shorty, 2022-12-02:
Perfect 👍

closed 2023-02-19 by proddy