emsesp/EMS-ESP32 #784: Thermostat data issue when tap water is active

PopaSchKing, 2022-12-01:
**Bug description**

Hi everyone,

I'm running into an issue where data from the thermostat (Hc1) is completely corrupted when tap water is active (boiler starts when tap water is pulled).
Date/Time, Current room temp., Current room target temp., Hc1 summer mode, ... change as soon as tap water is pulled and goo back to normal values afterwards.

This happens without any human interaction on the thermostat/boiler/Home Assistant/EMS-ESP.
Tx write is disabled, 100% quality on Rx/Tx/MQTT, 0 fail on both Rx/Tx/MQTT (would mean it's not a communication issue with the EMS-ESP ?).

EMS-ESP v3.4.4
ESP32 / BBQKees interface board
Boiler: Condens 5000i/Greenstar 8000
Thermostat: RC200/CW100

I'm adding 2 screenshots:
- First one is a graph from Grafana (Home Assistant --> InfluxDB --> Grafana).
  It shows the current room temperature, temperature target and if heating is on/off.
- Second (and third) is a screenshot of various Home Assistant data (boiler and thermostat from EMS-ESP).
  You can see that Date/Time changes when tap water is active, and the graph is a zoom on temp. and temp. target that changes at the same time.

I'm thinking about a potential power issue on the EMS bus which would corrupt the data..
Tested with a beefier 5V adapter for the ESP32, same behavior.

This is an issue opened following a discussion on Discord, many thanks for the help !

---

**Steps to reproduce**
Have everything running normally.
Pull water from a tap.
Issue hap
pens.

---

**Expected behavior**
No change of behavior / no data corruption when tap water is being pulled.

---

**Screenshots**

Temperature, Target and Heating on/off
[image]

HA data (Date/Time is the strangest thing to me)
[image]

Temps again but zoomed in on the same time period as the previous HA data (current room temp and target)
[image]

---

**Device information**
```
{
  "System Status": {
    "version": "3.4.4",
    "uptime": "000+01:33:14.923",
    "freemem": 128,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -69,
    "IPv4 address": "192.168.0.18/255.255.255.0",
    "IPv4 gateway": "192.168.0.1",
    "IPv4 nameserver": "9.9.9.9",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": true,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 2034,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {},
  "API Status": {
    "API calls": 1,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 6277,
    "bus reads (tx)": 1230,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "CUSTOM",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 0,
    "pbutton gpio": 0,
    "led gpio": 0,
    "hide led": true,
    "notoken api": false,
    "readonly mode": true,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": false
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 5000i/Greenstar 8000",
      "device id": "0x08",
      "product id": 195,
      "version": "05.04",
      "entities": 60,
      "handlers received": "0xBF 0xC2 0x15 0xD1 0xE3 0xE4 0xE5 0xE9",
      "handlers fetched": "0x14 0xE6 0xEA",
      "handlers pending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0xD7 0xD8 0x02D6 0x2E 0xF7"
    },
    {
      "type": "Thermostat",
      "name": "RC200/CW100",
      "device id": "0x18",
      "product id": 157,
      "version": "41.05",
      "entities": 40,
      "handlers received": "0x06 0xA2 0x031D",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02F5 0x023A",
      "handlers pending": "0xA3 0x12 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x02CE 0x02D0 0x02D2 0x031B 0x",
      "handlers ignored": "0xE7 0x02E0 0x02EA 0xBF"
    },
    {
      "type": "Controller",
      "name": "8700i",
      "device id": "0x09",
      "product id": 229,
      "version": "36.05",
      "entities": 0
    }
  ]
}
```

PopaSchKing, 2022-12-02:
Hello, thanks for your reply!

I attached the system log, the activation of tap water happens on line 24 (MQTT publication).
[ems-esp-log.txt](https://github.com/emsesp/EMS-ESP32/files/10144740/ems-esp-log.txt)

I'm also adding captures of HA regarding temperature (target and current). 
There are 2 mentions of the hc1 summer mode on lines 59 and 60.

`001+08:42:50.124 DEBUG 268186: [mqtt] Removing HA config for thermostat_hc1_summermode`
`001+08:42:50.225 DEBUG 268187: [mqtt] Publishing topic homeassistant/sensor/ems-esp/thermostat_hc1_summermode/config (#37928, retain=1, retry=1, size=0, pid=1)`

Regarding the other entities affected, I'd say it's only thermostat values from what I analyzed.
Of course, given that the thermostat starts/stops the boiler, sometimes if the current temperature jumps, heating starts/stops, ..

Here's a visual of what happened in the log :

[image]

[image]

Let me know if you need more data.
Cheers

MichaelDvP, 2022-12-03:
The summermode issue (switching to operationmode) is known and already fixed in v3.5. (see #747). 
The log is to short to see what happens with temperatures, there is only a single telegram 0x02A5 from thermostat, showing  18.0°C (00 B4) roomtemp and 17.0°C as setpoint (22). Datetime telegram )0x06) is not received in this log.
Please do another log with more than 1 min before and after the issue occures.

PopaSchKing, 2022-12-03:
Hi,

My bad for the previous log, here's a longer one with two water tap activation in the span of around 5 min.

[log_5min_2tap.txt](https://github.com/emsesp/EMS-ESP32/files/10146233/log_5min_2tap.txt)

[image]

MichaelDvP, 2022-12-05:
I can't see anything wrong in the log. Thermostat telegrams coming in. Datetime every minute with data only changing in minute, selected roomtemp stays at 17°C, measured roomtemp goes down from 18 to 17.9 °C and stay. 
The log does not show the mqtt payload, but the payload length and i can see that the length only changes 2 bytes as roomtemp goes from 18->17.9. Also when summermode is deleted and operationmode is published the payload lenght changes.
I've made a actual v3.5 dev with addition to sho the mqtt-payload in log. Please update to this version and log again. (The summermode/operationmode issue is fixed in v3.5dev). Download bin file from here: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

BTW: I don't know what the colors in the datetime bar should express. I think this should show a change, but what change?

PopaSchKing, 2022-12-07:
Hello!

Many thanks for your inputs, the logs indeed didn't show anything wrong with the EMS data.
It's only on Grafana (and thus Home Assistant + MQTT) that I witnessed the issues.

**After testing the new firmware, I'm glad to say that v3.5dev seems to have solved the issues !**

[image]
[image]

There's no longer a "cut" in the datetime data whenever tapwater is activated.
I tested with heater on, took a shower (heater stopped which is normal) and heater started again.
Same goes for the target/current room temperature, the issues met initially (drop to 15°C, curruption of current temp) are gone.

I'll keep monitoring everything tonight and tomorrow and will close the issue should the issues not reappear.

Answer to your BTW:
In my case, since I'm MQTTing every 60s, each color represents a datetime of 1min (eg 08:31:XX in blue, then 08:32:XX+1 in green, ...).
So when the issue occured, you had this weird datetime that would appear in a 1min "frame" and split it.

[image]

Once again thanks for your time !

PopaSchKing, 2022-12-08:
The problem did not appear under intense testing while dishwashing / heating.

[image]

[image]

No problem in the datetime data, and no erratic values in the temperatures (target/current).

Marking the issue as solved in v3.5dev.
Thanks a lot @MichaelDvP !

closed 2022-12-08 by PopaSchKing