emsesp/EMS-ESP32 #786: RC30 Data missing with SW Version 3.4.4 / 3.5b

Cretanios, 2022-12-04:
Coming from a premium V1.6 adapter with ESP8266 V2.2.2 Software I just siccessfully did an update  to the ESP32 with Software Version V3.4.4 and then V3.5b.

Using a GB142 with RC10, MM10, WM10 and UBA3 3.5 I checked the data submitted by the adapter and found there are a lot of parameters missing for the RC30. Starting with the measured room temperature.

It is a hydraulic with 3way valve for heating and DHW, the pump is feeding DHW and a hydraulic valve. There are two heating circuits after the hydraulic valve - HC1 is direct and HC2 is with mixing valve - eachnof which with separate pumps. The RC30 is dedicated to HC2.

List of parameters for V2.2.2:
[image]

List of parameters for V3.5b:
[image]

proddy, 2022-12-04:
Could you post the support details (https://emsesp.github.io/docs/#/Support?id=support) and also look at the EMS status or any Rx/Tx issues, and open the log (web or console) and see if you see any warnings.

Cretanios, 2022-12-11:
EMS status is:

Connection Status: Connected
Received Telegrams: 228.193
Rx line quality: 100%
Sent telegrams: 62.224
Ty line quality: 100%

proddy, 2022-12-11:
please attach the support info

Cretanios, 2022-12-17:
{
  "System Info": {
    "version": "3.5.0b10",
    "platform": "ESP32",
    "uptime": "000+00:01:24.907",
    "freemem": 118,
    "maxalloc": 69,
    "free_app": 73,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Info": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -44,
    "IPv4 address": "192.168.178.186/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 1,
    "temperature sensor reads": 16,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 244,
    "bus reads (tx)": 49,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "BK13/BK15/Smartline/GB1x2",
      "device id": "0x08",
      "product id": 64,
      "version": "03.05",
      "entities": 67,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC30",
      "device id": "0x10",
      "product id": 67,
      "version": "01.12",
      "entities": 18,
      "handlers received": "0x06 0xA2 0x12 0x3E 0x3D 0x48 0x47 0x38 0x39",
      "handlers fetched": "0xA5 0x37",
      "handlers pending": "0xA3 0x13 0x3F 0x42 0x49 0x4C 0x52 0x51 0x53 0x56 0x5C 0x5B 0x5D 0x60",
      "handlers ignored": "0xAC 0x35 0x1A 0x9D"
    },
    {
      "type": "Mixer",
      "name": "MM10",
      "device id": "0x21",
      "product id": 69,
      "version": "02.00",
      "entities": 6,
      "handlers received": "0xAB",
      "handlers fetched": "0xAA",
      "handlers pending": "0xAC"
    },
    {
      "type": "Switch",
      "name": "WM10",
      "device id": "0x11",
      "product id": 71,
      "version": "02.00",
      "entities": 2,
      "handlers received": "0x9C 0x1E",
      "handlers pending": "0x9D"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20",
      "device id": "0x09",
      "product id": 68,
      "version": "02.03",
      "entities": 0
    }
  ]
}

proddy, 2022-12-18:
can you give an example of a single entity value that is missing between v2.2 and v3.5, which we can use to trace down the root cause? It's easier than wading through screen shots.

MichaelDvP, 2022-12-21:
This is a strange issue, seems that all values from thermostat heatingcircuits are missing. 
The system info is with uptime 1:24 minutes in the startup and maybe not all values fetched, but thermostat says the handlers for hc1 and hc2 monitor and settings (0x3D, 0x3E, 0x47, 0x48) are received, but only 18 entities is without the hc values.
There are only 2 possible reasons i can imaging:
- all hc entities are set to disable in customizations
- both hcs are disabled in thermostat.

peter--s, 2023-02-10:
I found this issue searching for the answer why my new EMS-ESP on v3.5 does not display any data of my RC30_N's HK1MonitorMessage (0x3E) much like in the screenshot above.
@MichaelDvP: in my case hc1 is enabled in the RC30 ant the RC30 is assigned to hc1 (moreover, it seems to function correctly). No customizations either.

**Is the telegram by chance filtered out based on length?**
My RC30 is sending 13B of data in HK1MonitorMessage.

While https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#uebersicht states that RC30 sends 2B less than RC35
a comment at the end of https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#hk1monitormessage suggests only the last byte should be missing.

watch raw 0x3e:
`000+01:18:06.212 N 32: [telegram] Rx: 10 00 3E 00 04 02 2B 00 D7 00 00 43 43 43 00 00 64 11 D8
`

watch on 0x3e:
`000+01:19:17.090 N 33: [emsesp] thermostat(0x10) -B-> All(0x00), RC35Monitor(0x3E), data: 04 02 2B 00 D7 00 00 43 43 43 00 00 64 11
`

show devices:
```
These EMS devices are currently active:

boiler: BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:02.03)
 Received telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x34
 Fetched telegram type IDs: 0x14 0x16 0x33
 Pending telegram type IDs: 0xBF 0xC2 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA
 Ignored telegram type IDs:

thermostat: RC30 (DeviceID:0x10, ProductID:67, Version:02.02)
 Received telegram type IDs: 0x06 0xA2 0x12 0x3E 0x3D 0x48 0x47 0x38 0x39
 Fetched telegram type IDs: 0xA5 0x37
 Pending telegram type IDs: 0xA3 0x13 0x3F 0x42 0x49 0x4C 0x52 0x51 0x53 0x56 0x5C 0x5B 0x5D 0x60
 Ignored telegram type IDs: 0x1A 0x35

controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:01.05)

```
show:
```
EMS-ESP version 3.5.0

Kessel: BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:02.03) (61)
  Heizen aktiv: aus
  Warmwasser aktiv: aus
  Sollwert Vorlauftemperatur: 39 °C
  Heizungspumpe 1 Modulation: 63 %
  aktuelle Vorlauftemperatur: 36.4 °C
  Rücklauftemperatur: 36.1 °C
  Systemdruck: 1.2 bar
  Kesseltemperatur: 37.1 °C
  Gas: aus
  Gas Stufe 2: aus
  Flammenstrom: 0.0 µA
  Heizungspumpe: an
  Gebläse: aus
  Zündung: aus
  Heizen aktiviert: an
  Heizungstemperatur: 90 °C
  Kesselpumpen Maximalleistung: 100 %
  Kesselpumpen Minmalleistung: 50 %
  Kesselpumpen Modus: deltaP-2
  Pumpennachlaufzeit: 10
  Antipendelzeit: 10
  minimale Brennerleistung: 0 %
  maximale Brennerleistung: 99 %
  Einschaltdifferenz: -6 °C
  Ausschaltdifferenz: 6 °C
  Sollwert Brennerleistung: 99 %
  Brennerleistung: 0 %
  Brenner Starts: 98958
  Brenner Laufzeit: 1954 Tage 7 Stunden 47 Minuten
  Heizung Laufzeit: 1609 Tage 16 Stunden 18 Minuten
  Brenner Starts Heizung: 61078
  Anlagen-Gesamtlaufzeit: 7245 Tage 3 Stunden 48 Minuten
  Letzter Fehler: 3F(273) 08.02.2023 09:22 (0 min)
  Statusmeldung: 0Y
  Statusmeldungsnummer: 204
  Wartungsmeldung: H00
  WW Solltemperatur: 60 °C
  WW gewählte Temperatur: 60 °C
  WW Typ: Speicher
  WW Komfort: Heiß
  WW Vorlauftemperaturanhebung: 40 °C
  WW Zirkulationspumpe vorhanden: an
  WW Speicher-Ladungstyp: 3-Wege Ventil
  WW Einschalttemperaturdifferenz: -5 °C
  WW Ausschalttemperaturdifferenz: 0 °C
  WW Desinfektionstemperatur: 70 °C
  WW Zirkulationspumpen-Modus: 2x3min
  WW Zirkulation aktiv: aus
  WW aktuelle interne Temperatur: 55.3 °C
  WW aktuelle externe Temperatur: 55.3 °C
  WW externer Speichertemperatur: 55.3 °C
  WW aktiviert: an
  WW Einmalladung: aus
  WW Desinfizieren: aus
  WW Laden: aus
  WW Nachladen: aus
  WW Temperatur ok: an
  WW aktiv: aus
  WW 3-Wegeventil aktiv: aus
  WW Anzahl Starts: 37880
  WW aktive Zeit: 344 Tage 15 Stunden 29 Minuten

Thermostat: RC30 (DeviceID:0x10, ProductID:67, Version:02.02) (19)
  Letzter Fehler: A1(821) 10.02.2023 06:51 (0 min)
  Datum/Zeit: 10.02.2023 06:56
  Anzeige: Interne Temperatur
  Sprache: Deutsch
  Uhrkorrektur: 0 Sekunden
  Korrektur interner Temperatur: 0.0 °C
  min. Aussentemperatur: -16 °C
  Gebäudetyp: Mittel
  WW Modus: auto
  WW Zirkulationspumpen-Modus: auto
  WW Programmmodus: Standardprog.
  WW Desinfizieren: aus
  WW Desinfektionstag: Di
  WW Desinfektionsstunde: 6
  WW Maximale Temperatur: 60 °C
  WW Programm Schaltzeit: 00 mo 00:00 off
  WW Zirculationsprogramm Schaltzeit: 00 mo 06:00 on
  WW Feiertage: 01.01.2000-01.01.2000
  WW Urlaubstage: 09.11.2017-12.01.2018

Kontrollmodul: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:01.05) (0)
```

MichaelDvP, 2023-02-10:
Ah, that's it, byte 14 is used to detect if the heating circuit is active, so this gives always an inactive circuit. Please check the other circuits with `read 10 48`, `read 10 52`, `read 10 5C`  I'll check on my RC35, to find a clear sign for active/inactive circuit.

peter--s, 2023-02-10:
Thanks for your swift answer!

I suggest checking for the first byte>0 in 0x3D, 0x47, 0x51, 0x5B.
After checking my setup to answer your previous questions, it seemed obvious that this telegram conveys all hc related settings - and voilá:
0x3D starts with 4 while 0x47 starts with 0, both reflecting my thermostat's settings (RC30 supports two hc only).
(BTW: are you aware who maintains/how to correct/add to https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#hk1betriebsart? 0 appears to stand for hc disabled in thermostat.)

But if you prefer/need to stick with HK1MonitorMessage instead of HK1Betriebsart, B0>0 (or OR(b2,b7) or alike evaluation of the first byte) seems equivalent:
```
ems-esp:$ read 10 48
000+06:33:14.187 I 2: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC35Monitor(0x48), data: 00 00 00 00 00 00 00 05 05 14 00 00 00 00
ems-esp:$ read 10 52
000+06:33:27.909 I 3: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC35Monitor(0x52), data: <empty>
ems-esp:$ read 10 5C
000+06:33:39.285 I 4: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC35Monitor(0x5C), data: <empty>
```

MichaelDvP, 2023-02-12:
Yes, checking offset 0 should work, i've tested my RC35 and the active hc always sets a bit in offset 0 if hc is active.
I've made a dev-build with this fix, try: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest