emsesp/EMS-ESP32 #787: Dutch translation error: minuuten instead of minuten

joostvdl, 2022-12-04:
**Bug description**
In the GUI the term _Minutes_ is translated in _Minuuten_. This should be _Minuten_

**Steps to reproduce**
Just open the GUI in the dutch language

**Expected behavior**
Repair translation

**Screenshots**
n/a

**Device information**
{
  "System Info": {
    "version": "3.5.0b10",
    "platform": "ESP32",
    "uptime": "000+00:09:12.891",
    "freemem": 112,
    "maxalloc": 61,
    "free_app": 73,
    "reset reason": "Software reset CPU / Software reset CPU"
  },

**Additional context**
n/a

proddy, 2022-12-04:
thanks! Will be fixed in the next PR

joostvdl, 2022-12-04:
Addition: Same goes for _uuren_ instead of _uren_

proddy, 2022-12-04:
fixed in b11

closed 2022-12-04 by proddy