emsesp/EMS-ESP32 #789: Increase counter precision (analogsensors)

NFis, 2022-12-04:
I wanted to use the counter functionality for my gas meter. Every 0.01 m³ a pulse comes from my hall sensor (very good signal quality, hardware debounced). My gas meter has about 13000 m³ on its mechanical counter.  So I created an analogsensor with counter functionlity, Start Value 13xxxx and Factor 0.01: 
[image]
But after a while the counter in ems-esp differs from the real counter. EMS-ESP sensors internally use float values, so with its counter adding a small value (0.01) to a relative big one leads to deviations. 
My first workarund: configure the input as digital input and count the pulses externally in Home Assistant. 
Second workaround is a software change: use double in the Sensor class in analogsensors.cpp/h instead if float. In helpers.cpp/h the according render_value function has also to be adapted.
My version: 3.4.4

MichaelDvP, 2022-12-05:
Please test with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
I've changed to double, but have no impulse source to count and test.

NFis, 2022-12-05:
Ok, thank you. I installed the sw and watch the counter tomorrow. This night heating is off.

NFis, 2022-12-06:
The counter runs perfectly after 11 hours of heating now.

closed 2022-12-09 by proddy