emsesp/EMS-ESP32 #792: Nefit enviline 9KW has many outliers in energy report

Avd888, 2022-12-05:
**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*
I'm using EMS-ESP version 3.4.1 with an BBQKees E32 interface
boiler: Boiler: Buderus Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx

**Steps to reproduce**
Log energy values via MQTT with home assistant, overtime the graph shows many outliers, unfortunately to many to filter out. 
Affected values:
```
total energy consumption 
total energy consumption compressor 
energy consumption compressor heating 
energy consumption compressor dhw 
energy consumption compressor cooling 
energy consumption compressor pool 
total auxiliary electrical heater energy consumption 
auxiliary electrical heater energy consumption heating 
auxiliary electrical heater energy consumption dhw 
auxiliary electrical heater energy consumption pool 
total energy supplied 
total energy supplied heating 
total energy warm supplied dhw 
total energy supplied cooling 
```

**Expected behavior**
I expect to have less outliers and if there is an outlier it should better be an 'unavailable' instead of 0 or other low value

**Screenshots**
[image]

**Device information**
```
{
  "System Status": {
    "version": "3.4.1",
    "uptime": "050+04:05:56.194",
    "freemem": 109,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "94:3C:C6:39:15:B7",
    "IPv4 address": "192.168.0.29/255.255.255.0",
    "IPv4 gateway": "192.168.0.1",
    "IPv4 nameserver": "192.168.0.11",
    "phy type": 1,
    "eth power": 16,
    "eth phy addr": 1,
    "eth clock mode": 0,
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 1367515,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Status": {
    "API calls": 1,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 13812979,
    "bus reads (tx)": 566689,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 1626,
    "bus reads failed": 154,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 5,
    "tx gpio": 17,
    "dallas gpio": 4,
    "pbutton gpio": 33,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx",
      "device id": "0x08",
      "product id": 172,
      "version": "01.16",
      "entities": 108,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0xEF 0xF7 0x0491 0x0484 0x02CC 0x04A2 0xF6 0xF9 0xC6 0x2D 0xBB 0x04 0x0488 0xC7 0x0497 0x0499 0x04B4 0x0485 0x0486 0x048B 0x049F 0x04A0 0x04A3 0x0492 0x0493 0x049C 0x049D 0x05 0x01 0x03 0x0489 0x04A4 0xEB 0xFD02"
    },
    {
      "type": "Thermostat",
      "name": "RC100/Moduline 1000/1010",
      "device id": "0x38",
      "product id": 165,
      "version": "40.04",
      "entities": 0,
      "handlers received": "0x02A5 0x02B9",
      "handlers pending": "0x02AF 0x029B 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02F5 0x031B 0x031D 0x031E 0x023",
      "handlers ignored": "0x01 0x042B 0xF7 0xBF"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "13.07",
      "entities": 39,
      "handlers received": "0x06 0x02BA 0x02B0 0x029C 0x0472 0x02BB 0x02B1 0x029D 0x0473 0x02BC 0x02B2 0x029E 0x0474 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x02A6 0x02A7 0x02A8 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x031B 0x031E",
      "handlers ignored": "0x0269 0x026A 0x026B 0x026C 0x026D 0xF7 0xF6 0x01 0xEF 0xBF 0xC0 0xC1 0x0291 0x0292 0x0293 0x0294 0x02C3 0x02C4 0x02C5 0x02C6 0x02FF 0x0309 0x0449 0x044A 0x044B 0x044C 0xF9"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.08",
      "entities": 0
    }
  ]
}
```

**Additional context**
*Add any other context about the problem here.*

MichaelDvP, 2022-12-05:
Please update to actual dev buld v3.5.0b11 and check if the issue still exists.

Avd888, 2022-12-06:
I updated it to 3.5.0b11 and will monitor it for a while