emsesp/EMS-ESP32 #796: Add /metrics endpoint for prometheus

thegitarist, 2022-12-07:
Hello,

would be great to have an prometheus endpoint exposed (at default /metrics path) to easily add metrics to prometheus / grafana stack.

THNX & BR

proddy, 2022-12-09:
no idea how to do this. is this something you can implement? or come up with a spec and a design on how it should be implemented?

thegitarist, 2022-12-09:
its basically an additional endpoint with location /metrics and header Content-Type: text/plain; charset=utf-8 where the metrics, that are already accessible via mqtt are presented.

Currently I am running mosquitto and an mqtt-exporter to generate this /metrics endpoint, but the desired implementation would reduce complexity in a vanilla prometheus grafana architecture.

The metrics are then presented like this:
```
# HELP ems_currtemp metric generated from MQTT message.
# TYPE ems_currtemp gauge
ems_currtemp{topic="ems-esp_thermostat_data_hc1"} 18.6
```

If u create a basic skeleton with at least one metric, I will try my best to implement the rest

proddy, 2022-12-09:
can't you create a mapping in prometheus to read from the API? e.g. http://[ems-esp.local/api/thermostat already exposes the data in JSON

thegitarist, 2022-12-09:
good point. I can write an adapter that reads the api and exposes the /metrics. But the best part is no part :-)

this feature request has, of cause, no priority. maybe I will dig into the code and try it by myself

poelstra, 2023-01-29:
@thegitarist I needed the same thing and gave it a go: https://github.com/poelstra/ems-esp-exporter

Currently, it only polls the `boiler` device (that's the only one I have/need), but I'm happy to add generic support for the other devices too. Would appreciate it if someone could [give a hand figuring out](https://github.com/poelstra/ems-esp-exporter/issues/1) how to poll all devices.

thegitarist, 2023-01-30:
thanks and nice implementation! I'm familiar with js, but ts with generics looks interesting.

due to memory constraints on my rpi, I'll keep my current setup (mosquitto & mqtt-exporter = 16MB RAM)