emsesp/EMS-ESP32 #802: support silent mode option in Nefit Bosch heatpump

derSchreifritz, 2022-12-12:
Some settings available on the heatpump are not available through the bbqkees (yet) or I can't see or find them.

One of them is the silent mode option.

I'd like to see the option "Minimum outside temperature for silent mode" or other recognizable name in homeassistant
I did set it from 6 to 8, then to 9, 7 and back to 6, and exported the log that I saw during that in system at EMS-ESP. See attached txt file and foto.
[log(3).txt](https://github.com/emsesp/EMS-ESP32/files/10212029/log.3.txt)
[image]
The panel model on the heatpump is called HPC 410, and it says EMS 2 on the instruction book.

proddy, 2022-12-12:
thanks, we'll take a look.

derSchreifritz, 2022-12-13:
thanks, let me know if I can provide better information. or help in other ways.
At the moment, I'm even more interested in another setting I cant find via homeassistant, can I post that here as well?
It's the "auxiliary heater only" binary switch. I changed it twice from yes to no and back. A foto (this time in English) and log are below.
[log(4).txt](https://github.com/emsesp/EMS-ESP32/files/10215762/log.4.txt)
[image]
I have no idea how you can find anything in there, thanks in advance for any help!

MichaelDvP, 2022-12-13:
Thanks for the logs.
> I did set it from 6 to 8, then to 9, 7 and back to 6,

```
001+00:25:04.154 TRACE 33470: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x0484), data: 07 (offset 11)
001+00:25:13.616 TRACE 33559: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x0484), data: 09 (offset 11)
001+00:25:19.679 TRACE 33578: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x0484), data: 08 (offset 11)
001+00:25:27.979 TRACE 33640: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x0484), data: 06 (offset 11)
```
The complete telegram is very long, i thnk it is not only silent mode related.
Boiler(0x08) -> All(0x00), ?(0x0484), data: 00 00 14 28 0D 50 00 00 00 02 02 07 28 01 00 02 05 19 0A 0A 03 0D 07 00 0A
Boiler(0x08) -> All(0x00), ?(0x0484), data: 01 90 00 F6 28 14 64 00 00 E1 00 1E 00 1E 01 64 01 64 54 20 00 00 (offset 25)

> It's the "auxiliary heater only" binary switch. I changed it twice from yes to no and back

```
001+11:27:15.828 TRACE 190691: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x0491), data: 00 (offset 1)
001+11:27:20.340 TRACE 190709: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x0491), data: 01 (offset 1)
```
The complete telegram is:
`Boiler(0x08) -> All(0x00), ?(0x0491), data: 03 01 00 00 00 02 64 00 00 14 01 2C 00 0A 00 1E 00 1E 00 00 1E 0A 1E 05 05`
in offset 10 there is `01 2C`, could be the 300 for additional heater on delay.

Is your heatpump product-id 172 or 173?

I'll prepare a test build to check. The full implementation needs translations to nl, se, pl, no, fr.

MichaelDvP, 2022-12-13:
Testbuild: https://github.com/MichaelDvP/EMS-ESP32/releases

derSchreifritz, 2022-12-13:
> Is your heatpump product-id 172 or 173?
172

proddy, 2022-12-28:
@derSchreifritz working now?

derSchreifritz, 2022-12-28:
min outside temp for silent mode is working here, yes, thanks!

closed 2022-12-28 by proddy