emsesp/EMS-ESP32 #803: Support for Hybrid specific settings and statistics in Nefit Bosch Heatpumps

tvdhorst, 2022-12-14:
Some settings and statistics available in Nefit Bosch heatpumps do not seem to be available yet in EMS-ESP on BBQKees' hardware. Some of these are specific for Hybrid installations.

I played around with these settings and forced some changes, while monitoring them manually on the heatpump and logging on EMS-ESP. These are the setting and changes in their values:

Status aux heater with mix
- Add. heat source
  - ... Off
  - 09:36 On
  - 09:44 On
  - 09:51 On
  - 10:01 Off
- Mixing valve
  - ... 0%
  - 09:36 1%
  - 09:44 2%
  - 09:51 3%
  - 10:01 0%

Auxiliary heater with mixer
- Outs. temp par. mode
  - 09:07 0 degrees centigrade
  - 09:10 -1 degrees centigrade
  - 09:25 2 degrees centigrade
  - 09:37 1 degrees centigrade
  - 09:44 -2 degrees centigrade
  - 10:00 -5 degrees centigrade

Additional heater status
- ... Off
- 09:36 Heating
- 10:01 Off

Energy/ price ratio
- 09:07 3.5
- 09:10 4.0
- 09:25 4.3
- 09:36 4.7
- 09:44 4.9
- 10:00 1.7

Low-noise operation (als mentioned in #802)
- Min. outside temp.
  - 09:06 7 degrees centigrade
  - 09:09 5 degrees centigrade
  - 09:25 4 degrees centigrade
  - 09:37 3 degrees centigrade
  - 09:43 4 degrees centigrade

[log-ems-esp.txt](https://github.com/emsesp/EMS-ESP32/files/10226195/log-ems-esp.txt)

Would it be possible to add these? Happy to help by providing additional info!

MichaelDvP, 2022-12-14:
Please add your system info (from help page). We have hybrid settings with telegram 0xBB in thermostat, but for your thermostat it is shown as unknow. Please also mention if these are settings and the change is initiated by user (e.g. single value thermostats-W->boiler), or sensor-values that are published somewhere in a telegram.

tvdhorst, 2022-12-14:
Here's my system info: [emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10227579/emsesp_info.txt)
I edited the startpost to distinguish between the sensor values and settings.

MichaelDvP, 2022-12-14:
I've added the hybrid settings from #459 to this thermostat and from #802 to boiler, check if it works. (Testbuild: https://github.com/MichaelDvP/EMS-ESP32/releases)
The mixing valve seems to be the last byte in telegram 0x488. the other sensor values are difficult to find. For a 0/1 change we need more exact timing of the change, it's to much to check.

Your free memory is really low, was this the weblog buffer set to 100 (not needed if you leave the browser-page open) or do the heatpump values consume so much heap?

proddy, 2022-12-14:
> Your free memory is really low, was this the weblog buffer set to 100 (not needed if you leave the browser-page open) or do the heatpump values consume so much heap?

shall we remove 100, so it's only 25, 50 and 75?

tvdhorst, 2022-12-14:
> I've added the hybrid settings from #459 to this thermostat and from #802 to boiler, check if it works. (Testbuild: https://github.com/MichaelDvP/EMS-ESP32/releases) The mixing valve seems to be the last byte in telegram 0x488. 
Cool, will try this.

> the other sensor values are difficult to find. For a 0/1 change we need more exact timing of the change, it's too much to check.
These would be Add. heat source & Additional heater status? I would suppose these change at the same time as the change from mixing valve from 0 to anything above. Does that help?

> Your free memory is really low, was this the weblog buffer set to 100 (not needed if you leave the browser-page open) or do the heatpump values consume so much heap?
Yes I set it to 100 but later realised this didn't seem to be needed

tvdhorst, 2022-12-14:
The test build gives me 1 extra entity (energy cost ratio), but none of the others yet

derSchreifritz, 2022-12-14:
so also not the values from #802 ?
Auxiliary heater only- yes or no
and low noise mode min outside temp.?
I'd love to test, but don't know how to upgrade to this build from 3.4.4

tvdhorst, 2022-12-14:
No not the values from #802.

We seem to have the same heatpump, so if you'd also like to test: in the web gui, note the amount of entities for boiler and thermostat, then go to settings > Upload/Download and upload the above mentioned test build :)

derSchreifritz, 2022-12-14:
yes, I'm hackbard at tweakers. When I try upload the test build, I get "Upload failed (Internal Server Error)" .
I tried yesterday the flasher but couldn't get it work..

MichaelDvP, 2022-12-14:
@tvdhorst please make > a minute log, to see the telegrams (0xBB from thermostat, 0x488, 0x484 0x491 from boiler). The boiler should show a few more settings, add `ems-esp.local/api/boiler`

@derSchreifritz  If update via webpage gives repeated error you need to update once with usb and https://github.com/emsesp/EMS-ESP-Flasher/releases

derSchreifritz, 2022-12-14:
thanks, I tried, but the flasher didn't run on ubuntu, and on my windows I couldnt find the COM port... With command line, is it just copy and paste whats on https://emsesp.github.io/docs/#/Uploading-firmware ?
Sorry I guess this is offtopic, but I'm a bit stuck there..

tvdhorst, 2022-12-14:
Here's the output from ems-esp.local/api/boiler: [entities_boiler.txt](https://github.com/emsesp/EMS-ESP32/files/10228603/entities_boiler.txt)

Minute later the EMS-bus disconnects (after I try to connect over telnet?), problem persists after a restart. Trying to figure it out...

tvdhorst, 2022-12-14:
@MichaelDvP, I'm not sure of this is exactly what you're asking for, but here's the log: [telegram_log.txt](https://github.com/emsesp/EMS-ESP32/files/10228976/telegram_log.txt)

MichaelDvP, 2022-12-14:
Yes, that's what i'm asking for. The telegrams are only broadcasted on change, we have to request them. 
If you are in telnet please try:
`read 8 488`, `read 8 491`, `read 8 484` and also `read 8 486`, `read 8 492` and `read 8 4A2`
After that the values should be shown. (Please post the output from the reads here).
I'll add a fetch to these telegrams.
Seems working now, what was wrong? I've flashed now with the github-bin and it's the same as compiled here and working.

@derSchreifritz for windows flasher check if the com driver is installed (CH340 or CP210x). For commandline flash you need the bin files for partitions, bootloader, boot_app in the same directory. For windows also the com-driver installed.

tvdhorst, 2022-12-14:
Already realised that indeed :) So I was just logging telegrams while forcing the additional heating. 11 new values showed up! See [entities.txt](https://github.com/emsesp/EMS-ESP32/files/10229288/entities.txt)

Still, here are the requested outputs:
[read_8.txt](https://github.com/emsesp/EMS-ESP32/files/10229257/read_8.txt)

And another watch on output, there still seem to be unknown values?
[logtelnet2.txt](https://github.com/emsesp/EMS-ESP32/files/10229258/logtelnet2.txt)

derSchreifritz, 2022-12-14:
> No not the values from #802.
> 
> We seem to have the same heatpump, so if you'd also like to test: in the web gui, note the amount of entities for boiler and thermostat, then go to settings > Upload/Download and upload the above mentioned test build :)

I finally managed to upgrade :)
when I first opened the interface, I saw 118/4/48 number of entities, with the old version it was 106/47/4. However, after uploading my settings and customization files, I see only 108/4/49 entities. First I guess I will have to deal with the new entitiy ID's, and will do more testing later. 
Also to @MichaelDvP ,Thanks for your work!

MichaelDvP, 2022-12-14:
I've updated the test build to fetch the telegrams. Now the entities should be there 1-2 minutes after emsesp starts.
> there still seem to be unknown values?

Sure, a lot of unknown telegrams and in the known telegrams only a few known data and lots of unknown data.

derSchreifritz, 2022-12-14:
something is odd now, I can't change the manual temperature from the interface, "Write command failed"
Is that just me?

tvdhorst, 2022-12-14:
Running the test build now! Not much time right will report back, entities showed up from the start. Thanks!

tvdhorst, 2022-12-15:
So far this seems to be working fine. Entities show up and I can change the setting from ems-esp.

I also noticed when checking values on the heatpump that the mixing valve can still be 0% while Add. heat source & Additional heater status are On or Heating. That surprised me, but it also means that we can’t look at the valve for indications on when these two other sensors change.

How could I help to get more info on these values? What would you need? I can now force the additional heater from ems-esp, so that could help me to provide info :)

MichaelDvP, 2022-12-15:
What's the difference of Add. heat source and heater status? seem both are switching parallel. 
There is a change in first byte of telegram 0x0488, seem to me as a bitfield. Starts wiith 89, then 8D, 8C, 8E, and back to 89 when mixingalve goes zero. Bit 2 could be the heater status. But 488 is not broadcasted periodical, uncommon for a status message.
The telegrams 0x49F, 0x4A0, 0x4A2 are broadcasted very often but with no change over the complete log.

The better you can say the exact time of a change the better could we find the telegram/position.

derSchreifritz, 2022-12-15:
I upgraded again to the latest build. On the ems or ha I don't see add. heat source status. Should I again switch things on the boiler itself, and send the system logs?  I don't know about telegrams..
min. outside temp for silent mode, auxiliary heater only from #802 are working :)
as well as energy cost ratio and aux. heater mixing valve.
There is a few other things I'm interested in and I can't find yet, can I just post them here with logs?

tvdhorst, 2022-12-15:
Add. heat source or add. heater status or not integrated yet @derSchreifritz 

@MichaelDvP I was also wondering about the difference between the two, they seem similar, but in the gui on the heatpump one switches between off and heating and the other between off and on. I'll go ahead and force the additional heater while logging ALL on ems-esp. The values change 'immediately' when additional heating is forced on, so we can use the timestamp of the change in 'auxiliary heater only' as a guide. Might not be before the weekend though.

The switch off add. heater setting would also be useful.

MichaelDvP, 2022-12-15:
> I switched from no to yes, back, again yes, took the foto, and back to no

I can see offset 2 in 0x491 changing to 0, 1, 0, 1, so we have inverse logic? Or we name it `additional heater` with on/off and normal logic.
 > not working here:
Additional heater on delay

This was a guess, i found a 300 (01 2C)  in the telegram and thought that it could be, Now your picture show value 190, but i can't find a `00 BE` in the data. I'll remove it.

tvdhorst, 2022-12-15:
I would suggest to either:
- use the same name as in the gui, to make it recognisable (that would include inverse logic), or
- name it something like 'Enable additional heater', because the setting doesn't switch it on, it enables use of it.

Which brings to mind that Bosch Nefit uses both 'additional' and 'auxiliary' interchangeably. The 'Auxilliary heater only' setting involves the same external heater as this setting.

derSchreifritz, 2022-12-16:
> > not working here:
> > Additional heater on delay
> 
> This was a guess, i found a 300 (01 2C) in the telegram and thought that it could be, Now your picture show value 190, but i can't find a `00 BE` in the data. I'll remove it.

shall we give it another try?
I just changed the 
Additional heater on delay from 190 to 220

also changed the on/off hysteresis from 40 to 145
on/off hysteresis is another feature I'd love to see.

Both values are in K*min

[log(6).txt](https://github.com/emsesp/EMS-ESP32/files/10246069/log.6.txt)
[image]

derSchreifritz, 2022-12-16:
> Would be helpfull if you can give me the parameter names in the languages the thermostat can show or from the manual. We have now EN, DE, FR, NL, NO, PL, SE
EN:
Additional heater on delay
On/off hysteresis heating
DE
Zuheizer verzögert ein
Ein/Aus-Hysterese Heizbetrieb
FR 
Chauff app tempo marche
Hystérésis Marche en mode chauffage
NL
Bijverw. vertraagd aan
Aan/uit-hysteresis in verw. bedrijf instellen
NO 
Tilleggsvarmer forsinket på
På/av-hysterese Oppvar.
PL
Opóźn. włączenie dogrz.
Histerez wł/wył Ogrzew.
SE
Tillskottfördröjning på
På/av-hystereses Husv.

derSchreifritz, 2022-12-16:
> > Both values are in K*min
> 
> The picture shows K/min, not K*min as the delay?

Yeah, for on/off hysteresis heating, the other one says "K*mi". The manual says K x min for both, K/min doesn't make sense to me anyway. So I would recommend K*min

tvdhorst, 2022-12-17:
Here's another log, where I:
- Around 09:53 switched 'additional heater only' to on
- Add. heat source & Additional heater status changed to 'on' and 'heating' respectively
- Around 09:54 switched 'additional heater only' to off
- Add. heat source switched to 'off' immediately
- Additional heater status switch to 'off' just a little later (5-10 seconds?)

[log-8.txt](https://github.com/emsesp/EMS-ESP32/files/10251349/log-8.txt)

@derSchreifritz where did you get the translations? I could look them up for these entities.

Hysteresis should indeed be K * m, not K/m.

derSchreifritz, 2022-12-17:
> 
> @derSchreifritz where did you get the translations? I could look them up for these entities.
> 
HPC410 manuals in EN, DE, FR, NL I found online, you want me to send them to you?/post here? PL, NO, SE I typed from the boiler screen, with some help of google translate to get the correct letters/special characters. Was a bit of work for me.

MichaelDvP, 2022-12-17:
Thans for the translations. I've added the translations so far. Please check the latest dev on my beta. 
Proddy likes to release v3.5 soon, if this is ok we can add it and continue with more values in the next beta.

@tvdhorst I can see the switches, but have not found a status-change in the broadcasted telegrams yet. It's not easy.

tvdhorst, 2022-12-17:
@MichaelDvP thanks for the effort, let me know if there’s anything I could provide to gain more insight.

@derSchreifritz could post the links here? Which translations are still missing @MichaelDvP?

MichaelDvP, 2022-12-17:
See current translations of new entities here: It's always Tag, mqtt-name, en, de, nl, se, pl, no, fr 
https://github.com/MichaelDvP/EMS-ESP32/blob/ab3b9f13b574c3ca20917d06f896e60245f6bcfa/src/locale_translations.h#L371-L378
In the next block (L388-L400) of hybrid heatpumps there are also some nl, se, no translations missing (""), empty translations fall back to "en".

If you like to help with translations you can also download the file `locale_translations.h`, edit, and send it here.

tvdhorst, 2022-12-17:
I’ll have a look at translations after the weekend.

Are logs via telnet of the telegrams more helpful in finding the entities?

I do think I understand the difference between the two now. Add. heat source is the actual heat request to the heat source, my gaspowered boiler immediately reflects the change in its value. Whereas the Additional heater status seems more of a summary of the current status according to all settings and logic of the heatpump combined.

MichaelDvP, 2022-12-17:
I've searched again, the only change i've found is (as before) telegram 0x488, offset 0, bit2. I've added as auxHeaterStatus. Please check if the values over time are ok.

Logging in telnet or web is the same. Telnet is more flexible with watch or read command to track single telegram if we know what we are looking for.

derSchreifritz, 2022-12-17:
> 
> @derSchreifritz could post the links here? 
DE https://www.manualslib.de/manual/751630/Bosch-Hpc-410.html?page=1#manual
EN https://www.manualslib.com/manual/2343751/Bosch-Hpc-410.html
FR https://www.manualslib.fr/manual/564464/Bosch-Hpc-410.html
NL https://nefit-nl-nl-b.boschtt-documents.com/download/pdf/file/6721824065.pdf?sid=voueb1k6g2phanb54s1pav2td1

tvdhorst, 2022-12-17:
Nice work @MichaelDvP, the status of the additional heater seems to come through correctly! Other new entities come through as well. I’ll monitor it and do some more checking tomorrow.

There’s a small error in the “Enable auxilliary heater”: the logic value (on/off) corresponds with the heatpump, but the name is reversed. You could fix it by changing the name to correspond with the heatpump: “Switch off auxiliary heater”, I guess it would be best to stick with the original names and values?

tvdhorst, 2022-12-17:
One more detail: Auxiliary heater on/off hyst should be Heating on/off hyst

It’s the hysteresis setting of the heatpump itself (and works as expected).

Auxiliary heater on delay doesn’t seem to come through in home assistant, its behaves correctly in ems-esp but doesn’t show up in home assistant.

Enable auxiliary heating works as expected when the name is changed to switch off.

MichaelDvP, 2022-12-18:
Ok, i've added the translation from the manuals and fixed the name and scales. For the hysteresis there are in manual also settings for cooling and pool and default values in offset 33 and 35. I've added but maybe exchanged, please check.

I don't know what's wrong with HA, i don't use it, but there is no dfference on side of ems-esp for these values.

proddy, 2022-12-18:
@tvdhorst I can test on the HA side

> But changing either of these values in ems-esp or home assistant results in a strange value. Setting to 150 resorts to 700 for example.

Can you do this in the EMS-ESP Web UI and then HA and see what is show in the logs when debug is on

> Cost ratio works as expected in ems-esp, but in home assistant it doesn’t allow decimals.

Can you send a screenshot?

MichaelDvP, 2022-12-18:
The 150/700 (forgot divider 5 when setting) is fixed in my latest commits. Only the missing decimals in HA  is unclear. 
The cost ratio is UINT, DV_NUMOP_DIV10 and DeviceValueUOM::NONE, maybe this confuses HA?

tvdhorst, 2022-12-18:
> The 150/700 (forgot divider 5 when setting) is fixed in my latest commits. Only the missing decimals in HA is unclear.
> The cost ratio is UINT, DV_NUMOP_DIV10 and DeviceValueUOM::NONE, maybe this confuses HA?

Cool, I’ll check this and other changes later today.

> > Cost ratio works as expected in ems-esp, but in home assistant it doesn’t allow decimals.
> 
> Can you send a screenshot?

In state attributes, step is set to 1, should be .1 I suppose?

```
min: 0
max: 20
step: 1
mode: auto
friendly_name: Energy cost ratio
```

MichaelDvP, 2022-12-18:
> In state attributes, step is set to 1, should be .1 I suppose?

Step is not set in discovery for this value (only for temperatures), where does it come from?

MichaelDvP, 2022-12-18:
I've added a step to all numerical values, please check and give feedback if it works now. hysteresis and delay now should have steps larger 1, ratio 0.1 and temperature settings mostly 0.5.

tvdhorst, 2022-12-18:
We're nearly there :)

Ratio works, I can set it at .1 intervals. Disable aux heater works as well, name and value correspond with the setting on the hp. Silent and parallel temp settings can be set, these are without decimal, this corresponds with the heatpump.

Three things to fix:
- Auxiliary heater on delay: setting I put is halved
- HP on/off host cooling/ heating: setting I put is doubled (pool setting is not visible in my setup)
- The names of the hyst heating/ cooling/ pool should either all start with 'HP' or all omit it (and maybe hp lowercase, as in other entity names?)

derSchreifritz, 2022-12-18:
amazing work! Thanks so much!

problems here seem reverse with tvdhorst, maybe we look at the same from different sides:
when I set auxiliary heater on delay, on either ha or ems interface, the displayed value is double.
with HP on/off hyst heat it is half.

I have one more wish, so I don't ever have to get out of my seat anymore.. 
Temp.diff. TC3/TC0 Heat
3...7...10 K: Set the permissible temperature differential
between the heat pump flow and return in heating mode.
I changed from 8 to 7.8. Pictured in NL

[image]
[log(7).txt](https://github.com/emsesp/EMS-ESP32/files/10254694/log.7.txt)

for another testbuild or so. I feel like it's Christmas already.

MichaelDvP, 2022-12-19:
fixed the double/half settings, added the difftemp. The manual says there is also a difftemp for cooling, i think it's somewhere near in same telegram (0x048D), 
This is the first time Bosch uses Kelvin for relative temperatures, before it was always °C.

derSchreifritz, 2022-12-19:
looking great.
Here is another log.
I changed the difftemp cooling from 3.0 to 3.2
then DHW comfort Switch-off temperature from 48 to 51 C
and DHW Eco Switch-off temperature from 48 to 49 C

I don't have eco+ mode, with dhw mode there is normal and comfort option, when I choose eco it switches to normal. So I guess normal in ha is whats eco in the manual/boiler, and eco means eco+ which I don't have, but thats ok.
[log(8).txt](https://github.com/emsesp/EMS-ESP32/files/10257577/log.8.txt)

MichaelDvP, 2022-12-19:
The telegram for DHW settings is `Boiler(0x08) -B-> All(0x00), ?(0x0499), data: 31 33 3F 3B 01`
31 = 49°C, Eco switch off
33 = 51°C Comf. switch off
The manual says there is a corresponding switch on. Is it 3F = 63°C ECO and 3B= 59°C Comfort? Or do the values represent switch off EM and ECO+? Or something completly different?

With the new entities, how is your free ram from system page (free heap/max alloc)?

derSchreifritz, 2022-12-19:
switch offs are workin fine.
The corresponding on switches are called dhw selected temperature for comfort, and dhw selected lower temp for eco (or normal in my case).
I can't set them lower than 40°C unfortunatly, and they must be at least 8K apart from the off switch.
I can't find a value with 63°C. 
59°C is dhw disinfection temperature here.
Do you want me to log more? I'm quite happy with what I have for now.
It's now 127/49/4 entities.

>With the new entities, how is your free ram from system page (free heap/max alloc)?

Heap (Free / Max Alloc)
115 KB / 40 KB

MichaelDvP, 2022-12-19:
> Do you want me to log more? I'm quite happy with what I have for now.

I think istallation settings, that are only set once at installation time are not really needed in ems-esp. Setting which can be used to tweak operation are good to have. I have no heatpump and do not know what is usefull. You decide what you need.
We can close this now , use the settings so far in the christmas release v.3.5 and reopen in a new beta if you want more to add.
> Heap (Free / Max Alloc) 115 KB / 40 KB

Hmm, 115 k is ok, but 40k largest block is low, could you go to help page, download the system info and post it here.

derSchreifritz, 2022-12-19:
[emsesp_info(1).txt](https://github.com/emsesp/EMS-ESP32/files/10259368/emsesp_info.1.txt)

tvdhorst, 2022-12-19:
Hyst and delay all work fine now. As does tc3/tc0 temp diffs.

Memory is 95/ 39.

I can’t test the dhw entities, they’re not in my installation. Nothing left for me to test I think? If there is is: give me a shout :)

heennkkee, 2022-12-19:
I'm running the 3.5.0b12 version and am seeing some new values that I guess comes from the changes in this issue, but some of them seems to be incorrect/not applicable for my device.

For instance these:
```
  "maxheatcomp": "3 kW",
  "maxheatheat": "4 kW",
  "maxheatdhw": "6 kW"
```

I have my unit configured to (labels roughly translated):
* Limit aux effect when compressor is running: 6 kW
* Limit aux effect: 9 kW
* Limit effect dhw: 6 kW

When changing the values on my device nothing updates in the API response (the above json is still returned).

Writing this I realise this is probably just a case of these values not being read properly for my device and I should probably create a separate issue, or?

MichaelDvP, 2022-12-20:
This was added in #645. The telegram is fetched, the values should update, but maybe it takes a minute.
If not, add a log to see the telegrams whan changing a setting.

tvdhorst, 2022-12-22:
I occasionally get an error when clicking on the boiler or thermostat on the dashboard to the view the entities:
Problem loading (507: )

[log-10.txt](https://github.com/emsesp/EMS-ESP32/files/10284578/log-10.txt)

MichaelDvP, 2022-12-22:
To see fetched telegrams, the log needs to be > 1 minute. The limits are fetched from telegram 0x0492.

For 507 see #823.

tvdhorst, 2023-01-06:
@MichaelDvP I was a bit occupied during the holidays, but I think we can close this ticket now?

Thanks a lot for your effort in adding the new entities.

closed 2023-01-06 by MichaelDvP