emsesp/EMS-ESP32 #804: add flag to MQTT Settings to use the old v3.4 entity ID names in Home Assistant

proddy, 2022-12-14:
There may be unhappy users responding to the new format of the MQTT Entity ID (unqiueid in the MQTT Discovery config topic) that was introduced in 3.5.0b11 and uses the `shortname` (as seen in the Configuration screen and MQTT payloads) instead of the `fullname` which is the English translation used in previous versions.

The suggestion is to add a flag to enable the old format from v3.4.

If we get more votes I'll add it as a feature.

sciurius, 2022-12-14:
I have a number of HomeAssistant scripts that would require updating. Then there are some dashboards.
This is the least problematic.

More important is that the entries are logged (recorded) in the database that, on its turn, is used by a lot of grafana graphs and other external tools. Hunting these down is interesting. Preventing discontinuity in the logged data is challenging.

Yes, the amount of work to be done is finite. The impact is just bigger than I anticipated. 

So I would very much appreciate a compatibility settings as suggested by proddy.

bbqkees, 2022-12-14:
@tp1de would also like this.

tp1de, 2022-12-14:
> @tp1de would also like this.

... I have adjusted HA dashboards and scripts in the meantime. Anyhow it is the recommendation from HA to keep unique_id and object_id identical. But several users in Discord are now facing the consequences. Whenever 3.5 is going to be final version I would expect quite some more questions /complaints.

proddy, 2022-12-14:
an alternative is to create a script using HA's CLI (https://github.com/home-assistant-ecosystem/home-assistant-cli) that renames all entities back to the old format.

tp1de, 2022-12-14:
I am still in the "learning phase" of HA and recognized quite some behavior I do not like. 
Any renaming of entities leaves orphaned statistics data in the database. I did a lot of testing to write statistics data directly with SQL statements to include the energy usage data from km200 and my PV into the energy dashboard and recognized that the data does not seem to be reorganized when entities do no exist anymore or are renamed.

A script to rename might be only a partial solution. But you are the experts ....

sciurius, 2022-12-14:
> an alternative is to create a script using HA's CLI (https://github.com/home-assistant-ecosystem/home-assistant-cli) that renames all entities back to the old format.

The names for the entities come from autodetection. Changing the 3.5 names to 3.4 (using the cli or whatever means) would not work since the 3.5 names will come back eventually.

proddy, 2022-12-14:
> The names for the entities come from autodetection. Changing the 3.5 names to 3.4 (using the cli or whatever means) would not work since the 3.5 names will come back eventually.

The script will change the v3.4 entity names to the newer v3.5 names, so retaining the history. You'll still need to update the scripts and Grafana, but it'll probably take you less time than for me to write a script or change the code to add a new flag ;)

sciurius, 2022-12-14:
I misunderstood -- I thought the name changing was from 3.5 to 3.4.
Interesting suggestion. Changing the names in the database and updating the gateway to v3.5 will provide continuity in the database.
Thereafter there is all time to chase name mismatches in scripts and graphs.

sciurius, 2022-12-15:
So I bit the bullet. This is what I did:
1. Create a rename script for the database entries (let me know if you're interested).
2. Modified EMS-ESP to not use MQTT (actually, I routed it to a test broker).
3. In the MQTT broker, I removed all autodiscovery items for EMS-ESP.
4. In HA, removed the EMS-ESP, boiler and thermostat from the MQTT integration.
5. Run the rename sql to update all database entries.
6. Reflashed the gateway with v3.5b11.
7. Re-attached to the boiler and verify it is working.
8. Modify EMS-ESP to use the real MQTT broker and enable autodiscovery.
9. Hunt down YAML files and change entity names.

Maybe some of the steps are unneccessary, but this worked for me.
Unfortunately, due to a recent change in HA the 'more info' popup will only show new data. 'show more' will show *all* data.

proddy, 2022-12-15:
i did a PR last night with the flag to support <3.5 MQTT entity formatting. It needs some changes which I'll do tonight, so that would be another solution to keep the old naming.

sciurius, 2022-12-16:
Great. Even though I could deal with the changes I'm not sure the above procedure would be safe in the hands of average users.

proddy, 2022-12-16:
> Great. Even though I could deal with the changes I'm not sure the above procedure would be safe in the hands of average users.

would still be great to see the scripts if you don't mind sharing.

sciurius, 2022-12-16:
Sure. I made a markdown document with all the entities and a small Perl program that reads the md file to produce a SQL rename script. I use PostgreSQL but I think the script will work with other SQL databases as well.

[rename.zip](https://github.com/emsesp/EMS-ESP32/files/10243512/rename.zip)

Note that the program generates a rename sql that groups everything in a single transaction, and rolls back at the end. This is for testing. If testing succeeds, or if you are reckless, change ROLLBACK to COMMIT and go.

mtc716, 2022-12-17:
> So I bit the bullet. This is what I did:
>
> 1. Create a rename script for the database entries (let me know if you're interested).
> 2. Modified EMS-ESP to not use MQTT (actually, I routed it to a test [...]

I did more or less the same by stopping recorder, modifying SQL DB and changing dashboard and yaml files. And it worked like charm. In the DB there are roughly 6 or 7 table columns at which you need to change the old to new name. And if you do not see the short term statistics, you need to check under developer settings and statistics. There might be the problem reported and by solving it there your statistics will be back.

proddy, 2022-12-17:
That's great we have a solution that others can use. Did you try using the CLI tool like `hass-cli entity rename <old> <new>` ?

sciurius, 2022-12-17:
> In the DB there are roughly 6 or 7 table columns at which you need to change the old to new name. 

Can you elaborate?

proddy, 2022-12-17:
I've added an option to the MQTT Settings that enables the 'old' entity ID way as it was in v3.4. For backwards compatibility

proddy, 2022-12-17:
@sciurius @mtc716 I'm closing this issue and moved the renaming HA chat to a new GH discussion here: https://github.com/emsesp/EMS-ESP32/discussions/814

mtc716, 2022-12-17:
> > In the DB there are roughly 6 or 7 table columns at which you need to change the old to new name.
> 
> Can you elaborate?

Just check in #814 , I added my SQL ;-)

proddy, 2022-12-17:
I added 3 options now
 - no base, use long descriptive entity names (v3.4 compatible)
 - no base, using the short names for the entity ID, matching the MQTT payloads
 - base prefixed, using short names as above supporting multiple instances of EMS-ESP

in b12.

mtc716, 2022-12-17:
Just checked the new b12 and it works like charm.
Only in the web, the dropdown field does not show the selected option once selected.
[image]

proddy, 2022-12-17:
yeah just spotted that myself and fixed it. I should really test before I commit, doh :(

mtc716, 2022-12-17:
> yeah just spotted that myself and fixed it. I should really test before I commit, doh :(

No big deal, well done anyhow...

sciurius, 2022-12-17:
What is the default value for this flag?

proddy, 2022-12-17:
> What is the default value for this flag?

the 2nd/middle option - no prefixed mqtt base name and using the shorter mqtt like entity names.

mtc716, 2022-12-17:
> What is the default value for this flag?

I would assume multiple instance short name, whereas it depends from which version the users are upgrading....

HansRemmerswaal, 2022-12-20:
I guess the v3.4 backwards compatibility mode is not future proof and will disappear over time? If so then it would be great if there is a kind of procedure to migrate from long name to short name. 

Is that script created by @mtc716 complete? I'm running HA on a Raspberry Pi, how to execute the script from there?
(I'm an expert in creating SQL on Oracle SQL but didn't manage yet to connect to the database on the Raspberry Pi)

mtc716, 2022-12-20:
> I guess the v3.4 backwards compatibility mode is not future proof and will disappear over time? If so then it would be great if there is a kind of procedure to migrate from long name to short name. I would be more than happy to support in this.
> 
> Is that script created by @mtc716 complete? I'm running HA on a Raspberry Pi, how to execute the script from there? (I'm an expert in creating SQL on Oracle SQL but didn't manage yet to connect to the database on the Raspberry Pi)

The perl script is made by @sciurius and I just checked the database at which the entity id's are used and provided the SQL Replace Statement.
So the easiest way to connect to the database is using the addon phpmysqladmin. By this you have direct access and can execute whatever SQL you need.

HansRemmerswaal, 2022-12-20:
In my HA I do not see the phpmysqladmin addon. I do see in the "Home Assistant Community Add-ons" the phpMyAdmin. But I guess that can't be use as my database is SQLite (sqlite:////config/home-assistant_v2.db)

sciurius, 2022-12-20:
Please note that my script generates SQL out of a description of entities (in the zip `names.md`). These are the entities for my Topline GB162 boiler and Moduline 400 thermostat. If you have different devices you may want to adapt the description file. And let us know.
And beware: the rename operation can take a long time (couple of hours)!

HansRemmerswaal, 2022-12-20:
But I guess that @proddy has the complete x-ref between the v3.4 and v3.5 names? As it's somehow used in the MQTT option.

HansRemmerswaal, 2022-12-20:
Hmm it will not be that easy. Different hardware, different databases behind it. Best would be something that is running in HA itself to convert the data.

zibous, 2022-12-22:
I have the same problem, so I can hardly install the newer versions because the name changes have a problem with my automation and scripts. 

In addition, I lose the history in the Influx database and in the Grafans representations. 

The only remedy would be the compatibility with previous version v3.5.0. 

Alternatively, I would have to switch off Homeassistant Disvovery and create a own one to map the new names to the old one.

proddy, 2022-12-23:
> I have the same problem, so I can hardly install the newer versions because the name changes have a problem with my automation and scripts.
>
> In addition, I lose the history in the Influx database [...]

In one of the last beta releases of v3.5 you can choose to use the old v3.4 MQTT naming so nothing should change in Home Assistant. If that is not the case let me know please.

sciurius, 2022-12-23:
Upgrading retains settings. Won't there be the risk that after the upgrade EMS-ESP contacts the MQTT server and spawns the new names to HomeAssistant before the user has had the chance to switch this setting?

proddy, 2022-12-23:
> Upgrading retains settings. Won't there be the risk that after the upgrade EMS-ESP contacts the MQTT server and spawns the new names to HomeAssistant before the user has had the chance to switch this setting?

good point, forgot about that. I'll add a check in the code that if it's coming from v3.4 to automatically keep the old MQTT setting. thanks!

zibous, 2022-12-23:
@proddy 
> In one of the last beta releases of v3.5 you can choose to use the old v3.4 MQTT naming so nothing should change in Home Assistant. If that is not the case let me know please.

Thanks, but how can i find the mapping for the version EMS-ESP-3_5_0b10-ESP32 ?

proddy, 2022-12-23:
> Thanks, but how can i find the mapping for the version EMS-ESP-3_5_0b10-ESP32 ?

What do you need exactly? The mapping of the old names to the new names?

I'll create a new GH issue to maintain a table of the shortname, fullname etc. We've done a few passes at this but its not consistent

closed 2022-12-24 by proddy