emsesp/EMS-ESP32 #805: Error while retrieving firmware file / 404 Client Error: Not Found for url

Mojtra, 2022-12-14:
Hi, I recieve the following error while flashing my ESP32:
`Unexpected error: Error while retrieving firmware file 'https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/bin/bootloader_dio_40m.bin': 404 Client Error: Not Found for url: https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/bin/bootloader_dio_40m.bin`

It looks like the file ending has changed from "bootloader_dio_40m.bin" to "bootloader_dio_40m.elf"

Mojtra, 2022-12-14:
Thank you. I was able to flash my ESP32 board with the flasher from you link. I've previously used the windows version of the flasher from here [https://github.com/MichaelDvP/EMS-ESP_Flasher/releases](url), which is linked in the following dokumentation [https://emsesp.github.io/docs/#/Uploading-firmware](url)

MichaelDvP, 2022-12-15:
Oh, thanks for the info, i'll update the link and remove this version from my GH.