emsesp/EMS-ESP32 #810: Question for Heating (/WW) Own Program Switchtimes, EMS-ESP32 function and what is already implemented?

giovanne123, 2022-12-15:
Hi, I was looking if it is already possible to already a) view and b) modify the daily switchtimes for heating (circle 1/2) by EMS-ESP32?

I found in EMS-ESP32 already:
[image]
[image]
[image]
[image]
[image]
[image]

But it is only ONE, only for ON and only for Monday!?
What about with OFF and the other weekdays?

In Buderus Software in the past I could see in graphical or table view (info: does not correlate with above EMS-ESP32 values because older screengrab):
HC1:
[image]
HC2:
[image]

I'm not sure if it is related e.g. for HC1 to "Telgram 0x3F"? 
-> HC1: https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#hk1schaltzeiten
-> HC2: (Typ 0x49)
-> "Schaltzeiten allgemein": https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#schaltzeiten_allgemein

**
Can this be also realized by EMS-ESP32?
- first to have a view with current own program settings (e.g. only in table view)
- second to be able to edit the settings by EMS-ESP32

or can you describe what is already implemeted in EMS-ESP32 and how it is working?**

And, maybe (but not so important for me) also for WW switchtimes?
[image]

proddy, 2022-12-15:
There's some info in https://github.com/emsesp/EMS-ESP32/issues/182#issuecomment-967149449

MichaelDvP, 2022-12-15:
There were a lot of discussions about that in different issue# for RC30, RC20, RC300, Junkers... 
Now it's poorly implemented for RC20/30/35. Send `no` only to read an entry, send full data to change an entry.
- We can not publish all entries in the mqtt, this blows the buffer. 
- also showing all on the thermstat webpage is to large
- the settings are different for other thermostats, so we need extra settings pages for each thermostat model
- same for holiday modes/dates

giovanne123, 2022-12-15:
@proddy 
Thanks, yes there are some info in the comment you linked to and your last comment (when closing) in the thread is, that it should be resoved in 3.4. So I think that is the state I have found in 3.5.0b10 and described above.

For me in a first step it would be enough to see the current on/off's of programm own1.
For a short look e.g. in HA to see what times currently the heating will go on/off.
(so that I do not have to scroll to the RC30 controller itself. That was to complicated for me in the past ;-) I used the Buderus Ecosoft to view and change the on/off's before - but replaced the Ecosoft by EMS-ESP32 :-))

No need for me to change from EMS-ESP32 so far, because I think I also haven't understud fully how the current implementation works... have to dig deeper into the linked thread...

@MichaelDvP , ok thanks for the infos.
So seems not so simple.
Maybe we can get the important part (or telegram data) somehow to HA and the encryption of the data needs to be done on that end?

But maybe to complicated for the effort and because the on/off's are not that often changed...

... then I will write the on/off times manually to a HA table ;-) - but therefore (without Ecosoft) I have to scroll to RC30 controller :-(

closed 2023-02-19 by proddy