emsesp/EMS-ESP32 #811: Provide numerical uptime in API response

heennkkee, 2022-12-16:
**Is your feature request related to a problem? Please describe.**
Uptime is only provided as a timestamp at the endpoint /api/system/info, I suggest that the numerical uptime is provided as well. Providing only the timestamp format puts higher requirements on the user to handle it correctly (parsing, more complex data type to store), while the numerical value is easier to "just grab and store". 

**Describe the solution you'd like**
I would like this line, https://github.com/emsesp/EMS-ESP32/blame/fc057d18c99e9ded6fe34f0bd3aa46480d9d0879/src/system.cpp#L983, to be uncommented, to avoid unnecessary calls perhaps it'd make a little more sense to provide the uptime in ms (and only get the uptime once when providing the system info output).

**Describe alternatives you've considered**
None

**Additional context**
I can't find any reason related to the change of this line that indicates why it was commented out, perhaps it was just a mistake :)

proddy, 2022-12-16:
np, will add back.

proddy, 2022-12-17:
in b12

closed 2022-12-17 by proddy