emsesp/EMS-ESP32 #812: EMS bus disconnected when thermostat is off

tolmalev, 2022-12-16:
**Question**
Hi, I just installed the device and it works perfectly when boiler is on. I can see all the information I need it HA and this is amazing!
However as soon as temperature reaches target and thermostat (simple on-off) turns the boiler off I stop receiving any data from ems-esp. Web interface shows an error "EMS bus disconnected"
Everything starts working right after thermostat turns on again.

**Screenshots**
[image]

**Device information**
```
"Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 2,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "disconnected"
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "40.00",
      "entities": 45,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26",
      "handlers ignored": "0x05 0x22 0x01 0x03 0x04"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "40.00",
      "entities": 0
    }
```

**Additional context**
Boiler model: Worchester Bosh Greenstar 18i

proddy, 2022-12-16:
can you turn debug on in the log (either web or console) and capture a few telegrams before and after the boiler is switched off

MichaelDvP, 2022-12-16:
Does the thermostat cut the power line of the boiler or is it connected to the WA clamps (relais input heat requirement)?

tolmalev, 2022-12-16:
That's a good question. I was sure that it's not happening but now I started doubting. Boiler was installed 10 years ago and it was before I moved to the property so I'm not 100%

I found a workaround that solves the problem for me in HA: created a template state 'boiler_ems_online' that switches to off of there were no updates from ems-esp in the past 30sec and use this state to get the real value for other important variables such as power consumed

bbqkees, 2022-12-19:
Please see the following note from the compatibility list:
https://bbqkees-electronics.nl/ems-device-compatibility/
[image]

closed 2023-02-19 by proddy