emsesp/EMS-ESP32 #820: Webinterface shows wrong values

MaartenRs, 2022-12-19:
Webinterdace shows values of 21.474.836 kWh, while Home Assistent shows -1.

EMS-ESP Version
v3.5.0b12

Apparaat (Platform / SDK)
ESP32 / v4.4.2

<html>
<body>

TYPE | DESCRIPTION | ENTITIES |  
-- | -- | -- | --
  | Boiler | Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i | 122 |  
  | Thermostat | RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410 | 48 |  
  | Thermostat | RC100H | 4 |  
  | Sensors | Attached EMS-ESP Sensors | 0

</body>
</html>

<html>
<body>

cooling control starts | 29 |  
-- | -- | --
dhw control starts2 | 21.474.836 |  
pool control starts | 21.474.836 |  
total energy consumption | 1.946 kWh |  
total energy consumption compressor | 1.947 kWh |  
energy consumption compressor heating | 1.898 kWh |  
energy consumption compressor dhw | 21.474.836 kWh |  
energy consumption compressor cooling | 22 kWh |  
energy consumption compressor pool | 21.474.836 kWh |  
total auxiliary electrical heater energy consumption | 21.474.836 kWh |  
auxiliary electrical heater energy consumption heating | 21.474.836 kWh |  
auxiliary electrical heater energy consumption dhw | 21.474.836 kWh |  
auxiliary electrical heater energy consumption pool | 21.474.836 kWh |  
total energy supplied | 7.394 kWh |  
total energy supplied heating | 7.357 kWh |  
total energy warm supplied dhw | 21.474.836 kWh |  
total energy supplied cooling | 36 kWh |  
total energy supplied pool | 21.474.836 kWh |  
compressor power output | 3,5 kW |  
 

<br class="Apple-interchange-newline">
</body>
</html>

MichaelDvP, 2022-12-19:
Confusion in browser cache? Shut down and restart browser and (to be sure) also ems-esp.

MaartenRs, 2022-12-19:
Restart PC, empty browser cache and restart ems-esp.
Error is still there. If I connect with another browser (Firefox), the result is the same.

heennkkee, 2022-12-19:
Do you expect these values to be read? I have similar values being sent out, like for instance 

total energy supplied pool: 21.474.836 kWh

and the exported value is -1 (it was  21.474.836 when running 3.5.0b4).

I assume this is some type of default value for when a device "could have" this value (so the EMS-ESP32 detects the value) but the value has never been set in the device. I for instance do not have any pool that my heater supplies heat to, hence that value is "empty" (or defaulted to 21.474.836). I have similar for cooling, solar systems and some more.

MichaelDvP, 2022-12-21:
Please test https://github.com/MichaelDvP/EMS-ESP32/releases and give feedback if it is fixed or not.

pswid, 2022-12-21:
I'm experiencing similar(?) issue. In my case `uptimecontrol` and `uptimecompheating` are not displayed at all:

[image]

v3.5.0b11 was OK:
[image]

Your b12 compilation doesn't solve the problem.

MichaelDvP, 2022-12-22:
> Your b12 compilation doesn't solve the problem.

Thanks, now i understand this strange value, it's an artifact from float conversion.
I'll update my testbuild. Please check.

MaartenRs, 2022-12-22:
Hi Michael,

 

This version solved the problem.

However, if I double click “boiler” it shows the four entities of the Thermostat RC100H. Double clicking Thermostat RC300.. shows the 116 entities if the Boiler and double clicking Thermostat RC100H shows the 48 entities of Thermostat RC300..

 

Furthermore the interface with Home Assistant is not working correctly anymore. Enitites are not found or not avaialbel.

 

Best regards,

 

Maarten Reintjes

MichaelDvP, 2022-12-22:
Nice finding, @proddy we have a large memory leak when doubleclick on a dashboard line. 

@MaartenRs reboot emsesp and only use single click and wait for data. 
The Homeassistant entities has to be configured in mqtt, use the v3.4.4 option to keep compatible to old version.

MaartenRs, 2022-12-22:
I solved the error "Problem loading" by power off the gateway restart didn't solve the issue.

Single click works fine.

My version of MQTT is: 
Mosquitto broker
Current version: 6.1.3

I have restarted HA, but still Enitites are not found or not available.
[image]

MichaelDvP, 2022-12-22:
[image]

pswid, 2022-12-22:
Small update. I'm already using "long name (v.3.4)" option. Regarding number of entities. I counted them and value over 100 is correct in B12/B13 (for my system). So indeed more than 20 new entities was added since B11. Nice!

MichaelDvP, 2022-12-22:
The bad side of more entities is ram and the 507 on dashboard/customization, see #823

MaartenRs, 2022-12-22:
Unfortenatly I don't see (anymore) MQTT at the left of my screen.
If I go to settings, I can't find this option.
[image]

MichaelDvP, 2022-12-22:
@MaartenRs : You are searching in Homeassistant., this is EMS-ESP. Open EMS-ESP webpage and set the options.

MaartenRs, 2022-12-22:
Stupid me 😡
After having set this 
[image]

I get this error message, even after re-booting the gatway (power off).
[image]

MichaelDvP, 2022-12-22:
see #823

MichaelDvP, 2022-12-22:
@pswid Could you please test with https://github.com/MichaelDvP/EMS-ESP32/releases
set a log to debug and navigate to dashboard boiler and Settings/Customization Boiler
The log (please attach) should show the buffer used for the display. I've reduced the buffer to 16k to avoid 507, if it is to small a few entities at the end will be skipped.

pswid, 2022-12-23:
Indeed the buffer is too small and the customization page is truncated (~50 entities from 104 total is missing!). Furthermore if I try to expand Boiler at dashboard only empty/white page is displayed. Log attached. It says `Customization buffer used: 7639`

[image]
[image]

[log.txt](https://github.com/emsesp/EMS-ESP32/files/10295384/log.txt)

MichaelDvP, 2022-12-23:
Sad, so the size measurement is packed json, but the buffer have to hold unpacked json. I'll set back to 20k.

MichaelDvP, 2022-12-27:
Back to OP: 
@MaartenRs Is the wrong value 21.474.836 solved in latest builds?
Can this be closed or i something to do?

MaartenRs, 2022-12-28:
Yes, you can close this.

Thanks for your help.

commit 6f7fa6ab (2022-12-20): fix #820, NOTSET check for 4 byte values.
commit ab1924d2 (2022-12-24): fix #820, 4byte-values
closed 2022-12-28 by MichaelDvP