emsesp/EMS-ESP32 #821: No telegram type handler found for ID xxx

HansRemmerswaal, 2022-12-19:
**Question**
I have installed EMS-ESP-3_5_0b12-ESP32 and in the logs I see...

No telegram type handler found for ID 0x239 (src 0x10)
No telegram type handler found for ID 0x507 (src 0x10)
No telegram type handler found for ID 0x50A (src 0x10)
No telegram type handler found for ID 0x49F (src 0x08)
No telegram type handler found for ID 0x4A0 (src 0x08)
No telegram type handler found for ID 0x4A3 (src 0x08)

I assume that these are sensor values that are not yet defined in the EMS-ESP32 software? 
If my assumption is correct, then how can I support in finding out what the related sensor values are?

Example:

2022-12-20 03:29:15.217 T 4800: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0x049F), data: 00 00 0A 8C 08 34 00 00 07 08 02 58 07 08
2022-12-20 03:29:15.217 D 4801: [emsesp] No telegram type handler found for ID 0x49F (src 0x08)

How to read that telegram message? If I know the value then I might be able to figure out what it is.

MichaelDvP, 2022-12-20:
There are a lot of unknown telegrams and a lot of unknown fileds in the known telegrams.
Normally we go the reverse way. If you are missing a value, log the telegrams and note the EXACT time when the value changes, Then you can check the telegram at this time (not only unknown) for this new value. Or to find a setting: log and change the setting on the thermostat, then check the log for that change-telegram. Because telegrams come in every few milliseconds, it does not help if people send a log and write " somewhere around 15:30 the value has changed".

A telegram is only a number of bytes, each byte could be anything and value can be scaled. In your example you have 0x0A in offset 2, this could mean:
- A measurement or setting with value 10
- A measurement or setting with value 1.0 and scale / 10
- A measurement or setting with value 5.0 and scale / 2
- A measurement or setting with value 100 and scale * 10
- A status byte with bits 2 and 4 set, maybe 2 valves switched
- A counter with 3 or 4 bytes and value 2748 (0x00000ABC)
- etc.

HansRemmerswaal, 2022-12-20:
Ok will check for values I'm missing. If I find anything I will open a new issue.

closed 2023-02-19 by proddy