emsesp/EMS-ESP32 #822: Set device_class to duration in HA sensor definitions

heennkkee, 2022-12-19:
**Is your feature request related to a problem? Please describe.**
When automatically detecting devices through MQTT in Home Assistant the uptime counters could have their `device_class` set to `duration`. I believe that the `unit_of_measurement` would also need to be changed from `minutes` to `min`.
This effects how the values are rendered by default in HA, and although I honestly prefer the way the default rendering for the values are "as is", flagging the values with the correct attributes are more correct and "future proofer".

With `device_class` set:
[image]

Existing settings:
[image]

**Describe the solution you'd like**
The above one

**Describe alternatives you've considered**
To ignore this and keep going as is works just fine, but thought this was an improvement.

**Additional context**
None

proddy, 2022-12-23:
I'll test it out, thanks for submitting.

proddy, 2022-12-23:
looks like HA does use "min" https://github.com/home-assistant/core/blob/d7ac4bd65379e11461c7ce0893d3533d8d8b8cbf/homeassistant/const.py#L412

closed 2022-12-28 by proddy