emsesp/EMS-ESP32 #823: WEB interface - Problem loading (507:)

HansRemmerswaal, 2022-12-19:
I upgraded today to the latest version of EMS-ESP-3_5_0b12-ESP32.bin, I already had v3.5 installed (the one with the defrost fix in the compressor status). In the web interface I see that the number of entities for the boiler has increased from 107 to 127. So far so good.

When I click in the WEB interface on one of the devices I can't see the sensors, instead I'm getting "Problem loading (507:).
I have tried with Chrome and Edge.

**Screenshots**
[image]

**Device information**
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10263223/emsesp_info.txt)

HansRemmerswaal, 2022-12-19:
Not sure what happened but suddenly I'm now getting the values...

[image]

MichaelDvP, 2022-12-20:
Known issue, see #769 and #800. 
http:507 is `Insufficient Storage`
For web output of dashboard and cstomization we need a large json buffer (20k). Sometimes there is no continous 20k block free in heap and  the http:get was answered by null, making the web unresponsable. Now it shows 507 and keeps responsable.
Check system page for free heap, if the max alloc block goes below ~39k you'll get the 507. But ems-esp bus is still read and mqtt sent.

Actually there is no solution how to save memory, reduce heap fragmentation or  switch to ESP32 wrover with PSRAM (or ESP32S3).

MaartenRs, 2022-12-22:
My Heap free shows 101kB and I still get this 507 error.
[image]

proddy, 2022-12-22:
@MichaelDvP I think we're hitting the 20k buffer and maybe need to switch to event streaming in a future version for sending data to the dashboard, rather than POSTing it. 

@MaartenRs when you get this error can you go into your chrome/edge browser, turn on developer mode (usually F12 on Windows), go to Network, hit Record, reload the page, click on the Boiler device so the error occurs again and then see what is in the devicData payload, especially the Content-Length

[image]

MichaelDvP, 2022-12-22:
> @MichaelDvP I think we're hitting the 20k buffer and maybe need to switch to event streaming in a future version for sending data to the dashboard, rather than POSTing it.

Or split into smaller packages (like boiler, boiler_ww). There is free heap, but 20k on block is difficult now.

MichaelDvP, 2022-12-22:
@proddy Have we set the buffer to 20k before using msgpack? I can't remember. I've logged the size and get with 54 entities only 5k buffer use. I think we can reduce to 16k for heatpump and ~110 entities. If buffer is to small, only a few entities are missing.

proddy, 2022-12-22:
> > @MichaelDvP I think we're hitting the 20k buffer and maybe need to switch to event streaming in a future version for sending data to the dashboard, rather than POSTing it.
> 
> Or split into smaller packages (like boiler, boiler_ww). There is free heap, but 20k on block is difficult now.

or load in batches from the server and show pages in the table (https://react-table-library.com/?path=/docs/client-vs-server--server-side)

MichaelDvP, 2022-12-23:
I don't fully understand solution 2. Does it mean we always need a webclient connected to create the ha discovery? 

I've made a test and measured the msgpack size for dashboard and customizations. Changed the code to register all ems/ems+/ht3/heatpump entities to my boiler. Most of the entities have no value, but shows up in customizations. 

Customization: 
Boiler 165 entities: with 122 untagged and 43 tagged ww. Buffersize used  8221 bytes.
Thermostat with 1 hc: 62 entities: 10 untagged, 13 ww and 39 hc2, Buffersize used 3597 bytes. Calculating 4 hc this expands to 182 entites and ~11k buffer.
Dashboard:
Boiler 51 entities: 31 untagged, 20 ww, buffer used 2450 bytes
Thermostat 54 entities: 6 untagged, 13 ww, 35 hc2, buffer used 3999 bytes. Calculating to 4 hc buffer needed is < 12k.

Seems we can reduce the buffer for both to 12k and have less http:507 errormessages. Splitting thermostat in packages like mqtt-single format with extra packages for each hc we only need 4k buffer, but boiler (heatpump) needs >6k and >2,5k for ww (results in 8k buffer allocation to fit all). 

With all these entities i have free heap 116k, max alloc 65k and no issues.

I'll change my testbuild to 12k buffersize.

proddy, 2022-12-23:
cool that it fits into 12k. I looked at how to compress the data even more (outside msgback) last night but came to dead ends. My solution 2 does mean calling a JS function via a tunnel to fetch the translation. Which is probably not a good solution

MichaelDvP, 2022-12-23:
BTW: reducing buffers:
on Apr. 4 in #444 we set this https://github.com/emsesp/EMS-ESP32/blob/9802581301369031923ae659fb2214f2b225290a/src/web/WebCustomizationService.cpp#L49-L50 to 4k and parallel you limit the entities changed at once to 50 to fit in old buffer. So we can reduce back to 2k.

MichaelDvP, 2022-12-24:
Oh, there was a bug in reducing the weblog buffer on low memory, the limit was reduced, but not the buffer. I've fixed it with a dynamic memory dependent limit, see https://github.com/MichaelDvP/EMS-ESP32/commit/1f1feed3ae1772d3cf73a7140cb3c9363e839832

MichaelDvP, 2022-12-27:
With b14 and pr #844 #845 the http::507 is gone. 
Future complains about same low memory problem will have theme "truncated dashboard/customization list" or similar.

proddy, 2022-12-28:
think we can close this.

closed 2022-12-28 by proddy