emsesp/EMS-ESP32 #826: Set temp not possible with 2 CR10 thermostats

marcstone-be, 2022-12-21:
**Bug description**
V3.4.4
Everything is working perfectly with 1 CR10 thermostat connected, either the one of HC1 or the other for HC2.
When both HC1 and HC2 are connected, the readings are all fine. But only the values can be written to HC1, and not to HC2.

**Steps to reproduce**
- Connect 1 thermostat to HC1. Set manual temperature works.
- Connect 1 thermostat to HC2. Set manual temperature works.
- Connect 1 thermostat to HC1 AND 1 thermostat to HC2. Set manual temperature only works on HC1.

**Expected behavior**
When both thermostats are connected, setting manual temp should work on both HC1 and HC2.

Sysinfo: [emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10279037/emsesp_info.txt)

**Logging when only HC2 thermostat (0x19) is connected:**
000+00:03:08.222 N 19: [emsesp] Me(0x0B) <- Thermostat(0x19), RC300Curves(0x029C), length: 0x19
000+00:03:08.269 N 20: [emsesp] Thermostat(0x19) -> Me(0x0B), RC300Curves(0x029C), data: 04 01 00 00 00 01 00 00 3C
000+00:03:13.657 N 21: [emsesp] Me(0x0B) -> Thermostat(0x19), RC300Set(0x02BA), data: 1D (offset 10)
000+00:03:14.250 N 22: [emsesp] Thermostat(0x19) -> All(0x00), RC300Monitor(0x02A6), data: 1D (offset 3)
000+00:03:14.436 N 23: [emsesp] Thermostat(0x19) -> All(0x00), RC300Set(0x02BA), data: 1D (offset 10)
000+00:03:15.747 N 27: [emsesp] Thermostat(0x19) -> All(0x00), RC300Monitor(0x02A6), data: 00 9C 21 1D 00 00 1D 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00
000+00:03:16.517 N 28: [emsesp] Me(0x0B) <- Thermostat(0x19), RC300Monitor(0x02A6), length: 0x19
000+00:03:16.633 N 29: [emsesp] Thermostat(0x19) -> Me(0x0B), RC300Monitor(0x02A6), data: 00 9C 21 1D 00 00 1D 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 06 1C 00
000+00:03:16.708 N 30: [emsesp] Thermostat(0x19) -> All(0x00), RC300Monitor(0x02A6), data: 06 1C 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 22)
000+00:03:16.902 N 31: [emsesp] Thermostat(0x19) -> Unknown(0x21), ?(0x02E2), data: 01 00 00 00 01
000+00:03:17.428 N 32: [emsesp] Me(0x0B) <- Thermostat(0x19), RC300Monitor(0x02A6), length: 0x20 (offset 25)
000+00:03:17.501 N 33: [emsesp] Thermostat(0x19) -> Me(0x0B), RC300Monitor(0x02A6), data: 00 07 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)

Same in system log:
2022-12-21 15:39:38.844 D 900: [command] Calling command 'thermostat/seltemp' (selected room temperature) with value 14.0
2022-12-21 15:39:39.179 D 901: [telegram] Sending write Tx [#116], telegram: 8B 19 FF 0A 01 BA 1C
2022-12-21 15:39:39.202 D 902: [emsesp] No telegram type handler found for ID 0x2BA (src 0x0B)
2022-12-21 15:39:39.220 D 903: [emsesp] Last Tx write successful

**When both thermostats are connected, writing temperature to HC1 (0x18) works:**
000+00:59:53.062 N 29: [emsesp] Me(0x0B) -> Thermostat(0x18), RC300Set(0x02B9), data: 1C (offset 10)
000+00:59:54.362 N 30: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 37 02 2E 02 2E A1 00 00 03 00 00 84 6C 00 09 23 00 80 00
000+00:59:54.559 N 31: [emsesp] Thermostat(0x18) -> All(0x00), RC300Monitor(0x02A5), data: 1C (offset 3)
000+00:59:54.742 N 32: [emsesp] Thermostat(0x18) -> All(0x00), RC300Monitor(0x02A5), data: 1C (offset 6)
000+00:59:55.067 N 33: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 00 ED 00 00 00 02 00 C0 80 00 02 2E 80 00 00 00 FF 00 00 00 CB 00 00 00
000+00:59:55.671 N 34: [emsesp] Me(0x0B) <- Thermostat(0x18), RC300Monitor(0x02A5), length: 0x19
000+00:59:55.796 N 35: [emsesp] Thermostat(0x18) -> Me(0x0B), RC300Monitor(0x02A5), data: 00 BF 21 1C 00 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 07 75 00
000+00:59:55.860 N 36: [emsesp] Thermostat(0x18) -> All(0x00), RC300Set(0x02B9), data: 1C (offset 10)
000+00:59:56.627 N 37: [emsesp] Me(0x0B) <- Thermostat(0x18), RC300Monitor(0x02A5), length: 0x20 (offset 25)
000+00:59:57.444 N 38: [emsesp] Me(0x0B) <- Thermostat(0x18), RC300Monitor(0x02A5), length: 0x20 (offset 25)
000+00:59:57.535 N 39: [emsesp] Thermostat(0x18) -> Me(0x0B), RC300Monitor(0x02A5), data: 00 07 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)

syslog:
2022-12-21 16:42:54.632 D 4532: [command] Calling command 'thermostat/seltemp' (selected room temperature) with value 14
2022-12-21 16:42:54.632 D 4533: [emsesp] Write command successful
2022-12-21 16:42:54.975 D 4534: [telegram] Sending write Tx [#226], telegram: 8B 18 FF 0A 01 B9 1C
2022-12-21 16:42:55.062 N 4535: [emsesp] Me(0x0B) -> Thermostat(0x18), RC300Set(0x02B9), data: 1C (offset 10)
2022-12-21 16:42:55.062 D 4536: [emsesp] No telegram type handler found for ID 0x2B9 (src 0x0B)
2022-12-21 16:42:55.065 D 4537: [emsesp] Last Tx write successful

**When both thermostats are connected, writing temperature to HC2 (0x19) does not work:**
000+00:53:50.827 N 6: [emsesp] Thermostat(0x19) -> All(0x00), RC300Monitor(0x02A6), data: 00 9C 21 1C 00 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00
000+00:53:51.767 N 7: [emsesp] Thermostat(0x19) -> All(0x00), RC300Monitor(0x02A6), data: 06 1C 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 22)
000+00:53:51.968 N 8: [emsesp] Thermostat(0x19) -> Unknown(0x21), ?(0x02E2), data: 01 00 00 00 01
000+00:53:52.096 N 9: [emsesp] Unknown(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+00:53:52.179 N 10: [emsesp] Unknown(0x20) -> All(0x00), ?(0x02D7), data: 00 00 00 80 00 00 FF 00 03
000+00:53:52.423 N 11: [emsesp] Unknown(0x21) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+00:53:52.517 N 12: [emsesp] Unknown(0x21) -> All(0x00), ?(0x02D8), data: 00 00 00 80 00 00 FF 00 03
000+00:53:52.815 N 13: [emsesp] Unknown(0x22) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+00:53:52.885 N 14: [emsesp] Unknown(0x22) -> All(0x00), ?(0x02D9), data: 00 00 00 80 00 00 FF 00 05
000+00:53:53.445 N 15: [emsesp] Thermostat(0x19) -> Unknown(0x21), ?(0x02EC), data: 00

System log:
2022-12-21 16:36:44.594 D 4138: [command] Calling command 'thermostat/seltemp' (selected room temperature) with value 15
2022-12-21 16:36:44.594 D 4139: [emsesp] Write command successful
2022-12-21 16:36:52.827 N 4143: [emsesp] Thermostat(0x19) -> All(0x00), RC300Monitor(0x02A6), data: 00 9C 21 1C 00 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00
2022-12-21 16:36:53.767 N 4144: [emsesp] Thermostat(0x19) -> All(0x00), RC300Monitor(0x02A6), data: 06 1C 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 22)
2022-12-21 16:36:53.968 N 4145: [emsesp] Thermostat(0x19) -> Unknown(0x21), ?(0x02E2), data: 01 00 00 00 01
2022-12-21 16:36:53.968 D 4146: [emsesp] No telegram type handler found for ID 0x2E2 (src 0x19)
2022-12-21 16:36:54.096 N 4147: [emsesp] Unknown(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
2022-12-21 16:36:54.096 D 4148: [emsesp] No telegram type handler found for ID 0x1A (src 0x20)
2022-12-21 16:36:54.179 N 4149: [emsesp] Unknown(0x20) -> All(0x00), ?(0x02D7), data: 00 00 00 80 00 00 FF 00 03
2022-12-21 16:36:54.179 D 4150: [emsesp] No telegram type handler found for ID 0x2D7 (src 0x20)
2022-12-21 16:36:54.423 N 4151: [emsesp] Unknown(0x21) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
2022-12-21 16:36:54.423 D 4152: [emsesp] No telegram type handler found for ID 0x1A (src 0x21)
2022-12-21 16:36:54.517 N 4153: [emsesp] Unknown(0x21) -> All(0x00), ?(0x02D8), data: 00 00 00 80 00 00 FF 00 03
2022-12-21 16:36:54.517 D 4154: [emsesp] No telegram type handler found for ID 0x2D8 (src 0x21)
2022-12-21 16:36:54.815 N 4155: [emsesp] Unknown(0x22) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
2022-12-21 16:36:54.815 D 4156: [emsesp] No telegram type handler found for ID 0x1A (src 0x22)
2022-12-21 16:36:54.885 N 4157: [emsesp] Unknown(0x22) -> All(0x00), ?(0x02D9), data: 00 00 00 80 00 00 FF 00 05
2022-12-21 16:36:54.885 D 4158: [emsesp] No telegram type handler found for ID 0x2D9 (src 0x22)
2022-12-21 16:36:55.445 N 4159: [emsesp] Thermostat(0x19) -> Unknown(0x21), ?(0x02EC), data: 00
2022-12-21 16:36:55.445 D 4160: [emsesp] No telegram type handler found for ID 0x2EC (src 0x19)

Thanks for looking into this.
Marc

MichaelDvP, 2022-12-21:
> 2022-12-21 16:36:44.594 D 4138: [command] Calling command 'thermostat/seltemp' (selected room temperature) with value 15

You are writing to thermostat/seltemp without specifing the hc, so it's always the first one written. 
Use ' thermostat/hc2/seltemp'

marcstone-be, 2022-12-21:
Maybe one thing to mention, is that I changed the temperature from the EMS ESP GUI itself.
So I have no impact about the syntax of the sent command, this is generated by EMS ESP.
Also from Home Assistant through MQTT this is the same result, but to be sure that it's not mqtt related I was only changing the temperature from within the EMS ESP webpage.

MichaelDvP, 2022-12-21:
Oh, in 3.4.4 the log do not show the hc. 
I think i now where the bug is, the command is send to the wrong thermostat, but with right hc,. But this should not log `Write command successful`. 
Please log all when sending the command, I like to see the send telegram Me(0x0B) -> Thermostat(0x,

marcstone-be, 2022-12-21:
OK here are the loggings:
First I set the temp of HC1 (0x18) to 14,5. Here are the syslog and watch log:
[hc1 14_5 watch.txt](https://github.com/emsesp/EMS-ESP32/files/10280765/hc1.14_5.watch.txt)
[hc1 14_5 syslog.txt](https://github.com/emsesp/EMS-ESP32/files/10280766/hc1.14_5.syslog.txt)

Then I set the temp of HC2 (0x19) to 13. Here are the syslog and watch log:
[hc2 13 watch.txt](https://github.com/emsesp/EMS-ESP32/files/10280767/hc2.13.watch.txt)
[hc2 13 syslog.txt](https://github.com/emsesp/EMS-ESP32/files/10280768/hc2.13.syslog.txt)

After these actions, hc1 was still at 14.5 and hc2 was not changed either.

MichaelDvP, 2022-12-22:
Sorry, log_debug+watch does not help, i need log_all to see the outgoing telegrams. 
I don't understand why the log shows `Write command successful` when a hc is addressed, that does not exist on this device.

proddy, 2022-12-22:
it would be easier to debug if @marcstone-be upgrades to v3.5?

MichaelDvP, 2022-12-22:
> it would be easier to debug if @marcstone-be upgrades to v3.5?

Yes, this will be required in future for testing possible solutions. 

@marcstone-be Your system info shows that you have 3 mixing circuits with product-id 193. If you tell me the module name i can add it correctly to database to support them. In my testbuild i've called it MM300 because of the 3 circuits, but afaik there is no module with this name from bosch. Use this build https://github.com/MichaelDvP/EMS-ESP32/releases to update. (with repeated update error 500 use usb-flashing https://github.com/emsesp/EMS-ESP-Flasher/releases)

@proddy I think we need some more rework in the command handling for multiple thermostats. Now we have only one command function per device-type (e.g. thermostat) and pass the hc as id to it. But this will result in a command telegram to wrong device-id. Working good with master thermostat and remotes, but not in case of single thermostats per hc. I expect the same problem for mixing commands. 
My idea so far: add device_type to command vector and in find_command check if the device supports the hc/wwc/hs the id points to (via emsdevice->has_tag()). Other solution is to add command functions per hc, but this will blow up the command vector. Spoofing the dest-id in emsdevice write_command (similar to the old masterthermostat-solution) is IMO more complicated for this case with extra thermostats.

proddy, 2022-12-22:
> My idea so far: add device_type to command vector and in find_command check if the device supports the hc/wwc/hs the id points to (via emsdevice->has_tag()).

this is a good solution. Shall we get it into 3.5? I'm still working this week and next so little time for coding.

marcstone-be, 2022-12-22:
Hi, thanks for all your effort!
I will upgrade to the build you mentioned with EMS-ESP Flasher.
In the mean time here are the log_all logs of 3.4.4.
- hc1 set to 14.5 at 12:10.513
[hc1 14_5 log_all.txt](https://github.com/emsesp/EMS-ESP32/files/10285888/hc1.14_5.log_all.txt)
- hc2 set to 15 at 00:14:20.481
[hc2 15 log_all.txt](https://github.com/emsesp/EMS-ESP32/files/10285893/hc2.15.log_all.txt)

I notice that in the hc2 there is no Me(0x0B) -> Thermostat(0x19), RC300Set..

marcstone-be, 2022-12-22:
I upgraded to 3.5.
Sysinfo: 
[emsesp_info (3).txt](https://github.com/emsesp/EMS-ESP32/files/10286577/emsesp_info.3.txt)
I notice that since 3.5 I get EMS read errors, never had that before. Tried all tx modes but still the same.
Also in the emsespinfo file is stated that I am using HT3 Tx mode, while it's configured to EMS.

Here are the new logs:
- hc1 to 14 degrees (succesful)
[Uploading hc1 to 14.txt…]()
- hc2 to 15 degrees (failed)
[hc2 to 15.txt](https://github.com/emsesp/EMS-ESP32/files/10286559/hc2.to.15.txt)

MichaelDvP, 2022-12-22:
Ah, now we have `[emsesp] Write command failed callback function failed (Invalid)` what i have expected.
HT3 is how the protocol addresses the clients and is autodetected, tx-mode is timing. 

You are using HA, check the new option in mqtt settings to avoid HA complains.

I've pushed now the fist build with device-id checking (b13). I think (hope) this should work now. Please test.

The bus-connection in v3.5 is unchanged, but a lot more telegrams are processed. Maybe only after updateing some commands failed while bus in scanned. Check with longer uptime if errors continue.

marcstone-be, 2022-12-22:
Bingo, it's working! As well from the GUI as from HA.
[hc2 13 log_all.txt](https://github.com/emsesp/EMS-ESP32/files/10287425/hc2.13.log_all.txt)

However now I experience very much unstabilities:
In settings, now the board is automatically set on 'BBQKees Gateway S32'. When I change it to MH-ET D1 mini, everything restarts without changing the setting.
Also in system log, when changing e.g. from INFO to ALL, or changing the buffer size, EMS ESP restarts immediately.

I've updated the latest build through OTA. Should I have used EMS-ESP Flasher?

MichaelDvP, 2022-12-22:
> 've updated the latest build through OTA. Should I have used EMS-ESP Flasher?

The usb-flash is only needed if OTA does not work. 

Please restart browser and emsesp, maybe something in browser cached scripts causes the restarts/wrong settings.

MichaelDvP, 2022-12-22:
> However now I experience very much unstabilities:

Confirm. There is something extremly wrong, dashboard give no values or crashes esp, i have no idea what causes this. 
I suggest to use the last official version https://github.com/emsesp/EMS-ESP32/releases.

marcstone-be, 2022-12-22:
It seems to be better, but e.g. when changing the log buffer from 25 to 50, it restarts and the board settings are again put on default BBQKees ESP32. Changing the log detail is no problem anymore, which was the case in the previous build.
It has definitely something to do with the web interface, the previous instable build ran over 3 hours without problem, when not touching the UI.
I also tried flashing with ESP Flasher, different pc's and browsers (Chrome and Edge), all have show the same symptoms.
 I'll let it run overnight without using the webgui, and check if that's stable.

marcstone-be, 2022-12-23:
Goodmorning!
EMS ESP ran for 12 hours stable. The functionalities were all working during this time.
[emsesp_info (4).txt](https://github.com/emsesp/EMS-ESP32/files/10293931/emsesp_info.4.txt)
After a few hours, a new 'clock' device was found. 
But: when setting the slider in the log from 25 to 50, everything crashed again and board was reset to BBQKees ESP32.
Now however I can change it back to D1 Mini without crashing the system.

So I think the only bug now is to be searched in the logbuffer slider.

proddy, 2022-12-23:
ok, thanks for testing. does it happen every time? i.e. changing the log slider from 25 to 50?

MichaelDvP, 2022-12-23:
Thanks for the new system info. The max alloc is at minimum 39k and log-buffer was automatic reduced to 10 entries. I think we should block buffer expansion in this case. I've updated my testbuild.

marcstone-be, 2022-12-23:
OK sliding from 25 to 50 doesn't crash the system anymore :)
It just doesn't have any effect, when clicking away from the log and returning, it's still at 25. Or is this by design in the new fix?

> ok, thanks for testing. does it happen every time? i.e. changing the log slider from 25 to 50?

Yes it was consistently happening every time.

marcstone-be, 2022-12-23:
I'll monitor it further and let you know if I notice something not right.
Very happy with the multiple thermostats feature now working. I can now finally use the physical thermostat that's hanging on the wall for 2 years, in stead of the ESPHOME soft thermostat of HA. (Wife happy)
Thanks for the quick support, you guys rock!

MichaelDvP, 2022-12-23:
> Yes it was consistently happening every time

I think this is fixed,please check.

MichaelDvP, 2022-12-24:
Another fix for weblog-buffer in my testbuild.

marcstone-be, 2022-12-24:
Yes Michael, you nailed it! Tested and works now.

proddy, 2022-12-26:
@marcstone-be close this issue?

marcstone-be, 2022-12-26:
Issue closed, thanks!

commit 9cbb810f (2022-12-24): add device_id to commands, fix #826
closed 2022-12-26 by marcstone-be