emsesp/EMS-ESP32 #828: Create table of all EMS-ESP entities, by device, shortname/fullname and characteristics

proddy, 2022-12-23:
> Thanks, but how can i find the mapping for the version EMS-ESP-3_5_0b10-ESP32 ?

What do you need exactly? The mapping of the old names to the new names?

I'll create a new GH issue to maintain a table of the shortname, fullname etc. We've done a few passes at this but its not consistent

_Originally posted by @proddy in https://github.com/emsesp/EMS-ESP32/issues/804#issuecomment-1363723789_

proddy, 2022-12-23:
My first idea for this is to do it in the code (test.cpp) which will simulate adding all devices and then printing out their details (similar to the API call for each entity).

sciurius, 2022-12-23:
I have the entities for my boiler+thermostat in an md file that I can easily transform into any form you like.

zibous, 2022-12-23:
I have the problem that I have to adjust my automation, scripts and graphics when I use the new version b12. So I'm looking for the mappings so I know which names to change.

|Device            |EMS-ESP-3_5_0b10-ESP32                                     |single instance b12                       |single instance  v3.4|Multi instance b12|FIELD6|FIELD7|
|------------------|-----------------------------------------------------------|------------------------------------------|---------------------|------------------|------|------|
|EMS-ESP           |sensor.system_ems_bus                                      |sensor.system_bus_status                  |                     |                  |      |      |
|                  |sensor.system_free_memory                                  |sensor.system_freemem                     |                     |                  |      |      |
|                  |sensor.system_mqtt_fails                                   |sensor.system_mqttfails                   |                     |                  |      |      |
|                  |binary_sensor.system_ntp_status                            |binary_sensor.system_ntp_status           |                     |                  |      |      |
|                  |sensor.system_rx_fails                                     |sensor.system_rxfails                     |                     |                  |      |      |
|                  |sensor.system_rx_received                                  |sensor.system_rxreceived                  |                     |                  |      |      |
|                  |entity: binary_sensor.ems_esp_status                       |entity: binary_sensor.system_status       |                     |                  |      |      |
|                  |sensor.system_tx_fails                                     |sensor.system_txfails                     |                     |                  |      |      |
|                  |sensor.system_tx_reads                                     |sensor.system_txreads                     |                     |                  |      |      |
|                  |sensor.system_tx_writes                                    |sensor.system_txwrites                    |                     |                  |      |      |
|                  |sensor.system_uptime                                       |sensor.system_uptime                      |                     |                  |      |      |
|                  |sensor.system_uptime_sec                                   |sensor.system_uptime_sec                  |                     |                  |      |      |
|                  |sensor.system_wifi_rssi                                    |sensor.system_rssi                        |                     |                  |      |      |
|                  |sensor.system_wifi_strength                                |sensor.system_wifistrength                |                     |                  |      |      |
|                  |                                                           |                                          |                     |                  |      |      |
|                  |                                                           |                                          |                     |                  |      |      |
|EMS-ESP Boiler    |sensor.boiler_current_flow_temperature                     |sensor.boiler_curflowtemp                 |                     |                  |      |      |
|                  |sensor.boiler_total_uba_operating_time                     |sensor.boiler_ubauptime                   |                     |                  |      |      |
|                  |sensor.boiler_outside_temperature                          |sensor.boiler_outdoortemp                 |                     |                  |      |      |
|                  |sensor.boiler_total_burner_operating_time                  |sensor.boiler_burnworkmin                 |                     |                  |      |      |
|                  |sensor.boiler_burner_starts                                |sensor.boiler_burnstarts                  |                     |                  |      |      |
|                  |sensor.boiler_burner_starts_heating                        |sensor.boiler_heatstarts                  |                     |                  |      |      |
|                  |sensor.boiler_burner_stage_2_operating_time                |sensor.boiler_burn2workmin                |                     |                  |      |      |
|                  |sensor.boiler_burner_current_power                         |sensor.boiler_curburnpow                  |                     |                  |      |      |
|                  |sensor.boiler_flame_current                                |sensor.boiler_flamecurr                   |                     |                  |      |      |
|                  |binary_sensor.boiler_gas                                   |binary_sensor.boiler_burngas              |                     |                  |      |      |
|                  |binary_sensor.boiler_gas_stage_2                           |binary_sensor.boiler_burngas2             |                     |                  |      |      |
[... cut ...]
|                  |??                                                         |sensor.thermostat_hc1_modetype            |                     |                  |      |      |
|                  |??                                                         |sensor.thermostat_hc1_summermode          |                     |                  |      |      |
|                  |??                                                         |sensor.thermostat_wwextra1                |                     |                  |      |      |

zibous, 2022-12-23:
@sciurius 
> I have the entities for my boiler+thermostat in an md file that I can easily transform into any form you like.
Where can if ind the md file ?

MichaelDvP, 2022-12-23:
The customization page shows all registered short and fullnames, For what an extra list?
Or is this issue only HA related? But also HA discovery uses the names that are listed in customization.

zibous, 2022-12-23:
> The customization page shows all registered short and fullnames, For what an extra list?
> Or is this issue only HA related? But also HA discovery uses the names that are listed in customization.

Because changing names will destroy all the statics data in Homeassistant, Automations base on the name do not work...

[image]

So i need the new names to fix this.

sciurius, 2022-12-23:
@zibous 
[rename.zip](https://github.com/emsesp/EMS-ESP32/files/10293997/rename.zip)

proddy, 2022-12-25:
> The customization page shows all registered short and fullnames, For what an extra list? Or is this issue only HA related? But also HA discovery uses the names that are listed in customization.

just to maintain a library of all entities for all devices and their details which we can add to the wiki. People ask what you can read and write and what all the entities mean.

proddy, 2022-12-25:
> Because changing names will destroy all the statics data in Homeassistant, Automations base on the name do not work...
> 
> So i need the new names to fix this.

@zibous if you want to retain your history without using the DB script, just set the Entity ID Format back to the old v3.4 style in the MQTT settings

proddy, 2022-12-27:
I did a rough implementation of this. How it works:

In the code, if compiled with the flag -DEMSESP_STANDALONE_DUMP it will automatically add a default value to an entity. Then a test script goes through all the known devices, adds them and dumps out the information to the Serial port. It's intended to be used once in standalone mode (i.e running on Linux) to create a file. A script basically calls it using shell command like `make clean; make ARGS=-DEMSESP_STANDALONE_DUMP; echo "test dump" | ./emsesp | python3 ./scripts/dump_entities.py > dump_entities.csv`

What is output is the device name, device type, shortname, fullname, type [(enum values) | (min/max)], uom, and the readable, writeable, visible flags. I could later if needed add the Home Assistant entity name. See the example attachments. The CSV file contains (at the time of build) around 3100 rows and can easily be sorted or filtered using Excel/Numbers etc.

I could add a GH action later for this so we always have the latest information on build.

[dump_entities.xlsx](https://github.com/emsesp/EMS-ESP32/files/10309352/dump_entities.xlsx)

[dump_entities.csv](https://github.com/emsesp/EMS-ESP32/files/10309351/dump_entities.csv)

proddy, 2022-12-28:
I added the product_ID and HA entity ID for completeness

MichaelDvP, 2022-12-28:
The thermostat hc specific entities are missing. Mixer shows hc entities but not wwc.

proddy, 2022-12-28:
Oops, missed that. I'll add that.

I'm also looking at ways to show the CSV (or markdown) in a table in the Wiki so people can just look up the data

proddy, 2022-12-28:
think I fixed it now.

[dump_entities.csv](https://github.com/emsesp/EMS-ESP32/files/10316520/dump_entities.csv)

HansRemmerswaal, 2022-12-29:
If you have a spare moment.
You have added the HA entity ID (3.5 version), could you also add the HA entity ID (3.4 version)? That way it's easy to do a mass update in the HA database if someone wants to migrate from 3.4 to 3.5

proddy, 2022-12-29:
> If you have a spare moment. You have added the HA entity ID (3.5 version), could you also add the HA entity ID (3.4 version)? That way it's easy to do a mass update in the HA database if someone wants to migrate from 3.4 to 3.5

sure.

MichaelDvP, 2022-12-29:
> think I fixed it now.

Not complete, the common values are there, but the model specific are missing. It should be `HeatingCircuit(1,model);`

proddy, 2022-12-30:
I've made the changes and added to https://emsesp.github.io/docs/All-Devices/ 

Although the table is sortable and searchable, I haven't found a JS plugin to filter yet.

Closing this and will re-open if changes are needed.

closed 2022-12-30 by proddy