emsesp/EMS-ESP32 #829: Remove the test.mosquitto.org URL in the MQTT settings

bbqkees, 2022-12-23:
In the MQTT settings, by default the host address is test.mosquitto.org.

I often get emails from customers who have issues with MQTT and then the result after some questions about their configuration is that they are actually using test.mosquitto.org as the MQTT broker.

If we remove the default URL here then they are forced to think one step further and fill out the IP of their actual local broker.

proddy, 2022-12-23:
ok, need to see how to do this as the value can't be blank.

bbqkees, 2022-12-23:
I think there are lots of users who are actually using the test server, not just those who send an email with an issue.
So we need to protect them from this insecure solution.

Maybe just use 0.0.0.0?
Or when they tick the 'Enable MQTT' tickbox make the host box empty?

proddy, 2022-12-23:
I have a solution now.

bbqkees, 2022-12-23:
Ok thanks.

closed 2022-12-23 by bbqkees