emsesp/EMS-ESP32 #831: Silent mode - Boiler(0x08) -B-> All(0x00), ?(0xF7)

HansRemmerswaal, 2022-12-23:
Can you please implement the entity for "Silent Mode"?

Below is the log file where I enabled and disabled the silent mode. The telegram is 0xF7.

Silent mode ON:
Boiler(0x08) -B-> All(0x00), ?(0xF7), data: FF 03 84 31 FE FF 0F FD 7F

Silent mode OFF:
Boiler(0x08) -B-> All(0x00), ?(0xF7), data: FF 03 84 31 F6 FF 0F FD 7F

When putting the silent mode to "auto" I was getting the same as "Silent mode OFF", so I guess when putting it to "auto" the "Silent mode ON" will be automatically send somewhere tonight.

Note: I'm having the Nefit Eviline A/W Split 7.0 E-S

┌──────────────────────────────────────┐
│ EMS-ESP version 3.5.0b12             │
│ https://github.com/emsesp/EMS-ESP32  │
│                                      │
│ type help to show available commands │
└──────────────────────────────────────┘

ems-esp:$ set timeout 0
Telnet timeout set to 0 minutes
ems-esp:$ watch 8 F7
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0xF7
000+00:07:16.247 N 0: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0xF7), data: FF 03 84 31 FE FF 0F FD 7F
000+00:07:54.309 N 1: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0xF7), data: FF 03 84 31 F6 FF 0F FD 7F
000+00:07:59.309 N 2: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0xF7), data: FF 03 84 31 FE FF 0F FD 7F
000+00:08:04.359 N 3: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0xF7), data: FF 03 84 31 F6 FF 0F FD 7F
000+00:08:09.309 N 4: [emsesp] Boiler(0x08) -B-> All(0x00), ?(0xF7), data: FF 03 84 31 FE FF 0F FD 7F

proddy, 2022-12-25:
thanks, we can surely add this. Do you know a good description for this 0xF7 telegram?

Maybe we can implement https://github.com/emsesp/EMS-ESP32/issues/150#issuecomment-1306970869 at the same time

HansRemmerswaal, 2022-12-27:
When I change the menu to English it's called "**Low-noise operation**" with the options "No", "Auto" and "On".
When I change the menu to Dutch it's called "Stiller gebruik" with the options "Nee", "Auto" and "Aan"

If the below log the initial status was "Auto". I then changed to "No", "On", "No", "On", "Auto".
Interesting in the the log I do see **HPSilentMode**, but I do not see it coming back as one of the values.

[log silent mode.txt](https://github.com/emsesp/EMS-ESP32/files/10306594/log.silent.mode.txt)

001+03:50:09.558 N 8: [emsesp] Me(0x0B) -R-> Boiler(0x08), HPSilentMode(0x0484), length: 0x19
001+03:50:09.597 N 9: [emsesp] Boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: 01 00 23 2D 0D 50 00 00 00 02 01 00 28 1E 02 02 05 19 0A 0A 03 0D 07 00 0A
001+03:50:09.858 N 10: [emsesp] Me(0x0B) -R-> Boiler(0x08), HPSilentMode(0x0484), length: 0x19 (offset 25)
001+03:50:09.894 N 11: [emsesp] Boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: 01 90 00 F6 28 14 64 00 00 E1 00 1E 00 32 0A 64 01 64 54 20 00 00 (offset 25)

001+03:50:45.087 N 85: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 00 (offset 10)
001+03:50:45.779 N 86: [emsesp] Boiler(0x08) -B-> All(0x00), ?(**0xF7**), data: FF 03 84 31 F6 FF 0F FD 7F
001+03:50:45.985 N 87: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 00 23 2D 0D 50 00 00 00 02 00 00 28 1E 02 02 05 19 0A 0A 03 0D 07 00 0A
001+03:50:46.181 N 88: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 90 00 F6 28 14 64 00 00 E1 00 1E 00 32 0A 64 01 64 54 20 00 00 (offset 25)

001+03:50:53.412 N 106: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 02 (offset 10)
001+03:50:53.755 N 107: [emsesp] Boiler(0x08) -B-> All(0x00), ?(**0xF7**), data: FF 03 84 31 FE FF 0F FD 7F
001+03:50:53.960 N 108: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 00 23 2D 0D 50 00 00 00 02 02 00 28 1E 02 02 05 19 0A 0A 03 0D 07 00 0A
001+03:50:54.157 N 109: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 90 00 F6 28 14 64 00 00 E1 00 1E 00 32 0A 64 01 64 54 20 00 00 (offset 25)

001+03:50:57.838 N 124: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 00 (offset 10)

001+03:50:58.954 N 129: [emsesp] Boiler(0x08) -B-> All(0x00), ?(**0xF7**), data: FF 03 84 31 F6 FF 0F FD 7F
001+03:50:59.160 N 130: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 00 23 2D 0D 50 00 00 00 02 00 00 28 1E 02 02 05 19 0A 0A 03 0D 07 00 0A
001+03:50:59.357 N 131: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 90 00 F6 28 14 64 00 00 E1 00 1E 00 32 0A 64 01 64 54 20 00 00 (offset 25)

001+03:51:03.763 N 149: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 02 (offset 10)

001+03:51:06.979 N 169: [emsesp] Boiler(0x08) -B-> All(0x00), ?(**0xF7**), data: FF 03 84 31 FE FF 0F FD 7F
001+03:51:07.185 N 170: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 00 23 2D 0D 50 00 00 00 02 02 00 28 1E 02 02 05 19 0A 0A 03 0D 07 00 0A
001+03:51:07.457 N 171: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 90 00 F6 28 14 64 00 00 E1 00 1E 00 32 0A 64 01 64 54 20 00 00 (offset 25)

001+03:51:09.658 N 186: [emsesp] Me(0x0B) -R-> Boiler(0x08), HPSilentMode(0x0484), length: 0x19
001+03:51:09.698 N 187: [emsesp] Boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: 01 00 23 2D 0D 50 00 00 00 02 02 00 28 1E 02 02 05 19 0A 0A 03 0D 07 00 0A
001+03:51:09.959 N 188: [emsesp] Me(0x0B) -R-> Boiler(0x08), HPSilentMode(0x0484), length: 0x19 (offset 25)
001+03:51:09.994 N 189: [emsesp] Boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: 01 90 00 F6 28 14 64 00 00 E1 00 1E 00 32 0A 64 01 64 54 20 00 00 (offset 25)

001+03:51:11.475 N 199: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 01 (offset 10)

001+03:51:12.528 N 207: [emsesp] Boiler(0x08) -B-> All(0x00), ?(**0xF7**), data: FF 03 84 31 FE FF 0F FD 7F
001+03:51:12.735 N 208: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 00 23 2D 0D 50 00 00 00 02 01 00 28 1E 02 02 05 19 0A 0A 03 0D 07 00 0A
001+03:51:12.932 N 209: [emsesp] Boiler(0x08) -B-> All(0x00), HPSilentMode(0x0484), data: 01 90 00 F6 28 14 64 00 00 E1 00 1E 00 32 0A 64 01 64 54 20 00 00 (offset 25)

[image]
[image]
[image]

HansRemmerswaal, 2022-12-27:
Related to issue 802

proddy, 2022-12-27:
In #802 Michael added some support, not sure if it enables the low noise operation from the 0x0484 telegram. What you could try yourself is writing to this from the command line manually to see which values work from on/off/auto

MichaelDvP, 2022-12-27:
> Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 00 (offset 10) (No)
Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 01 (offset 10) (Auto)
Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 02 (offset 10) (On)

Thanks, this is a addition to #802, i'll add it.

> And that that the boiler response with the 0xF7 with the HPSilentMode No / On.
Boiler(0x08) -B-> All(0x00), ?(0xF7), data: FF 03 84 31 FE FF 0F FD 7F
Boiler(0x08) -B-> All(0x00), ?(0xF7), data: FF 03 84 31 F6 FF 0F FD 7F

This changes in thermostat the visibility of telegram 484, offset 11 (Min outside temp).
```
Bitmask: 31 F6 FF 0F FD 7F => 10001100 01101111 11111111 00001111 10111111 1111111
Bitmask: 31 FE FF 0F FD 7F => 10001100 01111111 11111111 00001111 10111111 1111111
                                          ^
```

HansRemmerswaal, 2022-12-27:
> This changes in thermostat the visibility of telegram 484, offset 11 (Min outside temp).

Are you sure? I get the impression that I see that 0xF7 each time around the same time that I change the Silent Mode. And each time that FE / F6 in the middle is changing depending on Silent Mode on / off.

proddy, 2022-12-27:
> > This changes in thermostat the visibility of telegram 484, offset 11 (Min outside temp).
> 
> Are you sure? I get the impression that I see that 0xF7 each time around the same time that I change the Silent Mode. And each time that FE / F6 in the middle is changing depending on Silent Mode on / off.

I think the F7 and F8 are only for reading (broadcasting) and the write operations are done via x484

HansRemmerswaal, 2022-12-27:
When the Silent Mode is set to Auto then somewhere in the evening the Silent Mode will be "On" and somewhere in the morning it will be "Off". So maybe the 0xF7 will tell the actual state.

BTW, I do see the Silent Mode now in EMS-ESP with that latest build.

[image]

And in HA

[image]

MichaelDvP, 2022-12-27:
> Are you sure? 

99% sure. The setting telegram is 484 with a length of 47 bytes. The F7 telegram starts with the regarding telegram `FF 03 84`, referencing the 484 telegram (telegram 484 is `<src> <dst> FF <offset> 03 84 ..` and then the bitmask with 47 bits, each bit reference one byte in the 484 telegram.

If you set silentmode to off, byte 10 in 484 changes and bit 11 of F7 change, meaning that the menu entry of minTempSilent (byte 11 in 484) is not shown in thermostat menu. If you change silentmode again ,the entry reappears.

The F9 telegrams are more complicated, i think they are coding the min/max ranges for a menu entry. But i have not understood the scheme.

HansRemmerswaal, 2022-12-28:
We can see the menu setting now, but we can't see when the silent mode is enabled / disabled in case of auto.

My SilentMode is set to auto. Somewhere last night the compressor was put in silent mode (the compressor output is limited, the peak of the 2nd run is less high) but the select.boiler_silent_mode remains auto. So we can't see when the silent mode was enabled and disabled.

[image]

derSchreifritz, 2022-12-28:
as I understand the manual, silent mode (or low noise operation) in auto enables/disables according to a set time schedule and the min temp. If on, its only the min temp.
Thanks for requesting/ adding the on/off/auto switch!

HansRemmerswaal, 2022-12-28:
Yes, but it would be great to see if silent mode is on or off. So I keep searching, there must be a message telling the actual status. But that can also be added later on a new issue.

HansRemmerswaal, 2022-12-28:
Ok, you are right. I captured the full log from 15:38 till 00:22. Somewhere between 19:43 and 22:32 the compressor boiler went into silent mode and there is no 0xF7 in that timeframe. So if there is a message then it must be something different.

Silent Mode = Auto
Run started at 19:43 was not in silent mode
Run started at 22:32 was in silent mode

[log to capture start silent mode when mode is auto.txt](https://github.com/emsesp/EMS-ESP32/files/10316875/log.to.capture.start.silent.mode.when.mode.is.auto.txt)

MichaelDvP, 2022-12-29:
Finding a single bit change within 140000 lines will take longer than this life.

As far as i understood the silentmode switches dependent on outdoortemp and mintemp-setting. 
Try set silentmode on, mintemp low, if pump starts it should be starting in silentmode.
Than set mintemp higher than outdoortemp and silentmode should stop. We have the change of mintemp in log and one of the next telegrams will contain the silentmode indicator.

HansRemmerswaal, 2022-12-29:
That's actually a very good idea, didn't realized that we have the min temp setting.

Starting point:
Silent mode is put to "On"
Min Temp Silent is 20
Outside temperature is around 9.6

Changing Min Temp Silent to 0
Changing Min Temp Silent to 20
Changing Min Temp Silent to 0

[log (7).txt](https://github.com/emsesp/EMS-ESP32/files/10318917/log.7.txt)

MichaelDvP, 2022-12-29:
Maybe changed a bit to fast. I think you need to wait after a change for the next outdoot temperature mesaurement (~ once a minute). Is it possile to check if the pump really actvates silentmode (noise, power, etc.)?

HansRemmerswaal, 2022-12-29:
Starting point:
Silent mode is put to "On"
Min Temp Silent is 0
Outside temperature is around 8.5
Last run was in silent mode

18:00 - Changing Min Temp Silent to 20
18:05 - Changing Min Temp Silent to 0
18:10 - Changing Min Temp Silent to 20
18:15 - Changing Min Temp Silent to 0
18:20 - Changing Min Temp Silent to 10

See attached log. I will later on double check that indeed the next run is not in silent mode.

Around 18:22 the EMS-ESP restarted maybe because of the size of the log?
[image]

[log (8).txt](https://github.com/emsesp/EMS-ESP32/files/10320737/log.8.txt)

HansRemmerswaal, 2022-12-29:
Each time block starting with "Calling command 'boiler/mintempsilent'" I marked with a group number. 
Group 0 was before +/- 18:00 
Group 1 was before +/- 18:05
Group 2 was before +/- 18:10

Next I made a pivot on the message, group number and count of messages. I was looking for messages that only existed in the groups where the silent mode was enabled and messages that only existed in the groups where silent mode was disabled.

Excluding the messages related to setting the MinTempSilent to 0, 10 and 20, I found the following;

29-12-22	17:59:56	DEBUG	4573	[command] Calling command 'boiler/mintempsilent' (min. outside temp. for silent mode) with value 20
29-12-22	18:04:27	DEBUG	5920	[emsesp] No telegram type handler found for ID 0xBF (src 0x38)

29-12-22	18:05:18	DEBUG	6183	[command] Calling command 'boiler/mintempsilent' (min. outside temp. for silent mode) with value 0
29-12-22	18:08:11	DEBUG	6945	[emsesp] No telegram type handler found for ID 0xBF (src 0x10)

29-12-22	18:10:00	DEBUG	7424	[command] Calling command 'boiler/mintempsilent' (min. outside temp. for silent mode) with value 20
29-12-22	18:14:27	DEBUG	8758	[emsesp] No telegram type handler found for ID 0xBF (src 0x38)

29-12-22	18:14:52	DEBUG	8804	[command] Calling command 'boiler/mintempsilent' (min. outside temp. for silent mode) with value 0
29-12-22	18:18:13	DEBUG	9810	[emsesp] No telegram type handler found for ID 0xBF (src 0x10)

29-12-22	18:19:57	DEBUG	1027	[command] Calling command 'boiler/mintempsilent' (min. outside temp. for silent mode) with value 10

It could be a coincident, but it looks like that;
No telegram type handler found for ID 0xBF (src 0x38) is related to silent mode off
No telegram type handler found for ID 0xBF (src 0x10) is related to silent mode on

Else I have no clue...

MichaelDvP, 2022-12-30:
I think the compressor must be running. Change would have effect at 16:34 and 19:40, but not if compressor is off. 
I've checked the telegrams, i'd expect a bit somewhere in telegram 0x48D together with actual compressor power.

HansRemmerswaal, 2023-01-03:
I did several tests with silent mode on / off and compressor running / not running.

I could not find a message telling that the silent mode was enabled. So maybe there is not such a message and does the boiler keeps this internally by just limiting the compressor power.

Closing it for now. At least we have the option itself now to set the silent mode!

closed 2023-01-03 by HansRemmerswaal