emsesp/EMS-ESP32 #833: sensor.boiler_auxiliary_electrical_heater_energy_cons

HansRemmerswaal, 2022-12-24:
**Bug description**
On Dec 19th I upgrade to 3.5.0b12 and have put the Entity ID format to "Single instance, long name (v3.4).
Since then I noticed in Home Assistant that I'm getting the value "sensor.boiler_auxiliary_electrical_heater_energy_cons" instead of "sensor.boiler_auxiliary_electrical_heater_energy_consumption_heating".  Looks like the variable is not big enough?

SELECT *
FROM "states"
where entity_id in ('sensor.boiler_auxiliary_electrical_heater_energy_cons','sensor.boiler_auxiliary_electrical_heater_energy_consumption_heating','sensor.boiler_auxelecheatnrgconsheating')
order by last_updated 

(see screenshots)

**Expected behavior**
Get in Home Assistant
sensor.boiler_auxiliary_electrical_heater_energy_consumption_heating
instead of
sensor.boiler_auxiliary_electrical_heater_energy_cons

**Screenshots**
[image]
[image]

**Device information**
[emsesp_info (1).txt](https://github.com/emsesp/EMS-ESP32/files/10298384/emsesp_info.1.txt)

MichaelDvP, 2022-12-24:
Check MQTT -> Settings -> Entitiy ID Format

HansRemmerswaal, 2022-12-24:
[image]

proddy, 2022-12-25:
yes this is a bug, I need to make the string buffer larger for HA

proddy, 2022-12-25:
I made a PR, being reviewed

HansRemmerswaal, 2022-12-27:
After I installed EMS-ESP-3_5_0-dev_14-ESP32.bin the problem was resolved.

[image]

commit 83438129 (2022-12-26): fixes #833
closed 2022-12-27 by proddy