emsesp/EMS-ESP32 #834: sensor.boiler_auxelecheatnrgconsheating has unknown and NULL since upgrade to 3.5.0b12

HansRemmerswaal, 2022-12-24:
**Bug description**
Since the upgrade to 3.5.0b12 the sensor.boiler_auxelecheatnrgconsheating (and also the longname version) contains NULL and unknown.

**Screenshots**
[image]

**Device information**
[emsesp_info (1).txt](https://github.com/emsesp/EMS-ESP32/files/10298398/emsesp_info.1.txt)

proddy, 2022-12-25:
is this in the MQTT topic, the console, web, HA, ioBroker? Can you be more specific

HansRemmerswaal, 2022-12-25:
This is when I query the SQLite database in HA.

SELECT *
FROM "states"
where entity_id in ('sensor.boiler_auxiliary_electrical_heater_energy_cons','sensor.boiler_auxiliary_electrical_heater_energy_consumption_heating','sensor.boiler_auxelecheatnrgconsheating')
order by last_updated

proddy, 2022-12-27:
and do they show values in the console, web UI or in the MQTT topics? It will help me hunt down where the problem is

HansRemmerswaal, 2022-12-27:
It started since the upgrade to 3.5.0b12, this morning I installed EMS-ESP-3_5_0-dev_14-ESP32 and that corrected the entity_id but still seeing the NULL and unkown.

I see a value in web UI
[image]

But maybe because that is the last value being received in HA
[image]

Below is the export from HA
[states-export.csv](https://github.com/emsesp/EMS-ESP32/files/10307269/states-export.csv)

proddy, 2022-12-27:
I think that is just old state information in HA. If you see null or unknowns in the WebUI, Console or MQTT then we have a problem!

HansRemmerswaal, 2022-12-27:
The last entries are from 20 min ago.
I have deleted all the NULL and unknown from the HA database and will keep an eye on it.

[image]

HansRemmerswaal, 2022-12-27:
1) From HA database I have deleted from "states" the values NULL and unknown.
2) Restarted EMS-ESP module
3) In HA database I now see the NULL and unknown again

Looking at the timings the values are only NULL and unknown for second, so difficult to see them in the WebUI.
Is there a way that I can capture and share the MQTT messages with you?

[image]

proddy, 2022-12-27:
the entities are created using MQTT Discovery. Use something like MQTTExplorer and search for the config topic for this 'energy_consumption_heating' and it will point to the boiler_data MQTT topic and which json key it's pulling the data from. Then still using MQTTExplorer watch boiler_data and see if the value appears or disappears. EMS-ESP will only create the HA entities when there is a valid value, and if it not then it will quickly try and remove the HA config topic so its goes from HA. Otherwise HA will keep complaining. 

Also, look at the HA logs. You can increase HA's logging for the MQTT integration which will show you more detailed logs. 

I'll also look into the code to see if something is odd

HansRemmerswaal, 2022-12-27:
I just installed MQTT Explorer, let's see if I can capture something.

But I wonder...

When I restart the EMS-ESP then I'm getting these NULL and unknown also. In HA I then see the sensor.ems_esp_status disconnected / connected. I see many disconnected / connected, also when I'm not playing with the system. Around the same time I see these NULL and unknown. Maybe there is a link.

From HA:

SELECT *
FROM "states"
where entity_id = 'sensor.ems_esp_status'
order by last_updated

[image]

proddy, 2022-12-27:
I see null values too when querying the DB, but this may be left over residue from previous versions where the states haven't been cleaned up. For example `ems_esp_status` was renamed to `ems_esp_system_bus_status`.

[image]

I'll keep tracking it and see what and why this is happening. Apart from the DB, are you seeing strange readings in HA?

HansRemmerswaal, 2022-12-27:
> Apart from the DB, are you seeing strange readings in HA?
Apart from this, which started when I upgrade to 3.5.0b12, I do not see strange readings when selecting all devices in HA. All kind of patterns looks similar, some examples shown in the first 2 screenshots.

But having said that, I do notice that for some sensor values it's being split, see screenshot 3-5. Need to dig into this.

[image]
[image]

**WIFI strength**
[image]

**WIFI RSSI**
[image]

**Free memory**
[image]

HansRemmerswaal, 2022-12-27:
SELECT distinct entity_id
FROM "states"
where entity_id like '%WIFI%'

[image]

proddy, 2022-12-27:
to be honest I'm not seeing anything strange in that data you shared from HA. Do you have an example of a sensor that is showing incorrect values in HA (not quering the db).

HansRemmerswaal, 2022-12-28:
The first 2 screenshots show how I checked on high level for strange values. And all looks ok, same kind of patterns and no values that suddenly stop. 

But I did found 3 values which are "wrong", for those I have raised #850.

With respect to the original issues, unknown and NULL since upgrade to 3.5.0b12. I noticed that the EMS-ESP often lost the connection, maybe due to a restart, and each time it was connected again the unknown and NULL where recorded in the HA database for the sensor values. Not sure if EMS-ESP is sending these or that HA is automatically generating them. But since the last build of EMS-ESP-3_5_0-dev_14-ESP32 I have not seen that lost connection anymore and therefore also not that NULL and unknown anymore (apart from the times that I restarted the EMS-ESP myself).

So it could be that this issue has been resolved as a side effect of the other defects which has been resolved recently.

From my point of view we can close this issue, if I see these restarts again then I can raise a specific issue for that one.

closed 2023-02-19 by proddy