emsesp/EMS-ESP32 #835: Unrecognized EMS device (device ID 0x09, no product ID)

stmb-100, 2022-12-24:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/emsesp/EMS-ESP32/issues) (both open and closed)*
* *searched the [documentation help section](https://emsesp.github.io/docs)*

*Completing this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api/system*

**Additional context**
*Add any other context about the problem here.*

MichaelDvP, 2022-12-24:
@stmb-100 Please add the system info from help page. A controller without product-id is very uncommon. There is also a temininal attached to the bus producing inclomplete telegrams (timing?). Maybe the sender 09 is misinterpreted from 0A because of timing differs >9%.
A complete log (all) of about 1-2min would also help to understand what's going on..

stmb-100, 2022-12-25:
[emsesp162_info.txt](https://github.com/emsesp/EMS-ESP32/files/10299905/emsesp162_info.txt)

stmb-100, 2022-12-25:
Hi now i changed the busID to Gateway 1 or 2 or 3. the it works shortly, i get the Boiler GBx72/Trendline/Cerapur/Greenstar Si/27i, but the busid jump back to service key by self. 
now i set the busid to converter

[emsesp162_info.txt](https://github.com/emsesp/EMS-ESP32/files/10299977/emsesp162_info.txt)
[log162.txt](https://github.com/emsesp/EMS-ESP32/files/10299978/log162.txt)

MichaelDvP, 2022-12-25:
Last system info looks ok. You have aready termnal, service-key and gateway connected. With a bus collision ems-esp can not work and there will no answer from system/ems-devices. Maybe disconnect the bus, configure ems-esp properly and then connect the bus again. 

BTW: A debug-log does not help, i ask for a log (all) with bus communication included. But if now alls ems-devcies are recognized it not needed anymore.

cyclops1982, 2023-01-12:
I have a bosch/worcester greenstar 32CDI and see the same message.
Is that device supported and thus there is a timing issue?

MichaelDvP, 2023-01-12:
@cyclops1982  **Please add the support info from help page.**

cyclops1982, 2023-01-12:
> @cyclops1982 **Please add the support info from help page.**
```
{
  "System Status": {
    "version": "3.4.4",
    "uptime": "000+01:38:18.694",
    "freemem": 115,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -64,
    "IPv4 address": "192.168.4.15/255.255.255.0",
    "IPv4 gateway": "192.168.4.1",
    "IPv4 nameserver": "192.168.1.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/London"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 2355,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 0,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 1179
  },
  "API Status": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected, tx issues - try a different Tx Mode",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 1642,
    "bus reads (tx)": 0,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 8,
    "bus reads failed": 88,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 0
  },
  "Settings": {
    "board profile": "LOLIN",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 17,
    "tx gpio": 16,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "generic boiler",
      "device id": "0x08",
      "product id": 0,
      "version": "00.00",
      "entities": 38,
      "handlers received": "0x1C 0x18 0x19 0x34 0x2A",
      "handlers pending": "0xBF 0x10 0x11 0xC2 0x14 0x15 0x1A 0x35 0x16 0x33 0x26 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x0494 0x0495 0x048D 0x048F 0x048A"
    }
  ]
}```

MichaelDvP, 2023-01-12:
Seems the board profile "Lolin" is wrong, tx is not working. Try "S32".

cyclops1982, 2023-01-12:
> Seems the board profile "Lolin" is wrong, tx is not working. Try "S32".

Which option in the dropdown would that be?

Also, what protocol should i be using with my boiler (bosch/worcester greenstart 32cdi)?

MichaelDvP, 2023-01-12:
> Which option in the dropdown would that be?

I mean: "BBQKees Gateway S32" or "NodeMCU32"  (is identical), 
What Gateway do you use? If it is homebuild, send a photo from gateway and esp32-gateway connection.
For "protocol" leave it on EMS, the options only have a bit different timing on tx, normally option EMS works good. You can test what option gives the less tx errors.
I expect this boiler as product-id 95, but maybe something other, let's see it what it report on version info.

cyclops1982, 2023-01-12:
Both S32 options don't work, and LOLIN D32 seems to work best... as it is a lolin D32 after all :-)

Here are the pictures:
[image]
[image]

MichaelDvP, 2023-01-13:
Oh, you have connected your own scheme. Then you have to use boardprofile "custom" and set the gpios as you have connexted:
Rx: 17, Tx 18

cyclops1982, 2023-01-13:
Ah, i thought i had the right pins based on the source code. Thanks. Now TX also works.

I was wondering - on the console with `show ems` the _bus protocol_ states HT3. Should i set it to that?

```
ems-esp:$ show ems
EMS Bus is connected.

EMS Bus info:
  Tx mode: 1
  Bus protocol: HT3
  #recognized EMS devices: 2
  #telegrams received: 76
  #read requests sent: 21
  #write requests sent: 0
  #incomplete telegrams: 0
  #read fails (after 3 retries): 0
  #write fails (after 3 retries): 0
  Rx line quality: 100%
  Tx line quality: 100%

Rx Queue is empty

Tx Queue is empty
```

closed 2023-01-18 by proddy