emsesp/EMS-ESP32 #841: console command crashes EMS-ESP

proddy, 2022-12-26:
```
ems-esp:# call thermostat hc1 mode eco

Connection closed by foreign host
```

The entire device was just restarted automatically.

_Originally posted by @akanto in https://github.com/emsesp/EMS-ESP32/issues/773#issuecomment-1365040363_

MichaelDvP, 2022-12-26:
I'm not sure what is the best fix. 
Because the commandline only uses a prefix with no command a nullptr is returned here: https://github.com/emsesp/EMS-ESP32/blob/d08a1224e2df894b937bbe575faf8f53faae8e5c/src/command.cpp#L228-L231
It does not crash if we return the empty string instead.

The crash happens with nullptr here: https://github.com/emsesp/EMS-ESP32/blob/d08a1224e2df894b937bbe575faf8f53faae8e5c/src/command.cpp#L253
in this strcmp: https://github.com/emsesp/EMS-ESP32/blob/d08a1224e2df894b937bbe575faf8f53faae8e5c/src/emsdevice.cpp#L280
We can check for nullptr or use std::string instead of const char* (like in is_readonly here: https://github.com/emsesp/EMS-ESP32/blob/d08a1224e2df894b937bbe575faf8f53faae8e5c/src/emsdevice.cpp#L636)

But with other commandlines like `call thermostat hc1`  it also crashes here: https://github.com/emsesp/EMS-ESP32/blob/d08a1224e2df894b937bbe575faf8f53faae8e5c/src/command.cpp#L265

I think an easy fix is inserting at the start of Command::call: 
```
    if (cmd == nullptr) {
        return CommandRet::NOT_FOUND;
    }
```
and the critical (crashing) functions are not called.

proddy, 2022-12-27:
thanks for hunting that one down. Yes, the easiest fix is as you suggested. Do you mind fixing it?

proddy, 2022-12-30:
seeing similar behaviour with testing the mixer

```
ems-esp:# call mixer info
{
  "wwc1 current temperature": 56.5,
  "wwc1 pump status in assigned wwc (PC1)": "off",
  "wwc1 temperature switch in assigned wwc (MC1)": 0,
  "wwc2 current temperature": 62.0,
  "wwc2 pump status in assigned wwc (PC1)": "off",
  "wwc2 temperature switch in assigned wwc (MC1)": 0,
  "hc1 mixing valve actuator (VC1)": 0,
  "hc1 setpoint flow temperature": 0,
  "hc1 pump status (PC1)": "off"
}

ems-esp:# call mixer hc1 info
make: *** [run] Segmentation fault: 11

```

MichaelDvP, 2022-12-30:
Hmm.
```
ems-esp:# call mixer hc1 info
Unknown command
Available commands are: activated commands entities flowsettemp info pumpstatus valvesettime

ems-esp:# call mixer hc2/info
{
  "Vorlauftemperatur HK (TC1)": 45.5,
  "Mischerventil Position (VC1)": 48,
  "Sollwert Vorlauftemperatur": 45,
  "Pumpenstatus HK (PC1)": "an",
  "Aktiviert": "an",
  "Zeit zum Einstellen des Ventils": 120
}

ems-esp:# call mixer info 2
{
  "Vorlauftemperatur HK (TC1)": 45.5,
  "Mischerventil Position (VC1)": 48,
  "Sollwert Vorlauftemperatur": 45,
  "Pumpenstatus HK (PC1)": "an",
  "Aktiviert": "an",
  "Zeit zum Einstellen des Ventils": 120
}

ems-esp:# call mixer info
{
  "HK2 Vorlauftemperatur HK (TC1)": 46.2,
  "HK2 Mischerventil Position (VC1)": 45,
  "HK2 Sollwert Vorlauftemperatur": 45,
  "HK2 Pumpenstatus HK (PC1)": "an",
  "HK2 Aktiviert": "an",
  "HK2 Zeit zum Einstellen des Ventils": 120
}

```

proddy, 2022-12-30:
I'm running from standalone simulating a mixer.  Maybe that's the issue

MichaelDvP, 2022-12-30:
No, i'm wrong, it crashes, my version was different ,i've tested with this https://github.com/emsesp/EMS-ESP32/blob/6c09134552271663576003d90951f36dd695b136/src/command.cpp#L229-L231
removed to give empty string back.

With nullptr returned it crashes here in strcmp https://github.com/emsesp/EMS-ESP32/blob/6c09134552271663576003d90951f36dd695b136/src/console.cpp#L371

commit 28de5bb0 (2022-12-27): fix #841
commit ba03add3 (2022-12-27): Merge pull request #842 from MichaelDvP/dev

fix #841
commit 34f6b412 (2023-01-05): fix #841, `call <device> hcx` show info