emsesp/EMS-ESP32 #843: NTP status "inactive"

HansRemmerswaal, 2022-12-27:
**Question**
On the EMS-ESP I always see the "NTP Status" as "Inactive" and the local time is incorrect. So I then use the "SET TIME" option to correct the time. How can I make the NTP Status working?

**Screenshots**
1) Shortly after reboot
[image]

2) Few minutes later after refresh
[image]

3) NTP settings
[image]

**Device information**

HansRemmerswaal, 2022-12-27:
After using the "SET TIME" option after the reboot the date / time stays correct. But with the next reboot the time is wrong again.

proddy, 2022-12-27:
I'm not sure why it would say Inactive. Did it work before on previous versions? Is your network blocking this perhaps? 

I tried the SET TIME, rebooted 3 times and it keeps the settings, so can't reproduce that one. Maybe Michael has an idea

MichaelDvP, 2022-12-27:
The esp32 rtc does not have a buffer-battery, after powercut the time is reset. 
I've tried different servers (time.google.com, europe.pool.ntp.org, de.pool.ntp.org, all working. But not working if i block ems-esp from internet access in the router. I have not tested, but i think it also does not work in guest network (only web/mail allowed).

My router (fritzbox) is capable to act as timeserver, I always use this for all internal devices, makes sure that all have identical time with low latency.

HansRemmerswaal, 2022-12-27:
Since I got my module from BBQKees, with 3.5 installed, I have this issue. 
From Windows I can do a ping to time.google.com.

[image]

MichaelDvP, 2022-12-27:
What do you want to compare? ICMP is not NTP, Your windows PC is maybe handled different by your firewall.
Your emsesp can not connect to time.google.com with NTP protocol on UDP, port 123. Make sure that this connection is allowed.

proddy, 2022-12-27:
I would also recommend to use the NTP server that comes with your router/modem (unifi, fritzbox etc)

HansRemmerswaal, 2022-12-27:
Interesting, I can connect to the EMS-ESP module (192.168.0.239) but the module is not visible in my router (DECO M5) in the list of connected devices.

HansRemmerswaal, 2022-12-29:
Connected the EMS-ESP module via a network cable, still not visible in DECO M5, but now the NAT status is active. So somehow related to connected via WIFI.

HansRemmerswaal, 2022-12-29:
Resolved... 

In the network settings I have defined a fixed IP address and defined the DNS 8.8.8.8 and 8.8.4.4.
Since then the NTP status is active.

closed 2023-02-19 by proddy