emsesp/EMS-ESP32 #844: EMS-ESP-3_5_0-dev_14-ESP32 - Problem loading (507) during "support info"

HansRemmerswaal, 2022-12-27:
**Bug description**
I have installed EMS-ESP-3_5_0-dev_14-ESP32. 
When I'm using the Help -> Support info I'm getting the "Problem loading (507) error and nothing is exported.

**Steps to reproduce**
Install EMS-ESP-3_5_0-dev_14-ESP32 and use the option Help -> Support info

**Expected behavior**
Export file

**Screenshots**
[image]

[image]

**Device information**
And older one...
[Uploading emsesp_info (1).txt…]()

**Additional context**
N/A

MichaelDvP, 2022-12-27:
@proddy 
I'll add a commit to the actual PR to avoid the 507 by reducing the buffer stepwise until it fits in memory. A few less users will complain that dashboard/customization or system-info will cut the last values. But that's a nice variety to all the 507 complains.

MichaelDvP, 2022-12-27:
Sorry, first commit has a copy/past error, dahboard and customization throw errors. Fixed in a second PR.

HansRemmerswaal, 2022-12-27:
Issue has been resolved for me with the latest build of EMS-ESP-3_5_0-dev_14-ESP32

[image]

closed 2022-12-27 by proddy