emsesp/EMS-ESP32 #846: Compress 7000AW missing sensors

sfbia2m, 2022-12-27:
Hi, 
I have been using your interface for some time now, but in one of the updates after Sep 13 2021. I lost two temperature sensores TR5 and TR6, is it possible to bring that to live again?
Also my homeassistant logs has a lot of warnings like this "Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.hc2.id}}'" this is also an isue witch started after Sep13 2021

I know the date because at that time you got the sommer mode to work for me :-)

I have installed the current firmware v3.5.0-dev.14 

[image]

MichaelDvP, 2022-12-27:
Please check in terminal `read 8 48F` and write down the actual values or TR5 and TR6.
The sensors were read from TR5: offset 20, TR6: offset 0, but there was a complain that the values are wrong. So we do not know where to read the values. If you could help to identify the tight offsets, i'll add them back.
> 'dict object' has no attribute 'id' when rendering '{{value_json.hc2.id}}'" this is also an isue witch started after Sep13 2021

https://emsesp.github.io/docs/#/Troubleshooting?id=home-assistant

sfbia2m, 2022-12-27:
Hi Michael,

Here is the result from mine. TR5 shows 1.2 and TR6 show 62.4. The values keeps changing at the moment so I home you can see them in the telegram

ems-esp:$ read 8 48f
000+08:36:05.906 I 2: [emsesp] Boiler(0x08) -W-> Me(0x0B), HpTemperatures(0x048F), data: 02 73 01 5E 01 61 01 19 00 00 00 00 00 08 02 33 01 45 00 00 00 0A 00 80
000+08:36:06.549 I 3: [emsesp] Boiler(0x08) -W-> Me(0x0B), HpTemperatures(0x048F), data: 02 6D FF D8 01 58 (offset 24)

sfbia2m, 2022-12-28:
Thanks it works now :-)

proddy, 2023-01-22:
in dev-16

commit 14f3d9ab (2022-12-27): Merge pull request #849 from MichaelDvP/dev

fix #846, #848, remove double values hpsuctionGas/hpHotGas, stick to hpTr5/hpTr6
closed 2023-01-22 by proddy