emsesp/EMS-ESP32 #847: Problem Loading (Extra x of y bytes(s) found at buffer[1])

akanto, 2022-12-27:
**Bug description**
When I click on the thermostat or boiler entries I get the "extra ... of ... bytes(s) found at buffer[1]" error message. I am using the latest 3.5 b14 version of the firmware.

**Steps to reproduce**
I have opened the network tab in chrome, and as you can see on the screenshot, the data is correctly returned from the backend and it seems that it is a frontend/javascript issue.

It is reproducible from Chrome and Safari as well.

I have seen a couple of "Problem Loading" github issues, but none of them are describing this exact error.

[image]

MichaelDvP, 2022-12-27:
You have catched a buggy, intermediate version, update again.

akanto, 2022-12-27:
Thanks, it looks good now.

MichaelDvP, 2022-12-27:
So close this?

akanto, 2022-12-27:
Yes, please. We can close this.

Thanks for the help.

closed 2022-12-27 by akanto