emsesp/EMS-ESP32 #848: Sensor TR5 and TR6

HansRemmerswaal, 2022-12-27:
**Question**
In the WebUI we have the TR5 and TR6 but for me no value is being shown.

[image]

But when I check on the display on my heatpump I do see values for these (pictures are available if needed).

[image]
The following log captures the period 15:40 - 15:48
[log (4).txt](https://github.com/emsesp/EMS-ESP32/files/10308829/log.4.txt)

Checking the WebUI I'm getting the impression that **suction gas** and **hot gas/compressed** are actually containing the TR5 and TR6 values.
[image]

It looks like that 
**compressor inlet temperature (TR5)** should contain the value from "Suction gas"
**compressor outlet temperature (TR6)** should contain the value from "Hot gas/compressed"

**Screenshots**
[image]

**Device information**
[emsesp_info (3).txt](https://github.com/emsesp/EMS-ESP32/files/10308833/emsesp_info.3.txt)

**Additional context**
Using EMS-ESP-3_5_0-dev_14-ESP32.bin

MichaelDvP, 2022-12-27:
Same is already in #846. 
The values were double. I think we stick to Tr5/Tr6 and delete the hpSuctionGas/HotGas, as hybrid Heatpumps also use the Tr5/Tr6 naming. Memory is low having2 entities and names for the same value is really not usefull.

sfbia2m, 2022-12-27:
Hi Michael,

Here is the result from mine. TR5 shows 1.2 and TR6 show 62.4. The values keeps changing at the moment so I home you can see them in the telegram

ems-esp:$ read 8 48f
000+08:36:05.906 I 2: [emsesp] Boiler(0x08) -W-> Me(0x0B), HpTemperatures(0x048F), data: 02 73 01 5E 01 61 01 19 00 00 00 00 00 08 02 33 01 45 00 00 00 0A 00 80
000+08:36:06.549 I 3: [emsesp] Boiler(0x08) -W-> Me(0x0B), HpTemperatures(0x048F), data: 02 6D FF D8 01 58 (offset 24)

HansRemmerswaal, 2022-12-28:
Great!

[image]

closed 2023-02-19 by proddy