emsesp/EMS-ESP32 #855: Web FW update Error 507 & OTA update not possible

SpaceTeddy, 2022-12-28:
**Bug description**
Not possible to update the EMS-ESP32 firmware in Web interface (upload/download). OTA update with espota.py tool is not possible as well. Any idea to get the FW updated w/o using serial connection? thx

**Steps to reproduce**
Installed FW 3.5.0b8
'''EMS-ESP Version
v3.5.0b8

Device (Platform / SDK)
ESP32 / v4.4.2

System Uptime
000+00:07:29.085

CPU Frequency
160 MHz

Heap (Free / Max Alloc)
128 KB / 79 KB

Flash Chip (Size / Speed)
4.096 KB / 40 MHz

Application (Used / Free)
1.887 KB / 33 KB

File System (Used / Free)
20 KB / 172 KB'''

**Expected behavior**
update FW version to 

**Screenshots**
Error 507
[image]

OTA update:
[image]

MichaelDvP, 2022-12-28:
Your ota partition is only 1920 kb, the filesystem 192 kb. You need to flash with ems-esp partiton table using USB and esptool or https://github.com/emsesp/EMS-ESP-Flasher.

SpaceTeddy, 2022-12-29:
alright, I used the [EMS-ESP-Flasher](https://github.com/emsesp/EMS-ESP-Flasher) and it works out of the box. After reflashing, the Web update worked. 
Thanks for giving the hint with the partitions.
-> closed

closed 2023-02-19 by proddy