emsesp/EMS-ESP32 #857: ahs / am200: web-api needs corrections

tp1de, 2022-12-28:
I got a GH issue in respect to my ioBroker adapter (https://github.com/tp1de/ioBroker.ems-esp/issues/12) opened by @tglynx.
The ioBroker Adapter is polling ems-esp by using web-api calls.  
I could find the following issues which need corrections:

1.  the  http://ems-esp/api/system command returns two times the device boiler. One for the boiler and one for the alternative heat source ahs. Device name should be unique either "boiler" and "boiler2" or even better using device name "ahs" for an am200 device
2. an http://ems-esp/api/boiler  returns the "ahs" as a substructure within json - I adjusted my ioBroker to get values. Better to get the ahs values by polling http://ems-esp/api/ahs
3. Reading single ahs values e.g. http://ems-esp/api/boiler/ahs/altflowtemp do return an error: { "message": "cannot find values for entity 'ahs/altflowtemp'" }. So single field attributes can't be read.

From my point of view I recommend to adjust.

MichaelDvP, 2022-12-28:
Please ask the user to participate here and post the system info. 
We can setup a own device-class ahs, but need someone with this device to test and give feedback.
Or only fix the prefix `ahs/` in point 3?

tglynx, 2022-12-29:
Hi, 

i am here and ready to test and give feedback. 

I think it would be the right way to implement a separate device class for the am200 module - as for mixers (MM100 and so on) or solar modules (SM100...).
However, it is a heat source module ... the alternative heat source module can be used to directly control a pellet stove with target temperature (or just a simple wood stove) and has the ability to block the primary heat source for a given time.

As i can see the implementation that has been done already also happened on user request and i really appreciate all the work you put into this. Great project!

MichaelDvP, 2022-12-29:
Yes, it's better to create a own device class. The boiler have a lot of entities making problems with buffer sizes.
I'll call the class `heatsource` and move the am200. 
For now we know only the AM200 with device-id 0x60, i'll add it unnested. If more modules are possible we will need to rename with tag HS1... 
The AM200 can be used as single boiler, pos 9/10. I think then it will use devcie_id 8 as a boiler and other telegrams. We have to check if someone uses this.

Boilers connected to a cascade module will still go to boiler class with tag HS1.., there are only a few values and the boilers keep there product-id, only changes device-id to 70...

I'll make a pr. @proddy please check if you agree that this is the right way.

MichaelDvP, 2022-12-29:
@tglynx Please check https://github.com/MichaelDvP/EMS-ESP32/releases

tp1de, 2022-12-29:
@MichaelDvP  what will be the device name within device-list of http://ems-esp/api/system ?

tglynx, 2022-12-29:
Hi,

installed that firmware and it is working!

[image]

From the log:
```
2022-12-29 13:13:20.989 D 123: [emsesp] Adding new device Alternative Heatsource (deviceID 0x60, productID 228, version 03.02)
2022-12-29 13:13:20.989 I 126: [emsesp] Recognized new heatsource with deviceID 0x60
```

I can also confirm that the web API now returns details for the entities of the heatsource.

http://ems-esp/api/heatsource/altflowtemp
```
{
  "name": "altflowtemp",
  "fullname": "alternative hs flow temperature",
  "value": 66.9,
  "type": "number",
  "uom": "°C",
  "readable": true,
  "writeable": false,
  "visible": true
}
```

So i would say it looks good ... 

@tp1de 
It also comes over very nicely in the ioBroker Adapter (in native and km200 structure mode).

[image]

[image]

In KM200 mode my personal preference would be that the AM200 values would come up as hs2 under heatSources.

Here is what http://ems-esp/api/system returns

```
{
  "System Info": {
    "version": "3.5.0-dev.14",
    "platform": "ESP32",
    "uptime": "000+00:15:40.323",
    "uptime (seconds)": 940,
    "free mem": 98,
    "max alloc": 38,
    "free app": 5,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "94:B5:55:FF:DF:D7",
    "IPv4 address": "192.168.123.146/255.255.255.0",
    "IPv4 gateway": "192.168.123.254",
    "IPv4 nameserver": "192.168.123.254",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
[... 167 lines cut ...]
      "product id": 220,
      "version": "01.05",
      "entities": 0
    },
    {
      "type": "Heatsource",
      "name": "Alternative Heatsource",
      "device id": "0x60",
      "product id": 228,
      "version": "03.02",
      "entities": 23,
      "handlers received": "0x054D 0x054E 0x0550",
      "handlers fetched": "0x054C",
      "handlers pending": "0x054F",
      "handlers ignored": "0xBF"
    }
  ]
}
```

For the EMS-ESP Dashboard / Devices & Sensors i would suggest that the Heatsource also gets the 'Boiler' icon and that it be named 'AM200 Alternative Heatsource'.

What else can we check? I think not all entities are identified already, is that correct? 

Great work! Thanks...

tp1de, 2022-12-29:
@tglynx 
Hi Michael I have changed the ioBroker adapter so that the "heatsource"  is translated to "heatSources/hs2" while using km200 structure mode. (as before).  It's on GH to be tested.

tglynx, 2022-12-29:
@tp1de 
Hi Thomas, can confirm that this is working!

[image]

MichaelDvP, 2022-12-29:
> For the EMS-ESP Dashboard / Devices & Sensors i would suggest that the Heatsource also gets the 'Boiler' icon and that it be named 'AM200 Alternative Heatsource'.

Ok, but i shorten the name to "AM200", all devices are nemd only RC300, i700, etc. the extrawas only because it wassorted as boiler but isn't one.

I've looked at #573, where we added the AM200. @OlliHein, please also check the changes in https://github.com/MichaelDvP/EMS-ESP32/releases If you are using homeassistant there will be the entiies changed. Sorry, but the class change can't be done without HA changings.

@tp1de If using a cascade module the boiler will have nests with boiler/hs1/ .. boiler/hs16/, make sure that will not conflict.

tp1de, 2022-12-29:
>If using a cascade module the boiler will have nests with boiler/hs1/ .. boiler/hs16/, make sure that will not conflict.

good point @MichaelDvP ... I was already thinking about it. How can I identify the nested boilers and how many there are?
If there are nested boilers, will there be then more tham one device entry boiler in the device-list? If yes, wouldn't it be better to give unique names ... e.g. boiler1 ... n ?

Or might it be easier to name the ahs module just "hsa?". @tglynx what about "hsa" instead of fixed "hs2"?

tglynx, 2022-12-29:
> Or might it be easier to name the ahs module just "hsa?". @tglynx what about "hsa" instead of fixed "hs2"?

sure, i think it is just a matter of preference as the original KM100/KM200 gateways do not support the AM200 module.

MichaelDvP, 2022-12-29:
> How can I identify the nested boilers and how many there are?

Mainly the Cascade MC400 is a single boiler to the bus with all entities. The bus supports only one boiler. For the connected boilers there is a new ems-bus for each and the MC400 is thermostat for  these busses. Only a few telegrams are forwarded in a kind of NAT (network translation), mapping the real boilers to device-id 0x70, 0x71, ...

Example: We have in database:
```
{208, DeviceType::BOILER, "Logamax Plus/GB192/Condens GC9000/Greenstar ErP", DeviceFlags::EMS_DEVICE_FLAG_NONE},
{210, DeviceType::BOILER, "Cascade MC400", DeviceFlags::EMS_DEVICE_FLAG_NONE},
```
and someone connects one MC400 and four GB192/20kW
Now the MC400 is one boiler with 80kW and depending on heat demand it starts one/two/tree/four GB192 boilers.
In web and system info the boilers show up seperatly, Boiler MC400 with device-id 0x08, Boiler GB192 with device-id 0x70, etc.
For the bus we have now in `/api/boiler` or mqtt boiler:
```
{
 "curflowtemp": 70.3,
 "selflowtemp": 52,
 "outdoortemp": 8.4,
 "boiltemp": 70.8,
 "wwsettemp": 48,
 "wwseltemp": 48,
 "wwtype": 3,
 .. all other boiler values...,
 "hs1": {
   "curflowtemp": 70.3,
   "selflowtemp": 52,
   "curburnpower": 50,
   "selburnpower":50,
   "burnworkmin": 12345
 },
 "hs2": {
   "curflowtemp": 0,
   "selflowtemp": 0,
   "curburnpower": 0,
   "selburnpower":0,
   "burnworkmin": 12000
 }
 "hs3": {
   "curflowtemp": 0,
   "selflowtemp": 0,
   "curburnpower": 0,
   "selburnpower":0,
   "burnworkmin": 11001
 }
 "hs4": {
   "curflowtemp": 0,
   "selflowtemp": 0,
   "curburnpower": 0,
   "selburnpower":0,
   "burnworkmin": 205
 }
}
```
The hs1..hs4 give only these 5 stati and can not be written directly, it's just info. All commands go to MC400.
The MC400 can be cascaded to support 16 heatingsources.

MichaelDvP, 2022-12-31:
@proddy Is this way ok? 
We can also move the few entities of hs1..hs16 to heatsource class. But then we should also make a ahs-nest/tag for the alternative heatsource. We can do this every time, but the tag will change the HA entities, so better decide now.
Unclear is, if there are more than one ahs possible (id 0x60, 0x61...). Use tag `ahs1` to reduce futur changes?
@tglynx  What would you prefere?

tp1de, 2022-12-31:
@MichaelDvP 
How would it look like on Hybrid Heating systems with gas boiler, heat pump and hm200 hybrid interface device?

proddy, 2022-12-31:
@MichaelDvP I'll go with whatever you think is the right way forward is. I haven't looked at your new code yet, but is the heatsource data being pulled into the boiler_data MQTT topic too?

tp1de, 2022-12-31:
Just to mention: 
A RC310 thermostat controlled heating system can support:
- an alternative heatsource (or more?) with am200
- a hybrid system with hm200
- a cascaded boiler (but quite exotic)
- 2 dhw systems (dhw1 and dhw2) with or without mixer
- up to 4 heating circuits with or without mixer 

Such a system with multiple / hybrid heat sources will have global (system-wide) fields/values as well as entries per heat source.
As far as I can see multiple dhw are not supported yet. The mixers are then linked to dhw or hc.

MichaelDvP, 2022-12-31:
> but is the heatsource data being pulled into the boiler_data MQTT topic too?

That's the point, for mqtt we can go to single topics `boiler_data_hs1` ,etc. but `api/boiler/info` grows.

I think it's better to move. Mixers also have only tagged entities, than we have heatsources always tagged with ahs1, hs1, hs2, etc. I'll stick to ahs1, so we can more if there are any.

@tp1de The hybrid system HM200 is listed as heatpump. Mixers could control hc1-8 or wwc 1-10, Poolmodule (PM100) or zonemodule MZ100. For RC300 some known dhw2 entities are already mapped, but nobody seems to use or miss it, so we don't know the telegrams/offsets.

tp1de, 2022-12-31:
@MichaelDvP I wonder if I understand correctly:

Heatsources are boilers, alternative heatsources (am200), heat pumps with or without hybrid manager (hm200):
Will then be boilers and am200  within boilers and heatpumps and hm200 within heatpumps? - Even if all are heatsources and am200 not boilers?

proddy, 2022-12-31:
I do like the generic term "heatsource". The word 'boiler' was back when I created the first version of this project in 2017. If we're going to do any massive renaming, now is time. And I like big disruptive changes because it's fun and annoying!

MichaelDvP, 2022-12-31:
@tglynx Please check my latest build and show us the system info, and check api returns and commands.

@proddy I'll do a PR so you check the code. Wait for confirmation or change requests before merging.

tglynx, 2022-12-31:
Installed latest build.

api/system: 

```
{
  "System Info": {
    "version": "3.5.0-dev.15",
    "platform": "ESP32",
    "uptime": "000+00:06:03.856",
    "uptime (seconds)": 363,
    "free mem": 107,
    "max alloc": 48,
    "free app": 27,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "94:B5:55:FF:DF:D7",
    "IPv4 address": "192.168.123.146/255.255.255.0",
    "IPv4 gateway": "192.168.123.254",
    "IPv4 nameserver": "192.168.123.254",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
[... 196 lines cut ...]
2022-12-31 18:07:31.680   ERROR 30:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/valvereturn' (Error)
2022-12-31 18:07:31.680   ERROR 31:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/altpumpmod' (Error)
2022-12-31 18:07:31.680   ERROR 32:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/vr2config' (Error)
2022-12-31 18:07:31.801   ERROR 33:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/ahsactivated' (Error)
2022-12-31 18:07:31.801   ERROR 34:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/apumpconfig' (Error)
2022-12-31 18:07:31.801   ERROR 35:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/apumpsignal' (Error)
2022-12-31 18:07:31.801   ERROR 36:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/apumpmin' (Error)
2022-12-31 18:07:31.801   ERROR 37:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/temprise' (Error)
2022-12-31 18:07:31.801   ERROR 38:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/setreturntemp' (Error)
2022-12-31 18:07:31.801   ERROR 39:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/mixruntime' (Error)
2022-12-31 18:07:31.801   ERROR 40:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/setflowtemp' (Error)
2022-12-31 18:07:31.801   ERROR 41:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/bufbypass' (Error)
2022-12-31 18:07:31.801   ERROR 42:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/bufmixruntime' (Error)
2022-12-31 18:07:31.801   ERROR 43:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/bufconfig' (Error)
2022-12-31 18:07:31.801   ERROR 44:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/blockmode' (Error)
2022-12-31 18:07:31.801   ERROR 45:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/blockterm' (Error)
2022-12-31 18:07:31.801   ERROR 46:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/blockhyst' (Error)
2022-12-31 18:07:31.801   ERROR 47:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1/releasewait' (Error)
2022-12-31 18:12:53.065   ERROR 48:     [emsesp]     Call failed with error: cannot find values for entity 'ahs1' (Error)
```

i like the approach with ahs1 as this does not conflict with hs1...hs2... 

> And I like big disruptive changes because it's fun and annoying!
Yes ... at the same time :)

MichaelDvP, 2022-12-31:
> api/heatsource/ahs1:

try again.

tglynx, 2022-12-31:
/api/heatsource returns
```
{
  "ahs1": {
    "altflowtemp": 69.0,
    "cyltoptemp": 55.7,
    "cylcentertemp": 59.3,
    "cylbottomtemp": 54.0,
    "valvebuffer": 50,
    "valvereturn": 50,
    "altpumpmod": 50,
    "vr2config": "bypass",
    "ahsactivated": "off",
    "apumpconfig": "on",
    "apumpsignal": "off",
    "apumpmin": 50,
    "temprise": "off",
    "setreturntemp": 60,
    "mixruntime": 120,
    "setflowtemp": 70,
    "bufbypass": "no",
    "bufmixruntime": 120,
    "bufconfig": "monovalent",
    "blockmode": "blocking",
    "blockterm": "n_o",
    "blockhyst": 5,
    "releasewait": 60
  }
}
```

/api/heatsource/ahs1 returns
```
{
  "altflowtemp": 69.3,
  "cyltoptemp": 55.7,
  "cylcentertemp": 59.3,
  "cylbottomtemp": 54.0,
  "valvebuffer": 50,
  "valvereturn": 50,
  "altpumpmod": 50,
  "vr2config": "bypass",
  "ahsactivated": "off",
  "apumpconfig": "on",
  "apumpsignal": "off",
  "apumpmin": 50,
  "temprise": "off",
  "setreturntemp": 60,
  "mixruntime": 120,
  "setflowtemp": 70,
  "bufbypass": "no",
  "bufmixruntime": 120,
  "bufconfig": "monovalent",
  "blockmode": "blocking",
  "blockterm": "n_o",
  "blockhyst": 5,
  "releasewait": 60
}
```

http://192.168.123.146/api/heatsource/ahs1/altflowtemp (as an example) returns
```
{
  "name": "altflowtemp",
  "fullname": "ahs1 alternative hs flow temperature",
  "circuit": "ahs1",
  "value": 69.8,
  "type": "number",
  "uom": "°C",
  "readable": true,
  "writeable": false,
  "visible": true
}
```

looks good for the ems-esp part i would say...
No errors in log when walking the API with the ioBroker adapter.

@tp1de will need to adjust the ioBroker adapter to support the heatsource structure in ioBroker now
Let me know if input is needed here ...

From my point of view it would be heatsource/ahs1/...entities... in native ems-esp mode and heatSources/ahs1/...entities... in KM200 mode.

proddy, 2023-01-05:
@tp1de can you check this again against the latest dev build and either comment or close it. thanks

tp1de, 2023-01-05:
> can you check this again against the latest dev build and either comment or close it. thanks

Everything works from my point of view ... I adjusted my ioBroker adapter to new version as well.

closed 2023-01-05 by tp1de

Snypes88, 2023-01-11:
Hello everyone, not sure if this is the right place to comment, but I also have an AM200 in my heating system and use a flue gas sensor on TF1 to measure the exhaust gas temperatures from my wood stove with a water jacket. My plan is when the temperatures fall below a certain value I trigger a notification through HA and know to throw some wood back on the fireplace without having to check every 30 mins XD. I do not see the flue gas temperature reading in the EMS ESP dashboard but do on my RC310. Anyone have the same behavior/use case?

Snypes88, 2023-01-11:
Hi Michael, thanks for the super quick response! 

I have attached a picture, however, the PT1000 isn't yet connected to the exhaust pipe, hence the low temperature reading as I'm still experimenting :) Tomorrow I should get around to installing the sensor in the exhaust and will add the telnet output when the fire is burning and I am picking up a feasible temperature reading from the sensor. So far when I run watch 60 I get the following from telnet

000+01:41:44.832 N 8: [emsesp] heatsource(0x60) -B-> All(0x00), AmTemperatures(0x054D), data: 80 00 80 00 80 00 02 BB 02 6E 02                                6C 80 00 02 44 00 00 80 00 01

would a screenshot from my EMS ESP dashboard also help?

do you mind 

[image]

tglynx, 2023-01-11:
okay, nice... you could already experiment with your current config by warming the PT1000 with your hands and see if you can identify the changing values in the AmTemperatures message (still assuming it is in that message) ... i also do not have a return temperature sensor but that value is implemented as far as i can see (so you can see AHS return temp in the dashboard, right)?

Snypes88, 2023-01-11:
Yea, a lot of Temperature readouts are actually already supported in the dashboard which is really great! It seems I'm quite lucky that support was recently added and so much dev work has been taking place to better the AM200 support 👌 I was experimenting with a heat gun and running upstairs to the RC310 controller and then back down to the AM200 in the basement and it is definitely picking up and spitting out feasible information! And seeing as it at least works in the EMS+ I will wait till my temperature resistant PT1000 arrives tomorrow and I get around to properly installing the sensor to the exhaust pipe to ping telnet.

I also already created an enhancement here https://github.com/emsesp/EMS-ESP32/issues/906 as per your recommendation, so if you prefer to continue the conversation I'm game 🙌

commit 29838a43 (2023-01-05): fix #857, add class heatsource