emsesp/EMS-ESP32 #860: Bosch Condens 5000I: dhw activation / deactivation no function

Hans-JR, 2022-12-29:
Boiler: Bosch Condens 5000I
Thermostat: Bosch CW 400
EMS-ESP: ESP32 with BBQKees Interface V3
Software: V3.44

**Question**
Changing the "dhw activated ON/OFF" doesn not work.
Changing the "heating activated ON/OFF" is OK.
Changing hc1 manual temperature" changing is OK. (I havent tried any other datapoint yet)

The system is running in "Auto" Mode.

**Device information**
{
  "System Status": {
    "version": "3.4.4",
    "uptime": "014+00:34:18.691",
    "freemem": 120,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -58,
    "IPv4 address": "192.168.178.89/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
[... 130 lines cut ...]

Man-Temp Change:

2022-12-29 09:57:15.757 D 2595: [command] Calling command 'system/info' (show system status)
2022-12-29 09:57:15.967 D 2596: [command] Calling command 'boiler/values' ()
2022-12-29 09:57:15.967 D 2597: [command] Calling command 'thermostat/manualtemp' (manual temperature) with value 19
2022-12-29 09:57:15.967 D 2598: [emsesp] Write command successful
2022-12-29 09:57:16.156 D 2599: [telegram] Sending write Tx [#81], telegram: 8B 10 FF 0A 01 B9 26
2022-12-29 09:57:16.156 D 2600: [emsesp] No telegram type handler found for ID 0x2B9 (src 0x0B)
2022-12-29 09:57:16.188 D 2601: [emsesp] Last Tx write successful
2022-12-29 09:57:16.188 D 2602: [telegram] Sending post validate read, type ID 0x2B9 to dest 0x10
2022-12-29 09:57:16.370 D 2603: [telegram] Sending read Tx [#82], telegram: 8B 90 FF 00 19 01 B9
2022-12-29 09:57:16.470 D 2604: [emsesp] Last Tx read successful
2022-12-29 09:57:16.502 D 2605: [command] Calling command 'thermostat/values' ()
2022-12-29 09:57:16.502 D 2605: [command] Calling command 'thermostat/values' ()
2022-12-29 09:57:18.621 D 2606: [emsesp] No telegram type handler found for ID 0xE7 (src 0x10)
2022-12-29 09:57:19.487 D 2607: [emsesp] No telegram type handler found for ID 0x2E0 (src 0x10)
2022-12-29 09:57:19.515 D 2608: [emsesp] No telegram type handler found for ID 0x2EA (src 0x10)
2022-12-29 09:57:52.876 D 2609: [emsesp] Fetching values for deviceID 0x08
2022-12-29 09:57:52.876 D 2610: [telegram] Tx read request to device 0x08 for type ID 0x14
2022-12-29 09:57:52.876 D 2611: [telegram] Tx read request to device 0x08 for type ID 0xE6
2022-12-29 09:57:52.876 D 2612: [telegram] Tx read request to device 0x08 for type ID 0xEA
2022-12-29 09:57:52.876 D 2612: [telegram] Tx read request to device 0x08 for type ID 0xEA

*Add any other context about the problem here.*

MichaelDvP, 2022-12-29:
Looks to me as the offset is wrong. 
Try if update to https://github.com/MichaelDvP/EMS-ESP32/releases works.

Hans-JR, 2022-12-30:
Hmm....
I tried to upload via OTA.
Got the message "Upload Failed. Internal Server Error"

Looks like I have to do a Serial Programming.
Is this still valid?:

[image]

MichaelDvP, 2022-12-30:
Yes this will work, but https://github.com/emsesp/EMS-ESP-Flasher/releases is easier.

Hans-JR, 2022-12-31:
Good news: The software update to Version 3.50.dev14 solved this issue.
Thank you!

commit bfcdf3ef (2023-01-05): fix #860, wwactivated in offset 5