emsesp/EMS-ESP32 #862: Upgrade to platform-espressif32 5.3.0

proddy, 2022-12-30:
@MichaelDvP 

platform-espressif32 5.3.0 was just released (https://github.com/platformio/platform-espressif32/releases/tag/v5.3.0)
and compared to 5.2.0 both RAM and Flash have gone up quite considerably. Shall we hardcode EMS-ESP to use 5.2.0 ?

_Originally posted by @proddy in https://github.com/emsesp/EMS-ESP32/discussions/388#discussioncomment-4552263_

proddy, 2022-12-30:
this needs some investigation. Also look into moving from C++11 to C++17

proddy, 2022-12-30:
In the compilation RAM went up by 5768 bytes, Flash by 24260 bytes and when running the available heap went down by 7KB. This is using v3.5.0-dev.14. 

For EMS-ESP v3.5.0 I suggest stick to espressif32 5.2.0

MichaelDvP, 2022-12-30:
My testbuild with some extra debug messages compiled with c++17 and platform 5.3.0:
ESP32 fits in partition with 627 bytes left. (S3 is 16M flash, C3 binary is larger than reported by build, can't upload to symetric ota1/ota2 partitions)
```
ESP32
RAM:   [==        ]  17.5% (used 57288 bytes from 327680 bytes)
Flash: [==========]  100.0% (used 2030989 bytes from 2031616 bytes)

ESP32 C3
RAM:   [==        ]  15.6% (used 51276 bytes from 327680 bytes)
Flash: [==========]  95.1% (used 1932090 bytes from 2031616 bytes)

ESP32 S2
RAM:   [==        ]  17.8% (used 58304 bytes from 327680 bytes)
Flash: [==========]  96.0% (used 1950086 bytes from 2031616 bytes)

ESP32 S3
RAM:   [==        ]  17.1% (used 55952 bytes from 327680 bytes)
Flash: [==        ]  23.4% (used 1946669 bytes from 8323072 bytes)
```
ESP32 flashed to MH-ET32 with factory partitions:
[image]

proddy, 2023-01-05:
I've created a new branch [tech-upgrade](https://github.com/emsesp/EMS-ESP32/tree/tech-upgrade) for testing these changes and monitoring memory usage and performance. These changes will most likely end up in EMS-ESP 3.6.0.

proddy, 2023-01-16:
and before you know it v6 is out with C+11 as default

https://github.com/platformio/platform-espressif32/releases/tag/v6.0.0

proddy, 2023-01-22:
with 4MB

5.2.0 it uses 15.7% RAM and 99.6% flash
5.3.0 and the latest 6.0.0 its 17.5% RAM and 100.8% flash (which doesn't fir the 4MB)

I still can't find why it the library uses so much more memory. I've been browsing the forums but haven't spotted anything yet

@MichaelDvP any ideas?

closed 2023-02-11 by proddy