emsesp/EMS-ESP32 #865: web api not returning entity details for second mixer (heating circuit 2)

tglynx, 2022-12-30:
Hi, this issue may be related to #857. When using the web API, details of an entity can be queried using the full path of an entity in the web API.

As an example http://ems-esp/api/boiler/wwsettemp returns details of the entity wwsettemp.

```
{
  "name": "wwsettemp",
  "fullname": "dhw set temperature",
  "circuit": "ww",
  "value": 13,
  "type": "number",
  "uom": "°C",
  "readable": true,
  "writeable": false,
  "visible": true
}
```

This  does not work with the mixer entities for the second mixer in the system.
Scenario: 

http://ems-esp/api/mixer returns

```
{
  "hc1": {
    "flowtemphc": 28.3,
    "valvestatus": 6,
    "flowsettemp": 29,
    "pumpstatus": "on"
  },
  "hc2": {
    "flowtemphc": 29.4,
    "valvestatus": 0,
    "flowsettemp": 0,
    "pumpstatus": "off"
  }
}
```

http://ems-esp/api/mixer/hc1/flowtemphc returns

```
{
  "name": "flowtemphc",
  "fullname": "hc1 flow temperature (TC1)",
  "circuit": "hc1",
  "value": 28.1,
  "type": "number",
  "uom": "°C",
  "readable": true,
  "writeable": false,
  "visible": true
}
```

Now the problem: 
http://ems-esp/api/mixer/hc2/flowtemphc returns

```
{
  "message": "cannot find values for entity 'flowtemphc'"
}
```

EMS-ESP logs the following errors when using the ioBroker Adapter to enumerate the hc2 entities:

```
001+03:01:10.313 E 203: [emsesp] Call failed with error: cannot find values for entity 'flowtemphc' (Error)
001+03:01:10.325 E 204: [emsesp] Call failed with error: cannot find values for entity 'valvestatus' (Error)
001+03:01:10.325 E 205: [emsesp] Call failed with error: cannot find values for entity 'flowsettemp' (Error)
001+03:01:10.325 E 206: [emsesp] Call failed with error: cannot find values for entity 'pumpstatus' (Error)
```

I would expect that the details of hc2 entities can be queried the same way as the details of the hc1 entities.

http://ems-esp/api/system returns

```
{
  "System Info": {
    "version": "3.5.0-dev.14",
    "platform": "ESP32",
    "uptime": "001+03:18:21.677",
    "uptime (seconds)": 98301,
    "free mem": 94,
    "max alloc": 38,
    "free app": 5,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "94:B5:55:FF:DF:D7",
    "IPv4 address": "192.168.123.146/255.255.255.0",
    "IPv4 gateway": "192.168.123.254",
    "IPv4 nameserver": "192.168.123.254",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 12610,
    "API fails": 46
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 434029,
    "bus reads (tx)": 45905,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 5,
    "bus reads failed": 12,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 5,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "web log buffer": 50
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Logamax Plus/GB192/Condens GC9000/Greenstar ErP",
      "device id": "0x08",
      "product id": 208,
      "version": "01.04",
      "entities": 73,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA",
      "handlers ignored": "0x2E 0x17"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.03",
      "entities": 79,
      "handlers received": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x02CE 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x02D0 0x02D2 0x031E 0xBB",
      "handlers ignored": "0x0292 0xF9 0x02E1 0x02E2 0x02EB 0x02EC 0x35 0x0291 0xBF"
    },
    {
      "type": "Thermostat",
      "name": "RC100/Moduline 1000/1010",
      "device id": "0x39",
      "product id": 165,
      "version": "12.03",
      "entities": 0,
      "handlers pending": "0x042B 0x047B 0x0273",
      "handlers ignored": "0x02BA 0x0292 0x042C 0x029C 0xF7"
    },
    {
      "type": "Mixer",
      "name": "MM100",
      "device id": "0x20",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D7",
      "handlers ignored": "0x1E 0x1A 0x0255 0xBF"
    },
    {
      "type": "Mixer",
      "name": "MM100",
      "device id": "0x21",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D8",
      "handlers ignored": "0x1A 0x0255 0xBF"
    },
    {
      "type": "Solar",
      "name": "SM100/MS100",
      "device id": "0x30",
      "product id": 163,
      "version": "27.07",
      "entities": 36,
      "handlers received": "0x0362 0x0363 0x0366 0x0364 0x036A",
      "handlers fetched": "0x0358 0x035A 0x0380 0x038E 0x0391",
      "handlers pending": "0x035D 0x035F 0x035C 0x0361",
      "handlers ignored": "0x0368 0x03A9 0x03AA 0x35 0xBF 0x038A"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.08",
      "entities": 0,
      "handlers ignored": "0x02B9 0x02F5"
    },
    {
      "type": "Controller",
      "name": "9000i",
      "device id": "0x09",
      "product id": 224,
      "version": "35.06",
      "entities": 0,
      "handlers ignored": "0x1A"
    },
    {
      "type": "Controller",
      "name": "BC30",
      "device id": "0x16",
      "product id": 220,
      "version": "01.05",
      "entities": 0
    },
    {
      "type": "Heatsource",
      "name": "Alternative Heatsource",
      "device id": "0x60",
      "product id": 228,
      "version": "03.02",
      "entities": 23,
      "handlers received": "0x054D 0x054E 0x0550",
      "handlers fetched": "0x054C",
      "handlers pending": "0x054F",
      "handlers ignored": "0xBF"
    }
  ]
}
```

proddy, 2022-12-30:
yup, a stinky bug. Thanks for reporting.

MichaelDvP, 2022-12-30:
I think same as #826, the  `get_device_value_info` needs to check the device_id to match the hc/wwc. I'll add it to my testbuild.

proddy, 2022-12-30:
thanks. I've added a test, `test mixer` should show the values

MichaelDvP, 2022-12-30:
@tglynx Please check https://github.com/MichaelDvP/EMS-ESP32/releases if it solves the issue.

tglynx, 2022-12-30:
Thanks, i tested that prerelease but something seems to be still broken ... 

http://192.168.123.146/api/boiler/wwsettemp
http://192.168.123.146/api/mixer/hc1/flowtemphc
http://192.168.123.146/api/mixer/hc1/flowtemphc
even
http://192.168.123.146/api/mixer/hc1/xxx  <- not existing

return just
`{}`

MichaelDvP, 2022-12-31:
Yes, this needs a different approach. Please try new build.

tglynx, 2022-12-31:
This is working in the latest build... Great work!

proddy, 2023-01-02:
nice, thanks Michael

commit 55750844 (2023-01-05): fix #865
commit b16a16d1 (2023-01-05): 2. fix #865
closed 2023-01-02 by proddy