emsesp/EMS-ESP32 #871: Change value: burner selected max power (selburnpow) does not have any effect

hansij66, 2022-12-31:
**Bug description**
I am trying to modify burner selected max power in dashboard - boiler.
Any change via EMS-ESP does not have any effect on max power setting of my Nefit.
If I change the max power settings directly on my Nefit, I do see the change in EMS-ESP

LOG:
2022-12-31 15:24:14.276 I 749: [command] Calling boiler command 'selburnpow', value 67, id is default
2022-12-31 15:24:14.276 I 750: [boiler] Setting burner max power to 67 

I am able to change other settings successfully.

**Expected behavior**
That max burn power setting is modified in the Nefit CV.

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api/system*
I only get { "message": "unknown device"} as output

EMS-ESP v3.3.1b1

Device (Platform / SDK) ESP32 / v3.3.5-1-g85c43024c

proddy, 2022-12-31:
Hi Hans, hope you're well! You're still on a really old un-supported version of EMS-ESP that's 12 months old. Can you upgrade to the latest dev version (3.5.0) and try again, just to rule out it being an old bug we may have fixed. 

On my Nefit HRC 30 I set the burner max power to 50% which lowers the kW to 15kW using that command.

hansij66, 2022-12-31:
Hey Paul
I am doing fine; thanks! And you?
I will update to latest version later this week. Is the 3.5 dev stable
enough? Or go for 3.4 production version?
I also have a HRC30, good to know that it should work.
Have a nice Sylvester and happy 2023!
BR Hans

proddy, 2022-12-31:
Go to 3.5.0. It's pretty stable. We should have released it earlier but keep adding to it! If you have one of Kees' boards then read this: https://emsesp.github.io/docs/Getting-Started/#upgrading-from-a-previous-release

See you in 2023!

hansij66, 2023-01-01:
Hi,

I was able to upgrade to 3.4 via WebUI System/Upload (but it does not solve
my problem)
Upgrading to 3.5 failed in the System/Upload (internal server error)
I do have a latest BBQKees E32 board (with internal usb connector)....is
this expected? As the documentation refers to older boards, while I have
the latest version.
Only way to upgrade to 3.5 is via USB/EMS-ESP flasher....or can I also use
OTA? Documentation is not xtal clear on this.

BR Hans

proddy, 2023-01-02:
To go to 3.5 you'll need to use the Flash tool - see https://emsesp.github.io/docs/Getting-Started/#upgrading-from-a-previous-release

hansij66, 2023-01-03:
I updated to v3.5
I did get it to work (but I guess it did work for the older versions).
I was changing the wrong parameter (selburnpow vs burnmaxpower); the latter works fine. Not sure what the first one is doing, as you can change this in EMS-ESP dashboard.

closed 2023-02-19 by proddy