emsesp/EMS-ESP32 #874: Plural of "days" under EMS Status not working for Swedish

mvjt, 2023-01-01:
**Bug description**
It seems plural for "days" for EMS Status is not working for Swedish but works for other languages. 
Day Singular = Dag
Day Plural = Dagar

In index.ts
NUM_DAYS: '{num} dag{{ar}}',

Screenshot where it should say "dagar" and not "dag"
[image]

Version: v3.5.0-dev.14
KONTROLLERA VERSION

MichaelDvP, 2023-01-01:
I think this is because the pl base have no fixed part: `NUM_DAYS: '{num} {{dni|dzień|dni|dni|dni}}',`
Changing the pl to: `NUM_DAYS: '{num} d{{ni|zień|ni|ni|ni}}',` should work,
 or set all other languages to `NUM_DAYS: '{num} {{dag|dagar}}',`

mvjt, 2023-01-01:
> I think this is because the pl base have no fixed part: `NUM_DAYS: '{num} {{dni|dzień|dni|dni|dni}}',`
> Changing the pl to: `NUM_DAYS: '{num} d{{ni|zień|ni|ni|ni}}',` should work,
> or set all other languages to `NUM_DAYS: '{num} {{dag|dagar}}',`

Thanks Michael, I'm not sure I understand the syntax but I used the EN one as reference...

proddy, 2023-01-01:
I think it's something else that is causing this. And it only happens in the Bus Status and only for SE. It renders correctly in the Dashboard for the "* burner operating time".

MichaelDvP, 2023-01-01:
For many days like in "burner operating time" the SE also renders correctly. And for minutes, hours also all languages work and PL defines cases for 1, 2, few, many, other languages only single/many. The only difference i see is that days are not defined as fixedpart/suffix, but use cases for each plural. My emsesp rarly reaches 2 days uptime without updating. Difficult to check.

proddy, 2023-01-01:
yeah, this is a weird one and related to how the page is rendered are the variables loaded. I have a workaround which I'll push shortly.

proddy, 2023-01-01:
I found the problem after tracing it down to internal JS library `Intl.PluralRules`. Turns out we're using the wrong locale for Swedish. It's SV and not SE (ISO 639). doh! Will do a PR...

mvjt, 2023-01-01:
Nice work. Translations finally look really good too. The only missing piece is #872.

commit 20627035 (2023-01-05): fix #874 plural of days
closed 2023-01-02 by proddy