emsesp/EMS-ESP32 #878: Auxiliary electrical heater advance settings

MichScha, 2023-01-01:
I had a little bit of time and just logged some settings for my auxiliary heater that would be awesome to have be included.

It is possible to configure the auxiliary electrical heater for specific jobs how much power it can consume.
There are 3 options available + Bivalenzpoint. Each option can use predefined values (0,2,4,6,9)
- limiting with compressor (in KW)
- Limit the auxiliary electrical heater itself (in KW)
- Limit auxiliary electrical heater for hot water (in KW)
- Bivalenzpoint (Celcius -30 to 40)

Telegrams
limiting with compressor (in KW)
```
-> 0
000+00:17:25.443 N 1465: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 00 (offset 2)
-> 2
000+00:16:14.405 N 1436: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 01 (offset 2)
-> 4
000+00:14:25.330 N 1371: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 03 (offset 2)
->6
000+00:12:31.442 N 1349: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 04 (offset 2)
-> 9
000+00:17:57.381 N 1477: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 05 (offset 2)
```
Limit the auxiliary electrical heater itself (in KW)
```
0
000+00:18:43.018 N 1488: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 00 (offset 3)
2
000+00:19:12.056 N 1528: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 01 (offset 3)
4
000+00:19:38.868 N 1541: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 03 (offset 3)
6
000+00:20:41.620 N 1580: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 04 (offset 3)
9
000+00:20:27.331 N 1571: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 05 (offset 3)
```

Limit auxiliary electrical heater for hot water (in KW)
```
0
000+00:21:19.706 N 1685: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 00 (offset 4)
2
000+00:21:31.744 N 1711: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 01 (offset 4)
4
000+00:21:44.543 N 1732: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 03 (offset 4)
6
000+00:21:59.056 N 1753: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 04 (offset 4)
9
000+00:22:12.756 N 1780: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HpHeaterConfig(0x0492), data: 05 (offset 4)
```

Bivalenzpoint (Celcius -30 to 40)
```
1
000+00:23:06.956 N 1891: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPAdditionalHeater(0x0491), data: 01 (offset 5)
2
000+00:23:29.906 N 1940: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPAdditionalHeater(0x0491), data: 02 (offset 5)
```

It is possible to Set a maximum temperatur for the Auxiliary electrical heater (off or value between 0.1 and 10.0 k)
```
->0.1
000+00:26:03.294 N 2241: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPAdditionalHeater(0x0491), data: 01 (offset 14)
->0.2
000+00:26:33.731 N 2311: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPAdditionalHeater(0x0491), data: 02 (offset 14)
->OFF
000+00:26:51.694 N 2332: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPAdditionalHeater(0x0491), data: 00 (offset 14)
```

It is possible to set the heating mode after EVU-Lock (ECO, COMFORT)
```
COMFORT
000+00:29:07.582 N 2595: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPAdditionalHeater(0x0491), data: 01 (offset 4)
ECO
000+00:29:45.207 N 2655: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPAdditionalHeater(0x0491), data: 00 (offset 4)
```

MichaelDvP, 2023-01-02:
Nice, but what version are you using. The limits should be already there, the bivalenz point is named "outside temperature parallel mode" (tempparmode). I don't know what is right.

I'll add the other values to my repo and the already opend PR.

MichScha, 2023-01-02:
I am using the latest beta. I have seen the values under Boiler but for me the values don´t fit to what RC300 says. 
[image]

All values are in the display of the RC300 are 6kw but the screens says different values. Changing those values in ems-esp doesn´t have an effect. Regarding bivalenz point - me neither. Maybe we name it like is displayed?

MichaelDvP, 2023-01-02:
Please use terminal to `read 8 492` and `read  10 492`
Maybe the values are also stored in thermostat. But normally the thermostat broadcasts it's own setting on change and writes to device if it is a device setting. So i expect the telegram is only stored in boiler.

MichScha, 2023-01-02:
ems-esp:$ read 8 492
000+12:41:06.486 I 0: [emsesp] Boiler(0x08) -W-> Me(0x0B), HpHeaterConfig(0x0492), data: 03 00 04 04 04

ems-esp:$ read  10 492
000+12:41:18.898 I 1: [emsesp] Thermostat(0x10) -W-> Me(0x0B), HpHeaterConfig(0x0492), data: 03 00 04 04 04

MichaelDvP, 2023-01-02:
Strange that emsesp shows different values, it reads from boiler.
Check whch one is writeable.
`call system send "0b 08 FF 02 03 92 03"` to set offset2 to 3
does it work and change the value permanent or is it overwritten by thermostat?

`call system send "0b 10 FF 02 03 92 03"` to set offset2 (maxHeatComp) to 3
does this work?

Please also check the new settings in https://github.com/MichaelDvP/EMS-ESP32/releases maybe for telegram 491 it's the same.

MichScha, 2023-01-02:
**Aux heater**
Changing through thermostat does work and is still present. Strange that in ems-esp it still shows the 6kw values from before.

- `call system send "0b 10 FF 02 03 92 03"`

- aux heater max temperature is in kelvin instead of celcius but value is correct. I have not tried it now but is it possible to set 0.1 as an example?

- heater aux heater mode - recognized correctly

**PV**
- PV settings looking all good (same as with temperature from aux heater - both values are with kelvin instead of celcius)

- Have you added - Cooling only with PV (YES, NO)? couldn´t find it

**Defrost**
- recognized correctly

MichaelDvP, 2023-01-02:
pvCooling should be there, also the limits. I've chnaged the requested to Kelvin.
Writing the limit is still to boiler, If it does not work we move the values to thermostat.

MichScha, 2023-01-02:
Alright just checked again with latest build.

Auxililiary
- Auxililiary does work now (r+w)!
- We are done with this part - great work!

PV
- just checked again regarding pvCooling. There seems to be still something wrong as also the console does not map it - still unknown
` Thermostat(0x10) -W-> Boiler(0x08), ?(0x0485), data: 01 (offset 21)`

I have missed another setting which initialy enables raising hw with PV (YES/NO)
->NO
000+00:33:40.857 N 2990: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0x023E), data: 00 (offset 3)
-> YES
000+00:34:06.682 N 3005: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0x023E), data: FF (offset 3)

MichScha, 2023-01-02:
Checked with latest build

PV features now work completely! tHx
I think I didn´t got the context right here. We should change the naming from "enable pv" to "enable raising ww" under the constraint of PV is been enable from an external contact.

MichScha, 2023-01-05:
Everything works like expected - tested. Nice!

closed 2023-01-05 by MichScha