emsesp/EMS-ESP32 #879: PV Settings

MichScha, 2023-01-01:
When enabling the pv settings through external contacts it is possible to set specific values for the heating system when the circuit for PV is closed.

Possible Settings
- Raise heating (in k from 0 to 5)
- Lower cooling (in k from -5 to 0)
- Cooling only with PV (YES, NO)

Telegrams (not all complete but I think those values can be calucalted though)
Raise heating (in k from 0 to 5)
```
->2
000+00:32:00.782 N 2889: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0x023E), data: 02
->3
000+00:32:57.682 N 2916: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0x023E), data: 03
```
Lower cooling (in k from -5 to 0)
```
-1
000+00:35:24.583 N 3034: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0x023E), data: FF (offset 5)
-2
000+00:35:54.545 N 3055: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0x023E), data: FE (offset 5)
-5
000+00:36:16.582 N 3058: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0x023E), data: FB (offset 5)
0
000+00:36:26.670 N 3069: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0x023E), data: 00 (offset 5)
```
Cooling only with PV (YES, NO)
```
YES
000+00:36:54.495 N 3117: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x0485), data: 01 (offset 21)
NO
000+00:37:15.370 N 3130: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), ?(0x0485), data: 00 (offset 21)
```

MichScha, 2023-01-02:
Latest dev-build from @MichaelDvP added all features. Works and can be closed with merge.

MichScha, 2023-01-05:
@proddy Part of release tag 3.5? As it is included in #870 and does work like expected.

proddy, 2023-01-05:
Yup, thanks. And we can probably close this if you have verified it.

closed 2023-01-05 by MichScha