emsesp/EMS-ESP32 #882: Unrecognized devices on EMS-bus

cgeboers, 2023-01-02:
After upgrading to 3.4.4 I see these errors in the EMS-ESP32 logs:

2023-01-02 10:26:00.736 W 44: [emsesp] Unrecognized EMS device (device ID 0x69, no product ID). Please report on GitHub.
2023-01-02 10:26:58.337 E 45: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E8 02 00 1B
2023-01-02 10:26:58.337 W 46: [emsesp] Unrecognized EMS device (device ID 0x68, no product ID). Please report on GitHub.
2023-01-02 10:27:02.503 E 47: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E9 02 00 1B
2023-01-02 10:27:02.503 W 48: [emsesp] Unrecognized EMS device (device ID 0x69, no product ID). Please report on GitHub.
2023-01-02 10:27:58.712 E 49: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E8 02 00 1B
2023-01-02 10:27:58.712 W 50: [emsesp] Unrecognized EMS device (device ID 0x68, no product ID). Please report on GitHub.
2023-01-02 10:28:01.144 E 51: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E9 02 00 1B
2023-01-02 10:28:01.144 W 52: [emsesp] Unrecognized EMS device (device ID 0x69, no product ID). Please report on GitHub.
2023-01-02 10:28:15.476 W 53: [telegram] Incomplete Rx: 08 00 18 01 30 80 00 64 0F 09 01 A5 40 80 00 02 33 80 00 FF FF FF 2D 48 00 C8 00 02 00
2023-01-02 10:28:58.310 E 54: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E8 02 00 1B
2023-01-02 10:28:58.310 W 55: [emsesp] Unrecognized EMS device (device ID 0x68, no product ID). Please report on GitHub.
2023-01-02 10:29:00.745 E 56: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E9 02 00 1B
2023-01-02 10:29:00.745 W 57: [emsesp] Unrecognized EMS device (device ID 0x69, no product ID). Please report on GitHub.
2023-01-02 10:29:58.621 E 58: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E8 02 00 1B
2023-01-02 10:29:58.621 W 59: [emsesp] Unrecognized EMS device (device ID 0x68, no product ID). Please report on GitHub.
2023-01-02 10:30:01.278 E 60: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E9 02 00 1B
2023-01-02 10:30:01.278 W 61: [emsesp] Unrecognized EMS device (device ID 0x69, no product ID). Please report on GitHub.
2023-01-02 10:30:59.578 E 62: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E8 02 00 1B
2023-01-02 10:30:59.578 W 63: [emsesp] Unrecognized EMS device (device ID 0x68, no product ID). Please report on GitHub.
2023-01-02 10:31:04.065 E 64: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E9 02 00 1B
2023-01-02 10:31:04.065 W 65: [emsesp] Unrecognized EMS device (device ID 0x69, no product ID). Please report on GitHub.
2023-01-02 10:31:59.349 E 66: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E8 02 00 1B
2023-01-02 10:31:59.349 W 67: [emsesp] Unrecognized EMS device (device ID 0x68, no product ID). Please report on GitHub.
2023-01-02 10:32:01.659 E 68: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B E9 02 00 1B
2023-01-02 10:32:01.659 W 69: [emsesp] Unrecognized EMS device (device ID 0x69, no product ID). Please report on GitHub.

I do know a little bit about ESP32, but I don't know anything about the EMS bus. This particuar system has two boilers, which are connected to each other via a device which in Home Assistant is visible as a "Switch", but it doesn't display a Product ID.

Any suggestions to what I can add for information, like system logs or anything?

cgeboers, 2023-01-02:
After restarting my browser, I can see this on the dashboard

[image]

proddy, 2023-01-02:
Michael added a change for missing product IDs in v3.5. See if you can upgrade to the that dev version.

And read https://emsesp.github.io/docs/FAQ/ on what to attach to this GH issue which will help us support you better

MichaelDvP, 2023-01-02:
Yes, please update, but 0x68 0x69 are unknown also in v3.5. 
Please describe your system, how are the 2 boilers connected to the bus? Where is emsesp connected?
It looks like something like the cascade module, but with boilers mapped to 0x68/69 and the module as unknown boiler at device-id 0x08.
With v3.5 make a log (level all) of > 2 minutes and attach it here as file. Also load from help page the supportinfo and also attach it.

cgeboers, 2023-01-02:
I don't really know how the boilers are connected to the EMS bus. What I discovered in the past is that when I connect my EMS-ESP32 to the service bus connector (3.5mm) of one of the boilers, I can only see the data of that boiler end not my RC35 for example. Currently my EMS-ESP32 is connected to the bus of the "controller" and in that case I can see bothe boilers, as well as the RC35 thermostat.

I also upgraded to v3.5 and have attached the requested info.
[emsesp_log.txt](https://github.com/emsesp/EMS-ESP32/files/10331773/emsesp_log.txt)
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10331774/emsesp_info.txt)

MichaelDvP, 2023-01-02:
Thanks, in first step i have added the cascade module as boiler/controller. his should give you the most values.
The boilers go to heatsources. The 0x68/0x69 devices are strange, we have to determine in a second step.
Please update this version to test: https://github.com/MichaelDvP/EMS-ESP32/releases

cgeboers, 2023-01-02:
Thanks for the fast return! Looks better, I still see some unrecognized devices in the system log, but the device status shows no more unknown devices.
[image]

The heatsources however still only show 0 entities, is this intended behavior?

[emsesp_log.txt](https://github.com/emsesp/EMS-ESP32/files/10332272/emsesp_log.txt)

MichaelDvP, 2023-01-02:
> The heatsources however still only show 0 entities, is this intended behavior?

Actually yes. For EMS+ cascade the boilers behind the cascade reports current flowtemp, burnerpower and workingtime, that's all. For this module we have to wait if there is something reported, Check the help/support_info after a few hours if there are ignored handlers for these devices.

@proddy What to do with the 0x68/0x69 devices, they are only registered in boiler devices-0x07 telegram, but no reply to version request and no other communication in this small log. Maybe register as controller to stop version requests and do not show them to dashboard? In support info they will show up with possible sent telegrams. I've added a test to my build. 
I've fond on @NorbertS1 list that 0x68/0x69 is also heatingsource 1/2. But we have the two boilers with version info on 0x71/0x72 and i think there are no other boilers. 
@cgeboers do you have more than these two boilers? Please update to new version from metioned repo, let it run some hours or a day and post the suport info here. A better description of yur system could be usefull. Make some photos. Also select in RC35 service menu the version info and tell us. (menu/ok+info+return keys simultanious, then select diagnostic, versions)

proddy, 2023-01-02:
> @proddy What to do with the 0x68/0x69 devices, they are only registered in boiler devices-0x07 telegram, but no reply to version request and no other communication in this small log. 

@MichaelDvP how about having an ignore/block list of device-id's we simply ignore, if they don't service any purpose

cgeboers, 2023-01-03:
I ran v3.5.0-dev15 all night and here are the results attached.
[image]
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10335495/emsesp_info.txt)

My system has only two boilers:

[image]

The MCM10 is in the middle of the two.

[image]

The RC35 versions info:

[image]

Everything seems to work fine now, the dashboard shows this:

[image]

The dasboard status shows some failed reads:

[image]

closed 2023-02-19 by proddy