emsesp/EMS-ESP32 #883: Additional HW Settings

MichScha, 2023-01-02:
RC300 / WLW196i latest dev build ([v3.5.0-dev.15](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest))

It is possible to set daily heatup of hw
```
yes
000+00:16:00.201 N 648: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 F5 0F (offset 1)
no
000+00:16:20.188 N 705: [emsesp] Thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 F5 0D (offset 1)
```

Couldn´t find that option. Circulating pump avaiable during WW operation (YES/NO)
```
NO
000+00:16:50.049 N 761: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 00 (offset 46)
YES
000+00:17:30.462 N 830: [emsesp] Thermostat(0x10) -W-> Boiler(0x08), HPSilentMode(0x0484), data: 01 (offset 46)
```

MichaelDvP, 2023-01-03:
F7 is not a setting telegram, it configures the menu entries of Thermostat. In thsi case it shows/hides the entry of wwCharge.

Is circulation pump from telegram 0x484 a different setting to the entitiy "circulation pump available" from telegram 0xEA? Or  a successor? Check `read 8 ea` and `read 8 484`

MichScha, 2023-01-03:
I have changed "Circulating pump avaiable during WW operation" and it doesn´t had an effect on "circulation pump available"

```
ems-esp:$ read 8 ea
000+10:51:21.266 I 1: [emsesp] Boiler(0x08) -W-> Me(0x0B), UBAParameterWWPlus(0xEA), data: 01 30 FE 00 19 01 01 41 00 01 12 41 3C 2A 0A 38 64 64 00 (offset 5)

ems-esp:$ read 8 484
000+10:51:37.512 I 2: [emsesp] Boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: 01 00 23 2D 04 50 00 00 00 02 01 F6 28 03 02 02 05 19 0A 0A 03 0D 07 00 0A
000+10:51:37.825 I 3: [emsesp] Boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: 01 90 00 F6 00 00 64 00 00 E1 00 00 00 00 02 64 00 64 54 20 00 01 00 00 05 (offset 25)
000+10:51:38.219 I 4: [emsesp] Boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: 46 1C 00 00 00 00 00 (offset 52)
```

MichScha, 2023-01-05:
Great - works!

MichScha, 2023-01-05:
@proddy same as with the other issue if you want to tag it with rel 3.5 - tested and works.

closed 2023-01-05 by MichScha

proddy, 2023-01-05:
> @proddy same as with the other issue if you want to tag it with rel 3.5 - tested and works.

thanks. 3.5.0 is going to be a huge release ;-)